arendst/Tasmota #1852: WS2812b wiring

leobel96, 2018-02-10:
Hi to all, I'd like to control my WS2812b led strip using a nodeMCU and, reading [here](https://learn.adafruit.com/adafruit-neopixel-uberguide/powering-neopixels#driving-5v-neopixels-from-3-dot-3v-microcontrollers), they say that, for 3V3 microcontrollers, you need a logic level shifter. All the devices in this library are based on an ESP8266 that is a 3V3 controller but I haven't read anywhere in the wiki about using level logic shifter or the problematics using these devices with WS2812b. How is it possible? Is it the same to use or not the level shifter? Thank you.

arendst, 2018-02-10:
Power the WS2812 by 5V. Connect the data line via a 100 - 220 ohm resistor to the datapin of the ESP8266 and you're good to go.

leobel96, 2018-02-10:
Ok, thank you!

closed 2018-02-10 by leobel96