arendst/Tasmota #1853: Home assistant does not auto discover the sonoff rf bridge with 5.12.0

ErikAndren, 2018-02-10:
This might not be a problem with Tasmota but my home assistant setup.
I'm new to both tasmota and home assistant. 
I have discovery: in my home assistant.

I've flashed the 5.12 firmware onto a sonoff rf bridge and enabled SetOption19 On on the console.
Now auto discover should work right? 
But what should happen next? 
Should a new item on my home assistant dashboard be added automagically or do I still need to do some setup in my home assistant configuration?

arendst, 2018-02-10:
You will have to configure a MQTT server in Hass in order to use the MQTT discover feature. If done it should come up automagically as either a light or switch.

ErikAndren, 2018-02-10:
Ah, I missed that the home assistant mqtt configuration needs a discovery: true
Thanks.

ErikAndren, 2018-02-10:
Ok, my slampher gets autodetected now but my rfbridge does not.
Does it support auto discovery?

arendst, 2018-02-10:
Nope. Currently only lights (well lights) and switches (relays).

ErikAndren, 2018-02-10:
Ok, thanks for the clarification

closed 2018-02-10 by ErikAndren

stavrosso, 2018-02-13:
Hi Guys ,
I am new to home automation and to github.

@arendst thank you for creating the amazing tasmota firmware. I flashed 5 sonoff switches and my rf bridge and everything went smooth.

I have a Sonoff DW1 door sensor and also a Sonoff PIR sensor which came as a package with my Sonoff Bridge. I would appreciate some help on setting them up and using them in Hassio . 

Could you please direct me to the proper topic if available

Thank you in advance

Stavros