arendst/Tasmota #1854: Random switch ch1

daspid92, 2018-02-10:
Hi, I've a sonoff 4ch pro where sometimes the 1st relay is turned on without user input, in the logs seems that this is caused by a random change of Button 1 (Tag: ESP-APP Message: Button 1 level 0-1), what can I try to do? Is it possible to virtually disable the Button 1 GPIO through template changes?

arendst, 2018-02-10:
Had it too!

Happend to be a learned 433 code after tapping the button twice. The 433 signal was something in the air at that time (coming from my temperature node).

Clear the codes by learning a know 433 signal or clear all 433 code by pressing a button (I think S5) on the board.

daspid92, 2018-02-11:
Thanks @arendst , today I've cleared all the codes, it's really strage because it happened 2 times in 2 days and before other 2 times 2 weeks ago and 1 month ago. If this doesn't solve the random power on what could it be? A bad joint? What happens if I remap the button 1 in the template?