arendst/Tasmota #1855: Feature request: Add JSONP

fry2k, 2018-02-10:
I have a few sonoffs with tasmota and they are great! Thanks for this project. 

I use them currently only with alexa. I would like to control the devices also with a easy smartphone app with one click instead of handling all the stuff with different webinterfaces or setup a small homeserver.

As a web developer I would like to code a small progressive web app (pwa) for controlling the tasmota devices.
The problem is that a pwa is only running with HTTPS and the tasmotas aren't. If I try to fetch a status command like http://ip/cm?cmnd=Power it ends with an mixed content error: 
"Mixed Content: The page at 'https://tasmota.jessie/' was loaded over HTTPS, but requested an insecure resource 'http://192.168.2.161/cm?cmnd=Power'. This request has been blocked; the content must be served over HTTPS."

A solution would be an additional JSONP API (https://en.wikipedia.org/wiki/JSONP)

I think this could be a 3 liner at the webserver.ino method HandleHttpCommand (https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/webserver.ino#L1482) arround line 1482. Sadly I'm not familiar with this coding language.

Let's say we would use something like this: http://ip/cm?cmnd=Power&callback=parseResponse
With PHP the code would look like this
```
if (isset($_GET['callback']) && !empty($_GET['callback'])) {
    $message = $_GET['callback'].'('.$message.');';
}
```
which results in:
`parseResponse({"POWER":"OFF"});`

instead of
`{"POWER":"OFF"}`

We just have to check if the GET parameter callback is set and wrap the json result inside the value of the GET parameter with a opening and closing round bracket.

Maybe someone could code this small pice of code for me at a fork/patch and I could test it at one of my devices. This would be great!

closed 2018-02-10 by fry2k

davidelang, 2018-02-11:
can you make your app talk to a mqtt service? that won't have these sorts of 
issues (they will have different ones, but those won't take more resources on 
the sonoff to fix :-) )

fry2k, 2018-02-12:
If I understand it correct I would need a mqtt server somewhere for connecting the tasmota devices, right? It seems like there are free cloud servers or I have to setup a local home server.

And that's exactly what I don't want. I just want to go with my smartphone to domain.com/pwa, get the progressive web app with offline functionality, configure the devices and it should work out of the box. As simple as possible.

But it isn't cause of the HTTPS problems ;o)
Right now I added a 5 liner php file to my local webserver and this script wraps the tasmota calls inside a HTTPS environment. It works but I need this small script somewhere in the local network with a valid (self signed) certificate. Sadly not as easy for a user which isn't a programer.

davidelang, 2018-02-12:
There have been many requests to add new protocols to interact with the devices, 
but space on the devices is very tight, and so adding such support would require 
not having a lot of other features.

As such, it's extremely unlikely to be added.

Tasmota is a way to manage esp-8266 devices with an MQTT server, there is a 
limited ability to do things via http (you can send commands to the device, but 
you can't have the device notify you of things that happen on it)

You would need to find someone to fork tasmota to add the functionality you are 
asking for. This is going to be much more work that supporting a local mqtt and 
home automation server
k

fry2k, 2018-02-12:
yeah sure I understand and you are totally right. It is not a important feature we need and we should focus more important ones first.

I found already a solution for my problem with an additional local script so far which is ok for me ;)
I will keep an eye to the protocols here, maybe there will be a better solution for me in the future.

Having a small control app for the tasmotas without any local homeserver would be great and I thought I could code this for the community in a few hours without big changes at the project itself.

Currently pwas aren't compareable with native smartphone apps and the scope is very limited. Hopefully this will change in the future and we may will get some exclusions for local networks.

thanks for your time and your detailed answer!