arendst/Tasmota #1857: Pow: Wrong Date and Time

Kraehe1806, 2018-02-10:
Hi everybody,

I have only one issue on a Sonoff Pow: He always shows me the wrong Date and Time although I've set the right timezone and NTP-Server (same settings work on a Basic).

He shows: 
`00:00:36 MQT: tele/sonoff/STATE = {"Time":"1970.01.01 00:00:36","Uptime":0,`

Any idea???? Thanx in advance...

Frogmore42, 2018-02-10:
Unplug the device for at least 10 min and then plug back in.

Kraehe1806, 2018-02-10:
...does't work...

Frogmore42, 2018-02-11:
There are several people who have had issues with the time not being set correctly.  There several reasons why this can not work correctly.  Your log snippet indicates that NTP did not work correctly.  This can be because the NTP servers are not set correctly.  Take a look at some of the past issues to see how to solve that.  There are commands for setting and verifying the NTP servers.  Make sure your router/firewall is not set to deny Internet access to this particular device.  What I found was even after doing all of those things, I had a device or several that were still not getting the correct time.  I even flashed new FW from the web interface and it still didn't get the correct time.  It wasn't until I removed the device and flashed new FW and reset all the configuration that the first one worked.  Since I had others that were in the same state, I tried just unplugging the device from ALL power for an extended period of time (to give the RTC enough time to discharge completely and erase all knowledge of everything).  I then plugged it back in without doing anything else and it worked fine.

You didn't indicate exactly which version you are using, 5.12.1 has a startup change to delay getting time from NTP to help not spam the NTP servers when a house gets power/internet back and all devices turn on at the same time.

I would manually flash the device (not OTA) and reset it and leave it unpowered for an extended period 30min or more and then try again.

curzon01, 2018-02-11:
@Kraehe1806 : Would you like to tell us which firmware you are using? 

`Status 2`?
[fill information here.... ]

Time information `Status 7`?
[fill information here.... ]

Kraehe1806, 2018-02-11:
@curzon01 

`Status 2:
00:00:24 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.12.0a","BuildDateTime":"2018.02.11 12:12:47","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}`

`Status 7:
00:04:50 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:04:50 1970","Local":"Thu Jan 01 00:04:50 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":99}}`

curzon01, 2018-02-11:
`NtpServer1` ?

`NtpServer2` ?

`NtpServer3` ?

NTP servers are public ones or within your intranet?

If private one within intranet, what kind of NTP services are using (own setup, router ones, if router which router)?

If using own NTP server: Is this service running proberly? What's the result of checks from a PC?

If using public NTP: Is IP network for Sonoff Pow allowed to connect to NTP (firewall/child protection on router...)?

Kraehe1806, 2018-02-11:
@curzon01 

Mixed.....and they are working for other Sonnoff Devices.....like the Basic....

`#define NTP_SERVER1            "de.pool.ntp.org"       
#define NTP_SERVER2            "fritz.box"    
#define NTP_SERVER3            "192.168.x.x"`

curzon01, 2018-02-11:
Yes, but other devices are not the Sonoff Pow - they have, for e.g., other IP address...

1. Is the output of cmnd `NtpServer<x>` same as you have defined in source? Verify it.
2. Check if IP for Sonoff Pow is allowed to connect the internet (verify "Filter" n AVM Box)
3. What's the result using firmware with lib 2.3.0?

I'm wondering why people do not want to place some additonal few minutes first checking all what's possible on their own side, even support is given not all questions are answered and not all possible solution steps tried before - comparing the time arendst raised for this wonderful project including the awesome documentation for absolutely free this is nothing.

Please: Give some answers to all the questions above and try additonal suggested steps and we will see.

If you belief you might have better way for analysing and solve the problem - continue.

Frogmore42, 2018-08-02:
No, not when you respond to a closed issue with none of the relevant information necessary to help.  If you went help, please open a new issue and include all the information listed in the template.