arendst/Tasmota #1858: Prevent rapid relay changes

Leseratte10, 2018-02-10:
When writing MQTT / automation rules in a home automation system (I am usng ioBroker, but it doesn't matter which one), I ended up with a situation where I messed up the action triggers and ended up changing the power state of my Sonoff device dozens of times per second (I accidentally had a "change switch value" action triggered by a change of the switch value, so I had an endless loop of changing the sonoff relay on and off and on and so on). 

I fear that that might either damage the Sonoff or the lights / device attached to the sonoff if this happens more often - neither the sonoff nor lights are designed to be turned on and off all the time. 

Would it be possible for Tasmota to add some detection like "hey this value has been changed X times in the last second something is probably wrong" and ignore any further MQTT / external switch requests until restarted (and possibly report some kind of warning over MQTT)? Preferably with configurable values for the amount and time of changes needed to trigger that.

reloxx13, 2018-04-06:
@Leseratte10 sry das ich das hier vergewaltigen muss :D
ich glaub dein kommentar zur letzten übersetzung hast du wieder geöscht.
ich denke das ist geschmackssache. da man auch gateway statt tor schreibt.
ich denke das hat sich so schon eingedeutscht mit subnetmask.
wenn du findest subnetzmaske ist besser, dann änder es ruhig wieder.

Subscribe ist ebenfalls so ein technisches eingedeutschtes wort meiner meinung nach.

ich will mich da nicht streiten, wenn du es wieder änderst weiß ich auch bescheid und dann beibt es so :)

viele grüße

nils

Leseratte10, 2018-04-06:
Das passt wohl besser in #2375 als hier rein.