arendst/Tasmota #1859: Need some help compiling tasmota

mattpurnell, 2018-02-11:
Hi - I am currently trying to flash 3 Sonoff T1 Touch light switches. I have had some limited luck before with Arduino IDE and the single-button version of the Touch, but now I have switches with 2 buttons I need a later version that will allow updating of the module via Web/MQTT.

I pulled down the latest version of Sonoff-Tasmota, and also to align with most users (and Tutorials) I abandoned Arduini IDE and gone with PlatformIO (via ATOM). Unfortunately I am having worse luck now; the firmware flashes with no issues, but I am getting some odd output from the serial monitor. I have re-written the user_config.h file numerous times (as that is the only thing I changed), but keep getting the same result (see below).

Can anyone please let me know what some good next steps would be, to try?

```
PS C:\Users\mattp\Downloads\platformio\Sonoff-Tasmota-master> pio device monitor --port COM6 --baud 115200
--- Miniterm on COM6  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
00:00:13 WIF: Connected
00:00:13 DNS: Initialized
00:00:13 HTP: Web server active on �������������������������������.local with IP address 255.255.255.255
00:00:13 UPP: Multicast (re)joined
00:00:19 MQT: Attempting connection...
00:00:24 MQT: Connect failed to �����������������������������������������������������������������������������������������������������������
�������������������������������������������������������������������������������������������������������������������������������������������
���������:65535, rc -2. Retry in 255 sec
00:00:29 UPP: Multicast (re)joined

00:01:14 RSL: ��������������������������������������������������������������������������������������������������� = {"POWER":"�������������
��������������������������������������������������������������������������������������������������������������"}
00:01:14 RSL: ��������������������������������������������������������������������������������������������������� = �����������������������
����������������������������������������������������������������������������������������������������
00:02:03 RSL: ��������������������������������������������������������������������������������������������������� = {"Time":"1970-01-01T00:

```

reloxx13, 2018-02-11:
r u sure that you OPENED the project and not created new project from existing files within atom?
https://github.com/arendst/Sonoff-Tasmota/wiki/Beginner-Guide---Create-your-own-Firmware-Build#building-the-firmware

try another usb cable
try another pin connector cables

mattpurnell, 2018-02-12:
Hi, yes I'm sure I opened the project..

I tried over and over and eventually went back to a clean install on Arduino, and I see the exact same result, question marks instead of the variables I declared in user_config.h (which I have rewritten more times than I can recall). Tried different cables - but the result is so consistent I don't think it's corruption. It's like some variables are filling to define properly. I think I'll try a standard esp8266 next.

Failing that, !?  I'm running out of sanity.. I think I'll have to reinstall windows 10 on my laptop.

mattpurnell, 2018-02-22:
OK after completely rebuilding my laptop, I had someone suggest to me that I should change the CFG_HOLDER value - so I looked it up and set it to todays date; straight away it started flashing correctly. All good now!

closed 2018-02-22 by mattpurnell