arendst/Tasmota #1860: Webserver is EXTREMELY slow

sebringsc, 2018-02-11:
I successfully flashed my Sonoff with TASMOTA (most recent build from 2/9/18).  Currently, my sonoff is about 6 inches from my router and it takes ~5 minutes for the webserver to load if and when it does and almost always times out if I click on any of the buttons on the server.

Any suggestions on how to improve speed?

Frogmore42, 2018-02-11:
Being that close to the router is generally not a good idea, but that is not likely your problem.  Did you do any searching of the issues for others with webserver issues.  There are many people who have had issues for several different reasons, too many UPnP devices on their network (using emulation mode), different versions of software in use.  Take a look and see if any of the solutions there solve the issue you are seeing.  If not, come back with all the things you tried that didn't help the problem and I am sure you will get some help.

sebringsc, 2018-02-12:
I have done some searching/reading and am extremely new to all of this.

I've ensured that emulation is off, used the most recent version with PlatformIO to compile and upload.  I also updated the board drivers in the platformio.ini to platform = espressif8266@1.6.0 which actually made the web server more responsive and allowed me to get in there to play around and set my MQTT server but after a few minutes, it disappeared and won't connect.

I cannot ping the Sonoff but my router sees it however I can issue commands through the serial monitor.  I've ensured that the Sonoff says that webserver is running as well and have tried accessing with Chrome, IE and Firefox.  It may load after a fresh flash of the device but after that, nothing.  I get an error in chrome that says -ERR_CONNECTION_TIMED_OUT

Here is my Serial Monitor:
05:25:52 CMD: webserver
05:25:52 MQT: stat/sonoff1/RESULT = {"Webserver":"Active for Admin on sonoff1-6383 with IP address 192.168.1.38"}
05:25:52 CMD:
05:25:52 MQT: stat/sonoff1/RESULT = {"Command":"Unknown"}

I'm at a loss now.  As I mentioned, after a fresh flash, it works for ~5 min but then that's it.  It just stops

oOPsychokillerOo, 2018-02-12:
Same here, the Sonoff RF was extremely slow on the web interface after the update from 5.10 to 5.12. After a few minutes, the web interface was no longer available! After a reset, I managed after a few attempts to flash the version back to 5.10 and the problem was gone immediately!

sebringsc, 2018-02-12:
@oOPsychokillerOo 
Thanks for the suggestion.  I have not gone back and tried to use an older version of firmware.  I will have to try that tonight when I get home.  I'll update tonight if I get the chance.

sebringsc, 2018-02-13:
@oOPsychokillerOo 
Flashed back to 5.10 and it works like a charm and has been up and running for over an hour after and I have moved it to 2 different outlets.

Now if I could just get HA to recognize it

closed 2018-02-13 by sebringsc