arendst/Tasmota #1862: Problem with Home Assistant

leobel96, 2018-02-11:
Hi. I have set my Home-Assistant configuration.yaml file as suggested in this way:
```
mqtt:
  broker: 192.168.1.72
  port: 1883
  client_id: sonoff1
  keepalive: 60
  username: !secret mqtt_user
  password: !secret mqtt_password
  protocol: 3.1
  # birth_message:
    # topic: "tele/sonoff/LWT"
    # payload: "Online"
    # qos: 1
    # retain: true
  # will_message:
    # topic: "tele/sonoff/LWT"
    # payload: "Offline"
    # qos: 1
    # retain: true
    
switch:
  - platform: mqtt
    name: "Sonoff power"
    state_topic: "cmnd/sonoff/Power1"
    command_topic: "cmnd/sonoff/Power1"
    availability_topic: "tele/sonoff/LWT"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    retain: true

```
And Tasmota in this way (where Host is the HA and Mosquitto IP):
[image]

But I keep obtaining this thing on Tasmota's console where the number of "stat/sonoff/RESULT" lines depends by how many times I switched on and off the switch on Home Assistant.
```
12:39:20 MQT: Tentativo di connessione...
12:39:20 MQT: Connesso
12:39:20 MQT: tele/sonoff/LWT = Online (salvato)
12:39:20 MQT: cmnd/sonoff/POWER = 
12:39:21 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:21 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:21 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:21 MQT: Tentativo di connessione...
12:39:21 MQT: Connesso
12:39:21 MQT: tele/sonoff/LWT = Online (salvato)
12:39:21 MQT: cmnd/sonoff/POWER = 
12:39:22 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:22 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:22 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:22 MQT: Tentativo di connessione...
12:39:22 MQT: Connesso
12:39:22 MQT: tele/sonoff/LWT = Online (salvato)
12:39:22 MQT: cmnd/sonoff/POWER = 
12:39:23 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:23 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:23 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:23 MQT: Tentativo di connessione...
12:39:24 MQT: Connesso
12:39:24 MQT: tele/sonoff/LWT = Online (salvato)
12:39:24 MQT: cmnd/sonoff/POWER = 
12:39:24 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:24 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:24 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:25 MQT: Tentativo di connessione...
12:39:25 MQT: Connesso
12:39:25 MQT: tele/sonoff/LWT = Online (salvato)
12:39:25 MQT: cmnd/sonoff/POWER = 
12:39:25 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:25 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:25 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:26 MQT: Tentativo di connessione...
12:39:26 MQT: Connesso
12:39:26 MQT: tele/sonoff/LWT = Online (salvato)
12:39:26 MQT: cmnd/sonoff/POWER = 
12:39:26 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:26 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:26 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:27 MQT: Tentativo di connessione...
12:39:27 MQT: Connesso
12:39:27 MQT: tele/sonoff/LWT = Online (salvato)
12:39:27 MQT: cmnd/sonoff/POWER = 
12:39:27 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:27 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:27 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:28 MQT: Tentativo di connessione...
12:39:28 MQT: Connesso
12:39:28 MQT: tele/sonoff/LWT = Online (salvato)
12:39:28 MQT: cmnd/sonoff/POWER = 
12:39:29 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:29 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:29 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:29 MQT: Tentativo di connessione...
12:39:29 MQT: Connesso
12:39:29 MQT: tele/sonoff/LWT = Online (salvato)
12:39:29 MQT: cmnd/sonoff/POWER = 
12:39:30 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:30 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:30 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:30 MQT: Tentativo di connessione...
12:39:31 MQT: Connesso
12:39:31 MQT: tele/sonoff/LWT = Online (salvato)
12:39:31 MQT: cmnd/sonoff/POWER = 
12:39:31 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:31 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
12:39:31 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
```
I'm using Mosquitto as MQTT broker on the same Raspbian distribution in which I've installed Home-Assistant. Thank you!

aferraz, 2018-02-11:
Hi @leobel96 
 you config, Has side, is wrong. Should be:
switch:
  - platform: mqtt
    name: "Sonoff power"
    state_topic: "**stat**/sonoff/Power1"
    command_topic: "cmnd/sonoff/Power1"
    availability_topic: "tele/sonoff/LWT"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    retain: true

Alessandro

leobel96, 2018-02-11:
Hi @aferraz 
I tried changing that line but nothing changed. I also noted that, if I press the switch in the HA frontend, the switch returns immediately to off position.

aferraz, 2018-02-11:
Hi @leobel96 
 I have the same configuration in HA and it work, could be a mutt broker problem.
Following my config:
- platform: mqtt
  name: "sonoff10"
  state_topic: "stat/sonoff10/POWER"
  command_topic: "cmnd/sonoff10/power"
  qos: 1
  payload_on: "ON"
  payload_off: "OFF"
  optimistic: false
  availability_topic: "tele/sonoff10/LWT"
  payload_available: "Online"
  payload_not_available: "Offline"   
  retain: true

leobel96, 2018-02-11:
Are you using mosquitto as broker? If yes, could you please tell me you configuration? Are you using the Hassbian distribution? Finally, could you please post a screen of your tasmota configuration? Thank you!

aferraz, 2018-02-11:
I use a Cento 7 Virtual Machine for HASS and another VM for mosquitto.
Tasmota side my configuration is like yours. 
Mosquitto side I don't change the default configuration.
You can try to subscribe on mosquitto topic, from Hassbian command line:
mosquitto_sub -v -t 'cmnd/sonoff/Power1'
and see what coming from HASS on the specific topic

leobel96, 2018-02-12:
I reinstalled mosquitto on Hassbian, reconfigured it creating again user:password file and modified the default mosquitto.conf file to indicate the new file. Then I noted that, if disable mqtt platform from HA, the problem is fixed while if I activate it the problem returns.

aferraz, 2018-02-12:
@leobel96 what about "mosquitto_sub -v -t 'cmnd/sonoff/Power1'" results ?

leobel96, 2018-02-12:
@aferraz with MQTT component disabled from configuration.yaml file, I obtain this thing in Tasmota's console running that:
`21:06:13 MQT: tele/sonoff/STATE = {"Time":"2018-02-12T21:06:13","Uptime":"0T00:13:14","Vcc":2.994,"Wifi":{"AP":2,"SSId":"WI-FI_EXT","RSSI":88,"APMac":"20:0C:C8:0D:B8:AE"}}`
all on one line.
Adding MQTT component in configuration.yaml I obtaint this thing in Tasmota's console continuosly:
```
21:29:16 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
21:29:16 MQT: Tentativo di connessione...
21:29:16 MQT: Connesso
21:29:16 MQT: tele/sonoff/LWT = Online (salvato)
21:29:16 MQT: cmnd/sonoff/POWER = 
21:29:17 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
21:29:17 MQT: Tentativo di connessione...
21:29:18 MQT: Connesso
21:29:18 MQT: tele/sonoff/LWT = Online (salvato)
21:29:18 MQT: cmnd/sonoff/POWER = 
21:29:18 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
21:29:19 MQT: Tentativo di connessione...
21:29:19 MQT: Connesso
21:29:19 MQT: tele/sonoff/LWT = Online (salvato)
21:29:19 MQT: cmnd/sonoff/POWER = 
21:29:19 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
21:29:20 MQT: Tentativo di connessione...
21:29:20 MQT: Connesso
21:29:20 MQT: tele/sonoff/LWT = Online (salvato)
21:29:20 MQT: cmnd/sonoff/POWER = 
21:29:21 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
```
And this thing in Hassbian's prompt:
```
pi@hassbian:~ $ mosquitto_sub -v -t 'cmnd/sonoff/Power1'
cmnd/sonoff/Power1 ON

```
Seems like the prompt keeps waiting for a command. I always have to press ctrl+c to exit from this situation.

aferraz, 2018-02-12:
@leobel96 mosquito_sub stay running and listen on the same Topic of Tasmota.
In this case you can see what came from HASS to Tasmota and understand what is wrong.
stat/sonoff/RESULT = {"Command":"Unknown"} means that unknown command arrives, since happening only when you enable MQTT on HASS means that the problem come from HASS.

leobel96, 2018-02-12:
Seems like there is a conflict. Is it correct to set in configuration.yaml for the broker the same ip of HA because Mosquitto runs on the same device?

aferraz, 2018-02-12:
This could be the problem, HASS should have an embedded mqtt broker. You talk about this broker or you have installed mosquitto also ?

leobel96, 2018-02-12:
I've have installed mosquitto also. I don't like the embedded broker....

leobel96, 2018-02-12:
I have tried changing module from Tasmota's configuration and now I can change the Power state!! Anyway it works only if I comment the state_topic in configuration.yaml. If I decomment it the switch in the HA frontend switches on and off immediately and sonoff's state doesn't change...

aferraz, 2018-02-13:
@leobel96 what module are you using, on a hardware level?
I use  state_topic: "cmnd/sonoff/Power1" but I have an Electrodragon with 2 relay.
In your case could be state_topic: "cmnd/sonoff/Power"

leobel96, 2018-02-13:
I'm using a simple Nodemcu with nothing connected. Before I set it as "Generic" and now it is set as "Sonoff Basic". It's strange because the switch without "state_topic" works but the device keep connecting and unconnecting as you can see here:

```
1518519271: Received PUBLISH from sonoff1 (d0, q0, r0, m0, 'cmnd/sonoff/POWER', ... (0 bytes))
1518519271: Received SUBSCRIBE from sonoff1
1518519271:     cmnd/sonoff/# (QoS 0)
1518519271: sonoff1 0 cmnd/sonoff/#
1518519271: Sending SUBACK to sonoff1
1518519271: Sending PUBLISH to sonoff1 (d0, q0, r1, m0, 'cmnd/sonoff/POWER', ... (2 bytes))
1518519271: Received SUBSCRIBE from sonoff1
1518519271:     cmnd/sonoffs/# (QoS 0)
1518519271: sonoff1 0 cmnd/sonoffs/#
1518519271: Sending SUBACK to sonoff1
1518519271: Received SUBSCRIBE from sonoff1
1518519271:     cmnd/sonoff1/# (QoS 0)
1518519271: sonoff1 0 cmnd/sonoff1/#
1518519271: Sending SUBACK to sonoff1
1518519271: Received PUBLISH from sonoff1 (d0, q0, r0, m0, 'stat/sonoff/RESULT', ... (14 bytes))
1518519271: Received PUBLISH from sonoff1 (d0, q0, r0, m0, 'stat/sonoff/POWER', ... (2 bytes))
1518519271: Socket error on client <unknown>, disconnecting.
1518519271: New connection from 192.168.1.72 on port 1883.
1518519271: Client sonoff1 already connected, closing old connection.
1518519271: Client sonoff1 disconnected.
1518519271: New client connected from 192.168.1.72 as sonoff1 (c1, k60, u'user').
1518519271: Sending CONNACK to sonoff1 (0, 0)
1518519271: Received SUBSCRIBE from sonoff1
1518519271:     tele/sonoff/LWT (QoS 1)
1518519271: sonoff1 1 tele/sonoff/LWT
1518519271: Sending SUBACK to sonoff1
1518519271: Sending PUBLISH to sonoff1 (d0, q0, r1, m0, 'tele/sonoff/LWT', ... (6 bytes))
1518519271: Received PUBLISH from sonoff1 (d0, q1, r1, m122, 'tele/sonoff/LWT', ... (6 bytes))
1518519271: Sending PUBACK to sonoff1 (Mid: 122)
1518519271: Sending PUBLISH to sonoff1 (d0, q1, r0, m1, 'tele/sonoff/LWT', ... (6 bytes))
1518519272: Received PUBACK from sonoff1 (Mid: 1)
1518519272: New connection from 192.168.1.77 on port 1883.
1518519272: Client sonoff1 already connected, closing old connection.
1518519272: Client sonoff1 disconnected.
1518519272: New client connected from 192.168.1.77 as sonoff1 (c1, k15, u'user').
1518519272: Sending CONNACK to sonoff1 (0, 0)
1518519272: New connection from 192.168.1.72 on port 1883.
1518519272: Received PUBLISH from sonoff1 (d0, q0, r1, m0, 'tele/sonoff/LWT', ... (6 bytes))
1518519273: Received PUBLISH from sonoff1 (d0, q0, r0, m0, 'cmnd/sonoff/POWER', ... (0 bytes))
1518519273: Received SUBSCRIBE from sonoff1
1518519273:     cmnd/sonoff/# (QoS 0)
1518519273: sonoff1 0 cmnd/sonoff/#
1518519273: Sending SUBACK to sonoff1
1518519273: Sending PUBLISH to sonoff1 (d0, q0, r1, m0, 'cmnd/sonoff/POWER', ... (2 bytes))
1518519273: Received SUBSCRIBE from sonoff1
1518519273:     cmnd/sonoffs/# (QoS 0)
1518519273: sonoff1 0 cmnd/sonoffs/#
1518519273: Sending SUBACK to sonoff1
1518519273: Received SUBSCRIBE from sonoff1
1518519273:     cmnd/sonoff1/# (QoS 0)
1518519273: sonoff1 0 cmnd/sonoff1/#
1518519273: Sending SUBACK to sonoff1
1518519273: Received PUBLISH from sonoff1 (d0, q0, r0, m0, 'stat/sonoff/RESULT', ... (14 bytes))
1518519273: Received PUBLISH from sonoff1 (d0, q0, r0, m0, 'stat/sonoff/POWER', ... (2 bytes))

```
Probably the problem with state_topic is generated by these continuous disconnections but cmnd doesn't care... As you can see mosquitto keeps receiving 'stat/sonoff/POWER' so it is right...

aferraz, 2018-02-13:
@leobel96 I encounter the same problem of continuous disconnections from broker in the last version.
@arendst I have the same problem reported by leobel96, only on nodemcu and electrodragon, sonoffs work fine. I also noticed that they often do not respond to pinging.
Thanks
 Alessandro

leobel96, 2018-02-13:
I also noted that, with "state_topic" option, If I press the ON/OFF button in Tasmota's homepage (so I try to turn it OFF without using MQTT), it keeps turning ON also if I want it to be OFF. The opposite thing that occurs with HA's button. Very strange.

leobel96, 2018-02-13:
Thi is what happens in HA:
[image]
And in Tasmota:
[image]

aferraz, 2018-02-13:
@leobel96 this is normal beacause you have, Hass side, problems with the state_topic so Hass don't know the relay status

leobel96, 2018-02-13:
@aferraz is it normal also at Tasmota's side? Shouldn't Tasmota know its own state?

Jason2866, 2018-02-13:
Hello,
if your Devices are "lazzy" means react slowly and have restarts. Compile V.5.12 in english.
The German V 5.12. has problems. With 5.12. (EN) everything works as it should.
Maybe the italianVersion has the same problems as the German....

leobel96, 2018-02-13:
@Jason2866 I have flashed the EN firmware but nothing changed

johoberm, 2018-02-13:
Just an idea.... Sorry i cant help you here. I use OpenHab as HA

aferraz, 2018-02-13:
You have flashed the standard .bin files or you have compiled firmware by yourself ?
You are using Nodemcu, this is the first firmware release flashed ?
I ask you these questions because in the past I had problems flashing Nodemcu with firmware compiled for sonoff. 
I always flash Nodemcu, for the first time, with a compiled version for Nodemcu and then upgrade via web interface with the standard sonoff version

leobel96, 2018-02-13:
I have flashed firmware bu by myself. I also tried disabling ws2812 and home-assistant commenting them in before flashing but nothing changed. It's first time I use Tasmota and the nodemcu is the first device I flash with this firmware :(

Jason2866, 2018-02-13:
I compile with my settings in user_config.h.
I can upload with no problems via Atom IDE (you have to remove the ; by Upload method!)
The software runs with no problems on the NodeMCU.
After that i flash my working devices....

leobel96, 2018-02-13:
@jason2866
Are you using Home Assistant?

Jason2866, 2018-02-13:
@leobel96 No, i use OpenHab

leobel96, 2018-02-13:
@Jason2866 ahhhhh. Sorry, I didn't understood.
I have tried to use the NodeMCU with Tasmota installed with another MQTT service (cloudMQTT) without Home-Assistant and it works fine so it isn't an upload or settings problem...

leobel96, 2018-02-13:
@Jason2866 @aferraz I RESOLVED THE PROBLEM:
The problem was that I set the same client_id for HA and Tasmota and this was causing a conflict:
```
mqtt:
  broker: 192.168.1.72
  port: 1883
  client_id: "sonoff1"  <-------- THIS ONE
  keepalive: 60
  username: !secret mqtt_user
  password: !secret mqtt_password
  protocol: 3.1
  birth_message:
    topic: "tele/sonoff/LWT"
    payload: "Online"
    qos: 1
    retain: true
  will_message:
    topic: "tele/sonoff/LWT"
    payload: "Offline"
    qos: 1
    retain: true
    
switch:
  - platform: mqtt
    name: "Sonoff power"
    state_topic: "stat/sonoff/POWER"
    command_topic: "cmnd/sonoff/POWER"
    availability_topic: "tele/sonoff/LWT"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    retain: false
```
[image]

I thought that in configuration.yaml I have to use the same name of the device I'll connect to it but, instead, it's the name used by Home Assistant to connect to Mosquitto's broker on the same machine. So using different client names is MANDATORY.
Thank you to all!!

closed 2018-02-13 by leobel96

bachoo786, 2018-02-14:
Ha doesn't report state when it restarts I get mine as off but the sonoff is on. Anyone has the same issue?

leobel96, 2018-02-14:
@bachoo786 probably you have set retain option to false?

bachoo786, 2018-02-14:
You mean retain in the sonoff switch ?

leobel96, 2018-02-14:
@bachoo786 in switch component and in mqtt component in configuration.yaml

bachoo786, 2018-02-14:
@leobel96 so my switch component for sonoff I have the retain as true. As for the mqtt settings u am using mosquitto, where do I set the retain option in the mqtt component?

leobel96, 2018-02-14:
@bachoo786 in birth_message and will_message. Look at my configuration many messages before

bachoo786, 2018-02-14:
@leobel96 ok let me try that

bachoo786, 2018-02-14:
@leobel96 will the birth and will message affect my other mqtt components?

leobel96, 2018-02-14:
@bachoo786 I don't know. Anyway, if you see problems with other components, you can disable it again and everything will be as before

bachoo786, 2018-02-14:
Ok thanks

leobel96, 2018-02-14:
@bachoo786 I noted now that if I switch off HA, then I change the Tasmota's state and then I switch on HA, Tasmota's state changes according to HA's switch one set before switching it off. If I try setting retain option to False, and I do the same passages, this thing doesn't happen. If you want the HA's switch state to be set like the Tasmota's one after reconnection, I don't know if it is possible. Anyway I suggest you to open a new Issue for it to have a better visibility...

bachoo786, 2018-02-14:
Ok I will try that when I am home

bachoo786, 2018-02-14:
@leobel96 I resolved the power issue by going into ‘Console’ in the Sonoff’s web UI, typing ‘PowerRetain 1’ and enter.

this remembers the state and even changes after restarting HA.

leobel96, 2018-02-14:
@bachoo786 Good!