arendst/Tasmota #1863: Sonoff 4ch killed?

kbaras, 2018-02-11:
Hi,

I have flashed tasmota to sonoff 4ch pro. At first it worked without any issues when testing(before took it to its place to switch the lights), and also after I connected the lights to it. But after I connected the second wall switch to the second GPIO (that was configured for switching) it stopped working. With the first switch it worked.
I see the first three leds are red, and it does not respond to any input. I don't see it on the LAN either.
I tried serial monitor with ftdi but no output when power-on. When I "push" gpio0 during starting (flash mode) only one charcter "�" is seen on the serial monitor.

Did I kill the wifi chip? If so, why could it happen? The cables of the wall switch weren't under electricity.

Any help would be appreciated!

Jason2866, 2018-02-11:
Hi,
already flashed again? 
If this doesnt help use esptool.
Erase!! first the flash memory and then flash it with esptool with option DOUT
Erasing should last 2 - 5  seconds. Flashing should be ready after 60 - 90 secpnds

kbaras, 2018-02-11:
Yes, earlier I had problem with flashing, so the process is already known to me. But the thing is, PC does cannot communicate with the sonoff now :(

I have a suspicion: when I connected the wall switch it it possible that some stray current was on the cable and maybe killed the wifi module.
Could you tell me a way how can I see if esp8285 is still functional? If it is dead, can I recplace it with another esp chip, or throw the whole device into the trash?

Jason2866, 2018-02-11:
If no communication is possible i think the esp is dead. 
All the devices i have (sonoff basic s20 and pow)
are the rx and tx direct connected to the esp chip.
Maybe this is different at the 4ch switch. Just have a look in the schematic.

Jason2866, 2018-02-11:
I took a look in the schematic. Bad news the RX and TX goes direct to the ESP.
So if it doesnt react. The chip is dead. One chance perhaps a other chip died a needs to much power on the 3.3 Volt Line. Just connect it to your flasher. Now measure the Voltage. If it is much lower than 3.3 Volts than maybe not the ESP is dead it could be a other IC or all of them... In this case its time to say goodbye

kbaras, 2018-02-11:
Ok, I mearured the voltages. They are correct (under 3,3V on all pin). So, it seems the chip went to heaven :(

I am quite new to electronics, therefore this question might be silly: would it worth to order another PSF chip from itead, remove the bad one, and solder the new? Is it worth a try?

Jason2866, 2018-02-11:
Theoreticall yes. But desoldering and resolder SMD Parts is really not easy without damaging the ICs or the PCB. I have some years experince.... But i dont do such things. I just have not the right tools for that nor the skills....