arendst/Tasmota #1864: Problem with sonoff Basic to Flash

surftier, 2018-02-11:
Hello,
i get an Sonoff basic th v1.1 2017-5-5 and a sonoff pow. The pow was easy to flash with arduino IDE,
but the basic will not work. If I flash it with the arduino IDE the progress bar went from 0 to 100%. After reboot no LED blink or something else. 
Then I Use the ESP 8266 Downloadtool from espressif with build .bin from arduino IDE. This work... but i can't change anything via Konsole or WebGUI or Serial.
After several tries with a sonoff.ino.bin i get this on  serial connection.
`
00:00:00 CFG: Standard verwenden
00:00:00 Projekt sonoff Sonoff (Topic basic01, Fallback DVES_067421, GroupTopic sonoffs) Version 5.12.0a

Exception (28):
epc1=0x4010442e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000483 depc=0x00000000

ctx: cont 
sp: 3fff3640 end: 3fff3950 offset: 01a0

>>>stack>>>
3fff37e0:  00000022 40104639 00040000 00000000  
3fff37f0:  3ffee730 40103616 00000000 fffffffe  
3fff3800:  ffffffff 3fffc6fc 00000000 4000050c  
3fff3810:  3fffc278 4010346c 3fffc200 00000022  
3fff3820:  3fff3830 3fffc6fc 00000001 3fff1bc0  
3fff3830:  40217d3c 00000030 00000017 ffffffff  
3fff3840:  40217d3c 00000000 00000000 00000000  
3fff3850:  0000000d 0000023d 00000000 fffffffe  
3fff3860:  ffffffff 3fffc6fc 00000058 3fff20b0  
3fff3870:  00000000 3fffdad0 3fff2924 00000030  
3fff3880:  00000000 3fffdad0 3fff2924 00000030  
3fff3890:  3ffef7b0 3fff4b4c 3fff03fc 401004d8  
3fff38a0:  00000001 00003efb 4021870c 3fff03fc  
3fff38b0:  00000001 00000001 00000000 40107851  
3fff38c0:  40207984 3fff2050 00000000 40208349  
3fff38d0:  3fff201c 3fff2050 3fff2001 40222120  
3fff38e0:  3fff20b0 00000000 3fff1bc0 402184b2  
3fff38f0:  3fff20b0 3fff23b0 3fff2930 00000000  
3fff3900:  3fffdad0 3fff2924 40221a5c 3fff2930  
3fff3910:  00000000 3fff1ba4 3fff291d 3fff2924  
3fff3920:  3fffdad0 00000000 3fff20b0 40217d3c  
3fff3930:  3fffdad0 00000000 3fff291d 40221aa8  
3fff3940:  feefeffe feefeffe 3fff2930 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

00:00:00 CFG: Standard verwenden
00:00:00 Projekt sonoff Sonoff (Topic basic01, Fallback DVES_067421, GroupTopic sonoffs) Version 5.12.0a
00:00:00 WIF: verbinden mit AP1 WLAN90a in Modus 11N wie basic01-5153...

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

Exception (9):
epc1=0x40254e37 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40254e2e depc=0x00000000
`
I don´t know what went wrong.
Is the ESP8266 Chip broken
Sorry for my bad English

Andre

Jason2866, 2018-02-11:
Hi,
did you flash with Option DOUT?
I had one NodeMCU which was "dead" i managed to recover it ONLY with esptool
[ESPTool_WriteFlash.txt](https://github.com/arendst/Sonoff-Tasmota/files/1714568/ESPTool_WriteFlash.txt)

surftier, 2018-02-17:
Hi, 
the Problem is solved.
It was caused of my self. I used the sonoff Basic with the Voltage out of the Programadapter to check it run. As I plugged the Basic to Mains the Basic work fine...

closed 2018-02-17 by surftier