arendst/Tasmota #1865: Sonoff & BME280

echo4019, 2018-02-11:
Hi,

Can anyone describe the process of connecting a BME280 sensor to a sonoff basic? 

Any help is appreciated.

curzon01, 2018-02-11:
1. based on [Sonoff-Basic GPIO Locations](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Basic#gpio-locations) connect 
```
BME280-3.3V -> Sonoff-3.3V
BME280-GND -> Sonoff-GND
BME280-SCL -> Sonoff-TX
BME280-SDA -> Sonoff-RX
```

2. On Tasmota web interface go to "Configration->Configure Module" and set
```
GPIO1 Serial Out -> I2C SCL
GPIO3 Serial In -> I2C SDA
```
Restart

echo4019, 2018-02-11:
When checking the Sonoff-Basic GPIO Locations it mentions to access GPIO 04 you must solder wire to pin on ESP chip. Is this referring to older versions of the sonoff basic?

Also on the tasmota web interface under configure module I have no option on the drop down list for I2C SCL or I2C SDA. 

Thanks

echo4019, 2018-02-11:
My bad I had disabled #define USE_I2C on the config file.

echo4019, 2018-02-11:
- platform: mqtt
    name: "BME280 Temperature"
    state_topic: "tele/bme280/SENSOR"
    unit_of_measurement: "°C"
    value_template: "{{ value_json.Temperature | round(1) }}"

Is this ok for the config.yaml file for home assistant?

curzon01, 2018-02-11:
In the configuration above I don't use GPIO4

jumblies, 2018-05-23:
Were you guys able to do this on a sonoff basic?  I'm struggling with this one.  Did you use a pullup on VCC and Data?

curzon01, 2018-05-23:
Comment above contains description for the 4 wires you need, link inside contains pictures of the sonoff basic. If you use a BME which comes soldered on one of common breakout boards you don't need any pulls, only connect the 4 wires and configure the Tasmota as described.

glamke, 2018-05-25:
Solved.  It was a circuitous route but it worked perfect once I managed to upgrade to the latest binary.  I had a very difficult time getting flashes done properly with the binaries.  Flashes worked, but then units wouldn't respond.  Another issue for another thread.  

Thank you for your help

curzon01, 2018-05-25:
So can you pls close your issue?

jumblies, 2018-05-26:
This closes issue #1865 for me, but I didn't open it

closed 2018-06-02 by ascillato