arendst/Tasmota #1866: Wrong flashing instructions for Windows 10

Cysign, 2018-02-11:
Just downloaded Tasmota to falsh a S20 socket - but https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE#copy-files tells me about a folder that doesn't exist:

portable\packages\esp8266\hardware\esp8266\2.3.0

Couldn't find the mentioned platform.txt on my Windows 10 machine. Where might it be located?

Jason2866, 2018-02-11:
Arduino IDE needs manual copying files.... I dont like that.
Just install Atom, and you have nothing to do manually.
Download the sonoff files extract the whole folder to a place of your choice and
in Atom go to Open Project folder. Point here to your tasmota folder. READY!

Cysign, 2018-02-11:
In general I'm fine with the ArduinoIDE. But I wonder if I have to create the folder 'portable' or if it should exist - maybe the author of the tutorial has been using another OS?

arendst, 2018-02-11:
The author was using windows and urges to get separate environments for normal arduino and esp8266 hardware. This allows for easy dedicated libraries for both esp/arduino versions 2.3.0 and 2.4.0 as well as for tasmota supported libraries.

Following the wiki you could have learned to create the portable folder first BEFORE executing arduino.exe as this program will then initialise this folder with correct file structure. Try it, it's fun.

Cysign, 2018-02-11:
Well...stupid me. That's it ;)
Thanks!

closed 2018-02-11 by Cysign