arendst/Tasmota #1867: MagicHome LED (ESP-IR-B-v2.3 / ESP8285MOD ESP-M2) not working with tasmota f/w

oxcxsxixbxaxn, 2018-02-11:
Hi I bought 2 of these devices cheap as chips off ebay.  V2.3 with no RX/TX pads and no IR connected.
I flashed one of them with the latest tasmota build using esptool (via raspi with serialUSB), making backup, erasing flash and uploading the sonoff.bin with no problems.  After completing the flash, powered off and disconnected (GND last as per wiki) and powered up again but initially saw no AP listening for initial setup.  Tried a few more times and eventually noticed an AP called ESP-<MACADR>, but I could not connect to it and eventually it would disappear, sometimes coming back again some time later, but still not connectable.
What is the expected behaviour?  These things have no button, so should the start in AP mode everytime until configured? Do I have to do something to trigger the AP mode?  Do I have to do something special to connect to it?

The second device I tried flashing with a slightly older tasmota f/w that I had used on my sonoff basic with success, but on the magichome, the same flaky behaviour as the first.  Suspecting dodgy hardware, I flashed the second device back to it's stock firmware, downloaded the magichome app and used that to set it up.  The device worked as expected again, magichome app found it as an AP, and connected it to my home wifi without issues, (other than I don't want to use magichome f/w!!)

So is this new unsupported hardware?  did I do something wrong in setup?  
Is there a way to debug this?  I tried pointing putty at the serial port connected to the device using 9600 and 115200, but both only showed garbage.

[image]

Jason2866, 2018-02-11:
Hi,
i have the same modul (it looks exactly as yours)
My first flash was not successful. I didnt connect to my Wlan.
After resolder the connections (i was sure it works...) the second attempt works as it should.
I have all my settings in user_config.h. No need to do anything on the module after power up.
Just a look in the router, what ip it has. Just launch browser and config the modul -> Ready and working
Hint: setOption15 1

oxcxsxixbxaxn, 2018-02-11:
Hi,
When you say you first flash was not successful, you mean the flash operation itself?  or the behaviour after a seemingly successful flash?

Thinking about it, one thing I did different is I forgot to disconnect the gpio-0 from GND through out the backup, erase and flash, only disconnecting it after "successful" flash completed and I disconnected everything...  would this make a difference?

This is how the flash output looked, seems fine??

# esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -fs 1MB -fm dout 0x0 sonoff.bin
esptool.py v2.3-dev
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 495808 bytes to 341889...
Wrote 495808 bytes (341889 compressed) at 0x00000000 in 31.8 seconds (effective 124.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

Jason2866, 2018-02-11:
Flash not successfull in my case ment everything looks good like your flash.
But after power it up it doesnt work. Second attempt was working.
The only difference between 1st and 2nd flash. At try 2 i did a erase with esptool.

oxcxsxixbxaxn, 2018-02-11:
I erased each time, only difference was leaving the gpio-0 connected to GND
so... 
connected GND, tx, rx, gpio-0 to GND and powered up,
Backed up with esptool and disconnected power, waited a few seconds
Reconnected power and Erased flash then disconnected power
Reconnected power and flashed f/w, when complete removed power, removed tx/rx/gpio-0 then GND

I'll try again with custom firmware with my wifi setting already set and see if that gets me further

Thanks

app41, 2018-02-11:
same model, arrived a few days ago from aliexpress.
it mounts the ESP-M2, a module with ESP8285
here the link for reference: [https://it.aliexpress.com/item/Magic-Home-Mini-RGB-RGBW-Wifi-Controller-For-Led-Strip-Panel-light-Timing-Function-16million-colors/32686853650.html?spm=a2g0s.9042311.0.0.O5enRa](url)

by the way, the label on the case was put upside down so the +VCC pin on led connector is reported in the opposite side

before starting to solder & wire it, I tested with myhome app and I have seen all the colors...

it is the model without IR receiver but with also W pin, even if I don't have an RGBW strip. 
the cost was quite the same..

I glued a connector to be able to make experiments without wiring problems
be careful with wire soldering, it is a tough small space

[image]

[image]

I didn't have any problem in flashing and tasmoting it
I made these configurations:
```
19:50:38 MQT: letto4/magicHome/INFO1 = {"Module":"MagicHome","Version":"5.10.0g","FallbackTopic":"DVES_96A3DE","GroupTopic":"sonoffs"}
19:50:38 MQT: letto4/magicHome/INFO2 = {"WebServerMode":"Admin","Hostname":"magicHome-0990","IPAddress":"192.168.10.205"}
19:50:38 MQT: letto4/magicHome/INFO3 = {"RestartReason":"Software/System restart"}
19:50:39 MQT: letto4/magicHome/RESULT = {"POWER":"OFF"}
19:50:39 MQT: letto4/magicHome/POWER = OFF
19:52:29 CMD: setoption15
19:52:29 MQT: letto4/magicHome/RESULT = {"SetOption15":"ON"}
19:52:32 CMD: color 1
19:52:32 MQT: letto4/magicHome/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF0000"}
19:52:35 CMD: color 2
19:52:35 MQT: letto4/magicHome/RESULT = {"POWER":"ON","Dimmer":100,"Color":"00FF00"}
19:52:37 CMD: color 3
19:52:37 MQT: letto4/magicHome/RESULT = {"POWER":"ON","Dimmer":100,"Color":"0000FF"}
```
but after boot and configuration, there is only the RED light switched on
it seems the other two colors are dead

I tried also to map where mosfets are mapped on GPIO on my board, to check on ESP pinout if it is a hardware/driver problem or a pin configuration problem, even if I don't have now an oscilloscope in my hands

Jason2866 and oxcxsxixbxaxn, are you able to drive all the 3 colors?
any suggestion?
maybe I forgot to configure any parameter?

can anybody suggest me the pinout from ESP for RGBW in magichome standard configuration?

Jason2866, 2018-02-11:
Hi app41,
i see no error. It should work. I will connect mine. At the moment it is not in use.
Just put it together for fun and testing...

oxcxsxixbxaxn, 2018-02-11:
I have still to flash tasmota, but when I plugged my rgbw strip into device running magichome firmware, all 3 colours and WW lit correctly.  Funnily enough I only checked the V+ was in the correct side and didn't look at the order printed on the strip...  that actually shows +GRBW...
[image]

Jason2866, 2018-02-11:
This is my setup with correct colors (Modul 37, SetOption15 1 and nothing else)

00:00:00 Project sonoff LED (Topic sonoff-959C17, Fallback sonoff-959C17, GroupTopic sonoffs) Version 5.11.1a
00:00:00 WIF: Connecting to AP1 Jason_Home_WLAN in mode 11N as sonoff-959C17-7191...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff-959C17-7191.local with IP address 192.168.2.135
00:00:04 UPP: Multicast (re)joined
00:00:05 UPP: Multicast (re)joined
20:43:59 MQT: Attempting connection...
20:43:59 MQT: Connected
20:43:59 MQT: tele/sonoff-959C17/LWT = Online (retained)
20:43:59 MQT: cmnd/sonoff-959C17/POWER = 
20:43:59 MQT: tele/sonoff-959C17/INFO1 = {"Module":"Arilux LC01","Version":"5.11.1a","FallbackTopic":"sonoff-959C17","GroupTopic":"sonoffs"}
20:43:59 MQT: tele/sonoff-959C17/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-959C17-7191","IPAddress":"192.168.2.135"}
20:43:59 MQT: tele/sonoff-959C17/INFO3 = {"RestartReason":"External System"}
20:43:59 UPP: Multicast (re)joined
20:44:00 MQT: stat/sonoff-959C17/RESULT = {"POWER":"OFF"}
20:44:00 MQT: stat/sonoff-959C17/POWER = OFF
20:44:07 MQT: tele/sonoff-959C17/STATE = {"Time":"2018-02-11T20:44:07","Uptime":0,"Vcc":3.268,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":80,"APMac":"00:A0:57:2A:BD:19"}}
20:47:09 CMD: State
20:47:09 MQT: stat/sonoff-959C17/RESULT = {"Command":"Unknown"}
20:47:19 CMD: Status
20:47:19 MQT: stat/sonoff-959C17/STATUS = {"Status":{"Module":37,"FriendlyName":"LED","Topic":"sonoff-959C17","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
20:47:40 CMD: Gpio
20:47:41 MQT: stat/sonoff-959C17/RESULT = {"GPIO14":"0 (None)"}
20:48:11 CMD: Modules
20:48:11 MQT: stat/sonoff-959C17/RESULT = {"Modules1":["1 (Sonoff Basic)","2 (Sonoff RF)","3 (Sonoff SV)","4 (Sonoff TH)","5 (Sonoff Dual)","6 (Sonoff Pow)","7 (Sonoff 4CH)","8 (S20 Socket)","9 (Slampher)","10 (Sonoff Touch)","11 (Sonoff LED)","12 (1 Channel)","13 (4 Channel)","14 (Motor C/AC)","15 (ElectroDragon)","16 (EXS Relay)","17 (WiOn)"]}
20:48:11 MQT: stat/sonoff-959C17/RESULT = {"Modules2":["18 (WeMos D1 mini)","19 (Sonoff Dev)","20 (H801)","21 (Sonoff SC)","22 (Sonoff BN-SZ)","23 (Sonoff 4CH Pro)","24 (Huafan SS)","25 (Sonoff Bridge)","26 (Sonoff B1)","27 (AiLight)","28 (Sonoff T1 1CH)","29 (Sonoff T1 2CH)","30 (Sonoff T1 3CH)","31 (Supla Espablo)","32 (Witty Cloud)","33 (Yunshan Relay)"]}
20:48:11 MQT: stat/sonoff-959C17/RESULT = {"Modules3":["34 (MagicHome)","35 (Luani HVIO)","36 (KMC 70011)","37 (Arilux LC01)","38 (Arilux LC11)","39 (Sonoff Dual R2)","40 (Arilux LC06)"]}
20:48:27 CMD: Setoption
20:48:27 MQT: stat/sonoff-959C17/RESULT = {"SetOption1":"OFF"}
20:48:57 CMD: Setoption15
20:48:57 MQT: stat/sonoff-959C17/RESULT = {"SetOption15":"ON"}
20:49:11 MQT: tele/sonoff-959C17/STATE = {"Time":"2018-02-11T20:49:11","Uptime":0,"Vcc":3.265,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":78,"APMac":"00:A0:57:2A:BD:19"}}
20:49:22 CMD: Color 1
20:49:22 MQT: stat/sonoff-959C17/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF0000"}
20:49:30 CMD: Color 2
20:49:30 MQT: stat/sonoff-959C17/RESULT = {"POWER":"ON","Dimmer":100,"Color":"00FF00"}
20:49:37 CMD: Color 3
20:49:37 MQT: stat/sonoff-959C17/RESULT = {"POWER":"ON","Dimmer":100,"Color":"0000FF"}

Jason2866, 2018-02-11:
I found the difference :-)  i use Modul 37 and you use Modul 34
Change this and yours will work too

app41, 2018-02-11:
aha, good hintm Jason2866
I tried to change to module 37 and I HAVE NOW 2 COLORS, still green missing
maybe a problem with pin mapping?
in my hardware, Color 1 is red, color 3 is blue, and so color 2 should be green

this hardware is made by some small companies in China, copying and recopying in a bad way the original idea. who can imagine how they mapped the 4 pins to drive the LEDs....? 
although my picture seems to be very similar, my hardware seems to be the same from oxcxsxixbxaxn 

lets make some experiment

Jason2866, 2018-02-11:
Take a look in sonoff_template.h  there is the definition for the GPIOs for the modules
Good starting point is the Arilux 01... Change your setup to module generic (or WeMos) 
and copy the Arilux Setup. Try to modify from this starting point. I am sure you will find a working setup

oxcxsxixbxaxn, 2018-02-11:
flashed a custom image and straight up and running with tasmota, except the colors are all wrong now :)
will look at this template too.  Arilux 01 only seems to use RRGGBB and no Warm White control at all.

Jason2866, 2018-02-11:
I have no RGB Strip with White Led so this line is not used in my setup....

app41, 2018-02-11:
by test on some compiling variation on my hardware I found
// GPIO05 - GPIO_PWM1 -- RGB LED COLOR 1 - RED
// GPIO13 - GPIO_PWM2 -- RGB LED COLOR 2 - BLUE
// GPIO15 - WHITE --- RGB LED WHITE

despite any effort I was not able to move the missing color, maybe something damaged on my hardware
I have now only this sample, maybe on Amazon...

jason and oxcxsxixbxaxn, can you please update me which GPIO you have mapped the PWM3?

I don't know how to manage the WHITE pin, I didn't found any in the standard command file

I think we have to pull a request to introduce in the standard configuration this new hardware MagicHome LED v2.3 (ESP-IR-B-v2.3 / ESP8285MOD ESP-M2) it is different from the MagicHome first generation

oxcxsxixbxaxn, 2018-02-12:
Well I worked mine out.  Color order was just a case of switching the PWM1 and PWM2 because my LED strip is +GRBW.  Then set some of the unused GPIO's to GPIO_USER and then tried them as PWM4 on the module configuration screen.
Actually the first one I tried was correct, I found the White PWM on GPIO15.
So here's what worked for my MagicHome LED v2.3 (ESP-IR-B-v2.3 / ESP8285MOD ESP-M2) 
 ```
 { "Arilux LC01",     // Arilux AL-LC01 (ESP8285) - MagicHome LED v2.3 (ESP-IR-B-v2.3 / ESP8285MOD ESP-M2)
                       //  (PwmFrequency 1111Hz)
     GPIO_KEY1,        // GPIO00 Optional Button
     0,
     GPIO_LED2_INV,    // GPIO02 RF receiver control
     0,
     GPIO_ARIRFRCV,    // GPIO04 IR or RF receiver (optional)
     GPIO_PWM2,        // GPIO05 RGB LED Red
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_PWM1,        // GPIO12 RGB LED Green
     GPIO_PWM3,        // GPIO13 RGB LED Blue
     GPIO_USER,       // GPIO14 ??
     GPIO_PWM4,       // GPIO15 RGBW LED White (optional - set to PWM4 for Cold White or Warm White)
     0, 0
  },
```
@app41

PC-Eggsberde, 2018-02-16:
Hi @app41 

Du hast die Rote Markierung in deinem Bild nicht auf V+ gesetzt sondern auf den White Pin

[image]

Gohico, 2018-02-17:
@oxcxsxixbxaxn 
I can confirm your config... In the beginning I thought my MagicHome has a hardware problem as the PWM was not running... But after applying the GPIO correction it looks good...
Awesome, thank you! :)

lance36, 2018-02-17:
Maybe is somewhat related to the issue i solved over here?
https://github.com/arendst/Sonoff-Tasmota/issues/1810

i have the RGB version tho

app41, 2018-02-17:
@PC-Eggsberde 
yes I took the picture before discovering that the common pin was wrong on the label. so red label is wrong. can you believe? the chinese manufacturer put the label on the wrong side of the plastic case, so it was impossible having led running

TKroenert, 2018-02-18:
After troubleshooting the following problems I believe Tasmota has some basically design issues with handling LED-stripes. Hopefully someone more into Tasmota could learn from my failures:
- @oxcxsxixbxaxn's configuration is for an "12V G B R W"- ledstripe, but I have a "12V B R G" - ledstripe. It fits into the controller, but the color mapping does not match.
- I have the same hardware-controller (MagicHome v2.3) as @oxcxsxixbxaxn, but using @oxcxsxixbxaxn's configuration for RGBW makes it necessary to push 4x values (RRGGBBWW) to the node. It does not response to RRGGBB values. (keep in mind: using an RGB-ledstripe on a RGBW-controller is totally legit)
- with the original firmware I was able to change the colors with the remote. But with Tasmota I only get strange looking IR Recv bytes over MQTT.

Nevertheless, for the basic functionality the following configuration works like a charm:
```

  { "MagicHome",     // Arilux AL-LC01 (ESP8285) - MagicHome LED v2.3 (ESP-IR-B-v2.3 / ESP8285MOD ESP-M2)
                        //  (PwmFrequency 1111Hz)
      GPIO_KEY1,        // GPIO00 Optional Button
      0,
      GPIO_LED2_INV,    // GPIO02 RF receiver control
      0,
      GPIO_IRRECV,    // GPIO04 IR or RF receiver (optional)
      GPIO_PWM1,        // GPIO12 RGB LED Green
      0, 0, 0, 0, 0, 0, // Flash connection
      GPIO_PWM3,        // GPIO13 RGB LED Blue
      GPIO_PWM2,        // GPIO05 RGB LED Red
      GPIO_USER,       // GPIO14 ??
      GPIO_USER,       // GPIO15 RGBW LED White (optional - set to PWM4 for Cold White or Warm White)
      0, 0
  },
```

oxcxsxixbxaxn, 2018-02-19:
@TKroenert IRRECV doesn't do anything more than send the IR button code to MQTT.  This is good and bad news.
The bad news is supplied remote doesn't do anything to the colours locally without additional handling set up on node-red or hass script or similar MQTT.
The good news is that the device will now respond to ANY IR remote... got an unused button on your tv remote? Use that to trigger any action in MQTT.

Actually my strip is +GRBW which should be opposite to your +BRG, I would have expected you to have PWM3 on GPIO5, PWM1 on GPIO12 and PWM2 on GPIO13 for the 2.3 version to light your colours correctly.  Your config would be light RBG on my setup.  I would suspect that it is the strips (yours or mine) which are incorrectly labelled...

If you include PWM4, the device expects RRGGBBWW, if PWM5 is included it expects RRGGBBCCWW.
So if you want RGB only, only use PWM1(red), PWM2(green) and PWM3(blue) on the GPIOs according to your strip.

bbanfai, 2018-02-26:
Hi,
I have the same MagicHome device with the new generation circuit board depicted above by @PC-Eggsberde  and @app41.  Could someone please post some instructions/additional pics where the actual wires need to be soldered for flashing the device?
Thank you in advance!

closed 2018-03-15 by oxcxsxixbxaxn

margusveer, 2018-03-28:
I flashed (TASMOTE'd) my MagicHome module ( without Rx/Tx pads) and from web view seems that the flashing was successful I can connect to module over web BUT - 
there are not possibility even on/off or something like that 
I tried  change the module to  34 MagicHome or 38 Arilux LC11, but nothing 

from console info : 
`00:00:00 Project sonoff ledtest (Topic ledtest, Fallback DVES_C1FBBD, GroupTopic sonoffs) Version 5.12.0g-2_4_0
00:00:00 WIF: Connecting to AP1 HAWiFi in mode 11N as ledtest-7101...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on ledtest-7101.local with IP address 192.168.200.220
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/ledtest/LWT = Online (retained)
00:00:05 MQT: cmnd/ledtest/POWER = 
00:00:05 MQT: tele/ledtest/INFO1 = {"Module":"MagicHome","Version":"5.12.0g","FallbackTopic":"DVES_C1FBBD","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/ledtest/INFO2 = {"WebServerMode":"Admin","Hostname":"ledtest-7101","IPAddress":"192.168.200.220"}
00:00:05 MQT: tele/ledtest/INFO3 = {"RestartReason":"Software/System restart"}
15:01:01 MQT: tele/ledtest/STATE = {"Time":"2018-03-28T15:01:01","Uptime":"0T00:00:13","Vcc":3.477,"Wifi":{"AP":1,"SSId":"HAWiFi","RSSI":94,"APMac":"46:D9:E7:03:C4:BB"}}`

seems the module is restarting but the module stays in the same Module config like I but 
 any ideas ?

Jason2866, 2018-03-28:
You have to SetOption15 to 1 first

margusveer, 2018-03-28:
Thanks - that SetOption 15 to 1 helps me little bit, but Module Type MagicHome My Home Assistant conf almost don't work in Arilux LC11 the config massed up as well 
 in color pallet only red works,
color temp could is but blue led to light and warm but the Green to work ;) 
and brightness  massed up all again

Jason2866, 2018-03-28:
One step behind the other...
Go to the web console from the module. Test color 1,2,3. Are the correct?
If not you have to change the config of the module. How? Read this article from the beginning ;-)

margusveer, 2018-03-28:
In the web module Arilux LC11 only have Cold,/ Warm slider and Dark / Bright slider and Toggle - I don't have option for three colors

Jason2866, 2018-03-28:
You have enter manualy in the web console the command 
Color 1 should be light red
Color 2 should be light green
Color 3 should be light blue
Read the Wiki -> Commands

margusveer, 2018-03-28:
yes I read the discussion  and I changed Module to 37 and now the color commands works and with color + command coming more interesting colors and end with white - seems the LED strip and module works fine through the WEB now I must find the right configuration for home assistant to describe the module for the system ;)

gedw99, 2018-04-25:
anyone willing to make a few board for me ?
I am in Berlin, Germany. Happy to pay someone for their trouble of course.

I am a software guy and useless at hardware :) I am also a building architect and really like the options Sonoff with custom firmware gives me.

Jason2866, 2018-04-25:
@gedw99 
I am from germany too near Ingolstadt, Bavaria. If you need only two pieces i could do it, 
if you are not in a hurry. It may took 2 weeks (job, family, friends, house...). 
I wouldnt take money only the cost for shipping and it would be nice if you would spend a few euros 
for a social organisation like "Die Tafel"
If you are interested, leave a message, with a contact
P.S. if you buy, buy only second one. The first is maybe different and cant be used...
Greets Jason2866

gedw99, 2018-04-26:
@Jason2866 
thank you and "The Tafel" sounds awesome. We have one here in Potsdam that we support too :)

You can grab me on:
gedw99 AT gmail DOT com 

Ok will get second one - thanks for the tip.

When i get your address i will order what you think i need and get it delivered to you.

qrioniclabs, 2018-06-04:
@Jason2866 I am from Germany, too. Although I have some experience in flashing Sonoff devices, as soon as these controllers arrived, I thought it might be a bit more complicated... would you be willing to flash two controllers for me, too? Would very much like to make a donation as well. You can reach me here: core [ATTTT] qrioniclabs.com

Would be much appreciated! Thanks!

mjmagro, 2018-06-15:
Hi all, Do you have any experience using Tasmota FW with these new controllers supporting WS2812 or SK6812 LED pixels? Not sure if flashing Tasmota to these devices with in-built ESP would then support operating the pixel LEDs.

Devices available here:

MagicHome for WS2812:

https://www.ebay.com/itm/Nexlux-WiFi-led-controller-Music-for-WS2811-2812-2801-2813-sk6812-IC-LED-Strip/323296343789?hash=item4b45f6b6edⓂ️m70salehAWTb2YznZ_5qfbg

H803WF:

https://www.aliexpress.com/item/led-wifi-controller-support-2048-pixels-full-DIY-WS2812-WS2811-UCS1903-DMX512-artnet-remote-control-Madrix/32861800586.html?spm=2114.search0204.3.48.31935a21CgFSwd&ws_ab_test=searchweb0_0,searchweb201602_3_10152_10151_10065_10344_10068_5722815_10342_10343_10340_5722915_10341_10698_5722615_10697_10696_10084_10083_10618_10307_10301_10303_5722715_5711215_10059_308_100031_10103_441_10624_10623_10622_5711315_5722515_10621_10620,searchweb201603_32,ppcSwitch_5_ppcChannel&algo_expid=2d02e428-4f58-4297-8cdc-4783cb9b222e-7&algo_pvid=2d02e428-4f58-4297-8cdc-4783cb9b222e&transAbTest=ae803_2&priceBeautifyAB=0

Thanks!

titomontes, 2018-07-24:
Hello

I just bought a version not mentionesd above:

[https://photos.app.goo.gl/YvPafjamLk2j45GN9](url)

app41, 2018-07-25:
chinese market is running fast, they are cloning one each other....
be pragmatic, don't worry about the local circuitry, but search for the ESP-M2 module pinout and you will find all the pins you need

titomontes, 2018-07-26:
So you mean this one?:
https://camo.githubusercontent.com/58f6cda8acd2097f44d240efb9c6192f6f271b3c/68747470733a2f2f692e696d6775722e636f6d2f415758725954702e6a7067

kpcz, 2018-08-16:
Hello. How do you enter in AP mode with magic home controller? flashed succesfully, but there is no new SSID showing up, as it is with Sonoff devices.

Stefan300381, 2018-08-17:
@oxcxsxixbxaxn 

I own a v23 board as well. Will any OTA upgrade break your changes for Arilux LC01?

Stefan300381, 2018-08-17:
@kpcz 
Just guess into the blue (as I never use AP mode and hardcode my Wifi credentials in the firmware). I was able to reset my device by plugging in and plugging out the 12v power supply three times in a row. like plugin, wait 1s, plugout, wait 1s, plugin....

Stefan

SniFManS, 2018-08-23:
Titomontes, same board here. Searching the write way to flash it so I can use it whit Hassio.

stewface, 2018-09-20:
I seem to be struggling getting this flashed too - trying to use the arduino software anyone know the settings to use? 

[image]

ascillato, 2018-09-20:
Hi,

Please read the wiki. All the information related to flashing is there.

Thanks

GasTurbineMan, 2018-10-11:
TitoMontes, I purchased what appears to be the exact same board as you are showing in your post above.  No "brand" but brief instructions direct you to download and install the Magic Home app.  I immediately flashed my controller board with Tasmota 6.2.1.14  using the ESP-M2 diagram you show above.  Only issue I had was that I could not get my ground wire to stick on the ESP-M2 GND pin so I just soldered my GND wire to the Power In ground wire (I checked with VOM and both Power In ground and ESP-M2 GND are connected).  I used a USB FT232 programmer and PlatformIO to build and flash my device.  NO NEED TO HOOK UP THE VCC PIN on your FT232 programmer since we are powering the device with 12V power supply.  Just make sure you short GPIO0 to GND to enable flashing (I left mine shorted together during the entire flash process). 
1.  Solder needed wires to LED Controller (ESP-M2)  (GND, GPIO0, RX, TX)
2.  Connect to USB programmer.  GND to GND, GPIO0 to GND, RX to TX and TX to RX.
3.  With 12v power supply to LED Controller UNPLUGGED, plug in USB Controller to PC.
4.  Now apply 12v power to LED Controller.
5.  Flash Device.
6. Unplug 12v power.
7.  Disconnect in this order remaining wires:  Jumper from GPIO0 to GND, RX, TX and finally your GND.

Now going to post my own questions in next post.

GasTurbineMan, 2018-10-11:
Now that I have my LED Controller flashed with Tasmota.  I am trying to determine how best to set up my Module Configuration.  I have read many older posts but my assumption is that those recommendations were made for older versions of Tasmota, maybe before Theo added specific support for say MagicHome and AriLux LCxx.  I have played around with my board configured as 34-MagicHome, 37-Arilux and 18-Generic.  So far I can only get every other LED in my string to light up, but can get the LEDs to change colors and brightness.  My LED string is RGBW.  My LED Controller has 5 pins out as follows:
Red Triangle (assume this is Vcc Power out), R, G, B, W

Based upon current version of Tasmota, what would be the "correct" configuration for my controller?

jasoncodes, 2018-10-11:
I also have a clone labelled “ZJ-ESP-IR-F V1.1” which appears to be the same as pictured above. Here’s the config I am using:

Module type: 18 Generic
GPIO4: 51 IRrecv
GPIO12: 37 PWM1
GPIO5: 38 PWM2
GPIO13: 39 PWM3
GPIO15: 40 PWM4
SetOption15 on
SetOption20 on
Speed 2
LedTable on

GasTurbineMan, 2018-10-12:
Jason, many thanks for sharing your configuration settings.  They appear to be spot on.  Now I am playing with the Tasmota commands for the LED string – manually entering them using the Console.

 

My ultimate goal is maybe to use these for Xmas Lights on the house.  My next step will be to figure out how I can “automate” this device for some sort of color theme changes.  I saw in one post how to use a Tasmota Rule, so that is an option;  but I  also have my RPi3 and Node-Red.  May set something up there to run a sequence that will add a variety of themes.

 

Neat thing to tinker with.  Thanks again for the help.

hiddit, 2018-10-29:
[image]
hi
i orderd the led controller and got this.
so far i have txd and rxd connected and gnd, and try to connect with i00 and v33 but that doesnt work.
can you guys help me out?

Jason2866, 2018-10-29:
Did you solder TX RX Gnd?  Connect Gnd to IO0
Connect extern 12 V power supply.  Connect USB Adapter to PC 
Start flasher! If it doesnt work swap TX RX connections

Kepete, 2018-11-16:
@jasoncodes @GasTurbineMan 
You have the same version as I -> ZJ-ESP-IR-F V1.1
I have used the settings provided and dimming, RGB colors work fine. But the W, additional channel never turns on. 
I tried setting the colour 12 (white) and it makes the white only with RGB leds, doesn't use the extra channel. 
Any ideas?

jasoncodes, 2018-11-17:
> the W, additional channel never turns on.

Tasmota does not control the white channel when given an RGB colour. As fas as I can tell, you either need to provide it RGBW values to `Color` or set `Channel4` separately.

zibous, 2018-11-21:
@hiddit

I'm trying also to flash the Led Controller:

[image]

```
1. GND ----> GND
2. RX ------> TX
3. TX ------> RX
4. GND ----> IO0
5. 12V power supply
6. Connect USB adapter to PC
7. Start flashing sonoff.bin
```

After compiling, an error message immediately appears that the firmware has not been transferred. Even if I change TX / RX comes the same error.

Would not I have to connect the 3.3V from the USB adapter with the device?

Kepete, 2018-11-21:
No you don’t have to connect the 3.3V. Actually you are not allowed to do it! 

First connect the USB to computer and after it 12v power supply.

zibous, 2018-11-22:
@Kepete

Thanks for the information. Works now for me. :)

Only the  RGB LED - IR remote to not work any more.
At the moment I can only select the colors via the "***color 1 - 12***" command.

```
CMD: Color 1
  --> Switch OFF ???
  --> MQT: stat/wzlicht1/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF0000","HSBColor":"0,100,100","Channel":[100,0,0]}
CMD: Color 2
  -->  11:41:22 CMD: Color 2
  --> 11:41:22 MQT: stat/wzlicht1/RESULT = {"POWER":"ON","Dimmer":100,"Color":"00FF00","HSBColor":"120,100,100","Channel":[0,100,0]}
CMD: Color 3
CMD: Color 4
.....
```
Used Version; Sonoff-Tasmota 6.3.0.11

hadahop, 2018-11-29:
Same for me:

Arilux LC01 Module: Colors o.k, IR remote Control do not work
MagicHome Module  : Colors wrong, IR remote  Control do not work

```
Program Version	6.3.0
Build Date & Time	2018-10-30T17:33:53
Core/SDK Version	2_3_0/1.5.3(aec24ac9)
Uptime	0T00:50:16
Flash write Count	400 at FB000
Binary from: http://thehackbox.org/tasmota/release/
Board: Magic Mini RGB wifi Controller,HaoDeng Bulb
```

joshdinsdale, 2018-12-08:
So weird, I've flashed a blank.bin to erase it and re-flashed the sonoff firmware, both of which go without a hitch, however it doesnt connect to my wifi and i cant get any serial output from it, is it dead??

ozett, 2018-12-16:
i can succesfully compile tasmota 6.3 in arduino ide 1.61.11 for the esp8266,
i got a binary without errors.

if i change to esp8285 board-definition,
i got errors in sonoff.ini at 1414" ShowSource(source);" 
dont know whats going wrong

i wanted to compile for 
https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-with-ESP8285
any hints to solve this riddle?

thanx

ozett, 2018-12-16:
@jasoncodes 

you posted the GPIOS on your module. https://github.com/arendst/Sonoff-Tasmota/issues/1867#issuecomment-429110835 i have the same, my colors are wrong,
i will try your settings.  as i am new to tasmota, i need help with the "setoption"

how do i set this? can i do this in the firmware-source, sonof_template.h ?
or in the rules on startup? or via mqtt?

thankx for any help on this..
at the moment i have sdk ver2.3 and the module running easy on wifi. looks promising so ar..
👍 

https://github.com/arendst/Sonoff-Tasmota/issues/1867#issuecomment-429110835
> I also have a clone labelled “ZJ-ESP-IR-F V1.1” 
> SetOption15 on
> SetOption20 on
> Speed 2
> LedTable on

jasoncodes, 2018-12-16:
> i need help with the "setoption"
> 
> how do i set this? can i do this in the firmware-source, sonof_template.h ?
> or in the rules on startup? or via mqtt?

You can run change those settings via MQTT or the web console.

ozett, 2018-12-17:
thanx,
i am still a bit clueless and would be very gladful for one hint more ... 😸 
i still have the ZJ-ESP-IR-F V1.1, and i flashed it now 10 times with tasmota 6.4 on my arduino ide 1.6.11
everything work fine, as i use board sdk v2.3 also now wifi works like charm.

BUT

i dont get blue color.

i tried to change the pwm`s on the GPIO 5/12/13/14/15 differently - with no usable result.
i dont recognize any system behind that, because:

if i only separately assign PWM1 to GPIO5 (only!) and set everthing else to 0 (zero),
and choose the "color 1"-cmd, i get no lights, if i chose color 2 and color 3 i get red and green.

this happens also, if i assign PWM2 to GPIO5 (only!) and leave all else untouched to 0 (zero).
if i change board-definiton to ARILUX_LC11, the behavior stays similar.

any hint appreciate to check colors and the right GPIOS separately.
do PWM to GPIO have a special system? 
or do i have to define the whole Color-Order elsewhere? 
my stripe is GRB (no white)
 
maybe also my hardware is damaged?

thanx a lot,
toz

[18 more replies]