arendst/Tasmota #1869: core_version.h not found during compilation

wsw70, 2018-02-11:
Following the instructions, I got stuck at the compilation stage:

```
Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Uno"

D:\arduino-1.8.5\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino-1.8.5\hardware -hardware D:\arduino-1.8.5\portable\packages -tools D:\arduino-1.8.5\tools-builder -tools D:\arduino-1.8.5\hardware\tools\avr -tools D:\arduino-1.8.5\portable\packages -built-in-libraries D:\arduino-1.8.5\libraries -libraries D:\arduino-1.8.5\portable\sketchbook\libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10805 -build-path C:\Users\yop\AppData\Local\Temp\arduino_build_213783 -warnings=none -build-cache C:\Users\yop\AppData\Local\Temp\arduino_cache_727316 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=D:\arduino-1.8.5\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=D:\arduino-1.8.5\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=D:\arduino-1.8.5\hardware\tools\avr -verbose C:\Users\yop\Desktop\arduino-1.8.5\portable\sketchbook\sonoff\sonoff.ino
D:\arduino-1.8.5\arduino-builder -compile -logger=machine -hardware D:\arduino-1.8.5\hardware -hardware D:\arduino-1.8.5\portable\packages -tools D:\arduino-1.8.5\tools-builder -tools D:\arduino-1.8.5\hardware\tools\avr -tools D:\arduino-1.8.5\portable\packages -built-in-libraries D:\arduino-1.8.5\libraries -libraries D:\arduino-1.8.5\portable\sketchbook\libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10805 -build-path C:\Users\yop\AppData\Local\Temp\arduino_build_213783 -warnings=none -build-cache C:\Users\yop\AppData\Local\Temp\arduino_cache_727316 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=D:\arduino-1.8.5\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=D:\arduino-1.8.5\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=D:\arduino-1.8.5\hardware\tools\avr -verbose C:\Users\yop\Desktop\arduino-1.8.5\portable\sketchbook\sonoff\sonoff.ino
Using board 'uno' from platform in folder: D:\arduino-1.8.5\hardware\arduino\avr
Using core 'arduino' from platform in folder: D:\arduino-1.8.5\hardware\arduino\avr
Detecting libraries used...
"D:\arduino-1.8.5\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-ID:\arduino-1.8.5\hardware\arduino\avr\cores\arduino" "-ID:\arduino-1.8.5\hardware\arduino\avr\variants\standard" "C:\Users\yop\AppData\Local\Temp\arduino_build_213783\sketch\sonoff.ino.cpp" -o "nul"
"D:\arduino-1.8.5\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-ID:\arduino-1.8.5\hardware\arduino\avr\cores\arduino" "-ID:\arduino-1.8.5\hardware\arduino\avr\variants\standard" "C:\Users\yop\AppData\Local\Temp\arduino_build_213783\sketch\sonoff.ino.cpp" -o "C:\Users\yop\AppData\Local\Temp\arduino_build_213783\preproc\ctags_target_for_gcc_minus_e.cpp"
C:\Users\yop\Desktop\arduino-1.8.5\portable\sketchbook\sonoff\sonoff.ino:31:143: fatal error: core_version.h: No such file or directory

 #include <core_version.h>                   // Arduino_Esp8266 version information (ARDUINO_ESP8266_RELEASE and ARDUINO_ESP8266_RELEASE_2_3_0)

                                                                                                                                               ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Uno.

```

The file is actually there (in `D:\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266`) but seems not to be found.

**I do not have experience with the Arduino IDE: should I instruct it where to look for the includes?**

The only thing which was incosenquent in my installation before compilation is that it was stated to 

> (...) replace in Arduino IDE folder `portable\packages\esp8266\hardware\esp8266\2.3.0 file platform.txt` with the Tasmota Source code file `arduino\version 2.4.0\platform.txt`.*

I only have a `portable\packages\esp8266\hardware\esp8266\2.4.0` folder (and thought it may be a typo)

Jason2866, 2018-02-11:
Using Atom is much easier:
https://github.com/arendst/Sonoff-Tasmota/wiki/Beginner-Guide---Create-your-own-Firmware-Build#building-the-firmware

wsw70, 2018-02-11:
@Jason2866 : I will give it a try, thanks. Once I have (hopefully) successfuly built the firmware, how to flash it to the Sonoff? (I assume that in the Arduino version this is done via the arduino)

Jason2866, 2018-02-11:
Flashing is included in Atom
Just use in the menu "Upload"

wsw70, 2018-02-11:
@Jason2866 : Thank you  -it is indeed way easier and the build worked great.

The upload did not (I will open a question specifically for this as I am getting 

```
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

```
when uploading (discussed in several threads - but none of the solution helped) - I would however hijack this issue for just one question: is the LED on the Sonoff supposed to be lighting up at any time (say, to inform that the device is in flash mode)?

Jason2866, 2018-02-11:
The leds are off during the whole flash process

claudioita, 2018-08-24:
I have same error (tried Atom but would not install platformio). Anyway to solve it with Arduino IDE?

carloszacu, 2018-10-23:
getting the same error... any ideas?

ascillato, 2018-10-23:
uninstall all your arduino or platformio and install it again and download all the libraries and all tasmota again. please, follow the instructions in the wiki

carloszacu, 2018-10-23:
hey, I found that updating the ESP8266 board library solved the problem for me, thanks!