arendst/Tasmota #1870: query please

handsomejackuk, 2018-02-11:
can anyone offer me some help how can i get the RSSI data as a mqtt topic so i can view its value.. in realtime, i can see it there in this string...

**tele/s1/STATE = {"Time":"2018-02-11T18:53:41","Uptime":"0T04:35:18","Vcc":3.230,"POWER1":"OFF","POWER2":"OFF","Dimmer":9,"Color":"22,22,22","Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"xxxxxxxxxxxxxx",**"RSSI":64_,"**APMac":"xxxxxxxxxxxxx"}}**

curzon01, 2018-02-11:
Message are in [JSON](https://en.m.wikipedia.org/wiki/JSON) format, pls take a research first.

handsomejackuk, 2018-02-11:
I have tried stat/s1/RSSI but no value is returned I see that the values of power1 and power2 are the relays and I can see those values being sent to Mqtt dash and Mqtt snooper when I run stat/s1/POWER1 values are on and off which I expected but no RSSI

Jason2866, 2018-02-11:
Example of items file for Uptime

String Sonoff_Basic_Up "Sonoff_Basic: Uptime [%s]" (gUptime)
    { mqtt="<[broker:tele/sonoff-yourxyz/STATE:state:JSONPATH($.Uptime)]" }

handsomejackuk, 2018-02-11:
phew thanks, i have a look at that.... I thought it would have just been a case of maybe in my mqtt dashboard ( android phone) using stat/s1/RSSI and that would have sent the RSSI command to mqtt dashboard. Sonoff sent the on / off values fine with the stat/s1/POWER1, POWER2 statement.. i sorry but this mqtt is all new to me... thanks Again...

Jason2866, 2018-02-11:
Your welcome,
i am new to this too. At the beginning it is really a lot of new stuff to understand and learn
But if you got the basic and the concept it is fun what is possible with MQTT (and this nice Sonoffs)  and OpenHab. With some simple parts Ir Receiver an Ir Diode you can manage many Tv, Amplifier. 
Combined with Alexa you can switch this on/off youst saying Tv on 😀

handsomejackuk, 2018-02-11:
i have already added ws2812b led strips to the sonoff basic, and added bmp180 barometric sensor and temp sensor to the i2c bus all work well, my hardware knowledge is good but programming skills need a bit more work, have used arduino before, and built a few quacdopters with atmel 328p so know my way around the hardware, but have been a bit spoilt with some of the excellent software in the quadcopter world.... as far as alexa, i have got google assistant running with sonoff original firmware but not got that running with tasmota yet, thats on my list of things to do.... the ir diode sounds good too i have loads of them here, will they work easily with the webinterface on tasmota or does it need more setting up ?

handsomejackuk, 2018-02-11:
https://www.youtube.com/watch?v=CPw5ixEF89Y

handsomejackuk, 2018-02-11:
used the 5v rail on the sonoff to power the ws2812 leds theres only 2 but anymore i think would need additional power supply.

Jason2866, 2018-02-13:
IR Send
https://github.com/altelch/SonoffIR

closed 2018-07-10 by ascillato2