arendst/Tasmota #1872: Sonoff Basic random exception restarts

sammyke007, 2018-02-11:
Hi

I have a Sonoff Basic running Tasmota 5.11.1. On GPIO14 and GND a hardware switch is connected. Serial logging is set to 0.

I have this running fine on 2 other Sonoff's (uptime +72 hours already) but a third one reboots randomly with the reason "Exception". Sometimes it's running fine for 10 hours and then it starts rebooting 5-10 times an hour...

Anyone has an idea what's wrong please?
I flashed all of my Sonoffs with ArduinoIDE without a problem.

Here are some log examples:

`00:00:00 Project sonoff Sonoff1_Slaapkamer (Topic sonoff1, Fallback DVES_096A35, GroupTopic sonoffs) Version 5.11.1
00:00:00 WIF: Connecting to AP1 Wifi13 in mode 11N as sonoff1-2613...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff1-2613.local with IP address 192.168.1.198
00:00:06 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/sonoff1/LWT = Online (retained)
00:00:06 MQT: cmnd/sonoff1/POWER = 
00:00:06 MQT: tele/sonoff1/INFO1 = {"Module":"Sonoff Basic","Version":"5.11.1","FallbackTopic":"DVES_096A35","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/sonoff1/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff1-2613","IPAddress":"192.168.1.198"}
00:00:06 MQT: tele/sonoff1/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x40229b38 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
00:00:07 MQT: stat/sonoff1/RESULT = {"POWER":"OFF"}
00:00:07 MQT: stat/sonoff1/POWER = OFF
00:00:07 MQT: domoticz/in = {"idx":145,"nvalue":0,"svalue":""}

----------

00:00:00 Project sonoff Sonoff1_Slaapkamer (Topic sonoff1, Fallback DVES_096A35, GroupTopic sonoffs) Version 5.11.1
00:00:00 WIF: Connecting to AP1 Wifi13 in mode 11N as sonoff1-2613...
00:00:07 WIF: Connected
00:00:07 DNS: Initialized
00:00:07 HTP: Web server active on sonoff1-2613.local with IP address 192.168.1.198
20:45:51 MQT: Attempting connection...
20:45:51 MQT: Connected
20:45:51 MQT: tele/sonoff1/LWT = Online (retained)
20:45:51 MQT: cmnd/sonoff1/POWER = 
20:45:51 MQT: tele/sonoff1/INFO1 = {"Module":"Sonoff Basic","Version":"5.11.1","FallbackTopic":"DVES_096A35","GroupTopic":"sonoffs"}
20:45:51 MQT: tele/sonoff1/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff1-2613","IPAddress":"192.168.1.198"}
20:45:51 MQT: tele/sonoff1/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x40106e6b epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
20:45:52 MQT: stat/sonoff1/RESULT = {"POWER":"OFF"}
20:45:52 MQT: stat/sonoff1/POWER = OFF
20:45:52 MQT: domoticz/in = {"idx":145,"nvalue":0,"svalue":""}

-----------

00:00:00 CFG: Loaded from flash at F4, Count 728
00:00:00 APP: Boot Count 169
00:00:00 Project sonoff Sonoff1_Slaapkamer (Topic sonoff1, Fallback DVES_096A35, GroupTopic sonoffs) Version 5.11.1
00:00:00 CFG: Saved to flash at FB, Count 729, Bytes 1648
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Wifi13 in mode 11N as sonoff1-2613...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on sonoff1-2613.local with IP address 192.168.1.198
00:00:07 APP: (UTC) Sun Feb 11 19:49:11 2018
00:00:07 APP: (DST) Sun Mar 25 02:00:00 2018
00:00:07 APP: (STD) Sun Oct 28 03:00:00 2018
20:49:12 MQT: Attempting connection...
20:49:12 MQT: Connected
20:49:12 MQT: tele/sonoff1/LWT = Online (retained)
20:49:12 MQT: cmnd/sonoff1/POWER = 
20:49:12 MQT: Subscribe to cmnd/sonoff1/#
20:49:12 MQT: Subscribe to cmnd/sonoffs/#
20:49:12 MQT: Subscribe to cmnd/DVES_096A35/#
20:49:12 MQT: Subscribe to domoticz/out/#
20:49:12 MQT: tele/sonoff1/INFO1 = {"Module":"Sonoff Basic","Version":"5.11.1","FallbackTopic":"DVES_096A35","GroupTopic":"sonoffs"}
20:49:12 MQT: tele/sonoff1/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff1-2613","IPAddress":"192.168.1.198"}
20:49:12 MQT: tele/sonoff1/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x4021ab92 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
20:49:13 MQT: stat/sonoff1/RESULT = {"POWER":"OFF"}
20:49:13 MQT: stat/sonoff1/POWER = OFF
20:49:13 MQT: domoticz/in = {"idx":145,"nvalue":0,"svalue":""}

----

0:00:00 CFG: Loaded from flash at F6, Count 734
00:00:00 APP: Boot Count 174
00:00:00 Project sonoff Sonoff1_Slaapkamer (Topic sonoff1, Fallback DVES_096A35, GroupTopic sonoffs) Version 5.11.1
00:00:00 CFG: Saved to flash at F5, Count 735, Bytes 1648
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Wifi13 in mode 11N as sonoff1-2613...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Connected
00:00:07 DNS: Initialized
00:00:07 HTP: Web server active on sonoff1-2613.local with IP address 192.168.1.198
00:00:07 APP: (UTC) Sun Feb 11 19:56:09 2018
00:00:07 APP: (DST) Sun Mar 25 02:00:00 2018
00:00:07 APP: (STD) Sun Oct 28 03:00:00 2018
20:56:11 MQT: Attempting connection...
20:56:11 MQT: Connected
20:56:11 MQT: tele/sonoff1/LWT = Online (retained)
20:56:11 MQT: cmnd/sonoff1/POWER = 
20:56:11 MQT: Subscribe to cmnd/sonoff1/#
20:56:11 MQT: Subscribe to cmnd/sonoffs/#
20:56:11 MQT: Subscribe to cmnd/DVES_096A35/#
20:56:11 MQT: Subscribe to domoticz/out/#
20:56:11 MQT: tele/sonoff1/INFO1 = {"Module":"Sonoff Basic","Version":"5.11.1","FallbackTopic":"DVES_096A35","GroupTopic":"sonoffs"}
20:56:11 MQT: tele/sonoff1/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff1-2613","IPAddress":"192.168.1.198"}
20:56:11 MQT: tele/sonoff1/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x402273a3 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
20:56:12 MQT: stat/sonoff1/RESULT = {"POWER":"OFF"}
20:56:12 MQT: stat/sonoff1/POWER = OFF
20:56:12 MQT: domoticz/in = {"idx":145,"nvalue":0,"svalue":""}
20:56:19 MQT: tele/sonoff1/STATE = {"Time":"2018-02-11T20:56:19","Uptime":0,"Vcc":3.504,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wifi13","RSSI":100,"APMac":"50:46:5D:6D:61:28"}}
20:56:19 MQT: tele/sonoff1/SENSOR = {"Time":"2018-02-11T20:56:19","Switch1":"ON"}

`

sammyke007, 2018-02-11:
Hi Jason, I have a pullup 10k.ohm installed between 3.3V and GPIO14 yes. I don't have flickering lights, just reboot problems.

Jason2866, 2018-02-11:
For finding the problem disconnect the switch on GPIO 14 in the Sonoff Modul.
If reboots are gone your problem is the connected switch. If not than the Sonoff is defect.
I think the problem comes with wire on GPIO 14. You maybe have build a "perfect" antenna for bringing your Sonoff in trouble. I am pretty sure thats your problem. Fixing it is the other side....

TLCary, 2018-02-13:
Same setup. Same problem.  Odd.   I removed the hardware and it still resets.   I go into configure modules and change it back to NONE in the software and it's fine.  That feels like a software problem.  Odd that I haven't seen it before.   Anyone else?  Any suggestions?

Jason2866, 2018-02-13:
Hi,
which Options have you set? 
I want to try your setup, if i can reproduce the error.
I use Version 5.12. master (en).  The german 5.12 seems to be buggy

sammyke007, 2018-02-14:
Upgraded to 5.12 and online fo 4 hours now... I will report back later!

sammyke007, 2018-02-16:
Online for 24 hours now after I flashed it with 5.12 and disabled all the sensors and the HomeAssistant part. Don't know what was wrong...

closed 2018-02-16 by sammyke007