arendst/Tasmota #1874: enhancement to webinterface.....

handsomejackuk, 2018-02-11:
would it be possible to add voltage of the 3v power supply rail to the webinterface, and rssi values  on the main page...?

just a thought as the data is there would be nice to see wifi signal strength, the original sonoff ewelink has this...

i dont know why 3v rail voltage would be needed but maybe handy if adding a lot of additional sensors, i have added my sensor to the 5v side of the sonoff not the 3v rail, i cant see where its taken from in the schematic to provide the 3v unless its internal to the esp8266.....

reloxx13, 2018-02-12:
check out [SonWEB](https://github.com/reloxx13/SonWEB).

flash is limited thats why the webinterface has to be "small".

handsomejackuk, 2018-02-12:
OK fair comment I understand there's only so much flash storage....I have a look into it myself.. And possibly remove some unnecessary code from my setup.... Not that I am that good on coding but I am gonna have a look... Would have been nice though just a rssi value displayed much the same as my barometer and temp is displayed..... Ahhh well