arendst/Tasmota #1875: BME280 MQTT

echo4019, 2018-02-11:
Hi, 

I've recently installed a BME280 on a sonoff basic. I'm receiving the sensor values on the Tasmota web interface but I'm unable to receive the sensor values on home assistant using MQTT. 

I've tried:

-platform: mqtt
 name: "BME280 Temperature"
 state_topic: "tele/bme280/SENSOR"
 unit_of_measurement: "°C"
 value_template: "{{ value_json.temperature | round(1) }}"
-platform: mqtt
 name: "BME280 Humidity"
 state_topic: "tele/bme280/SENSOR"
 unit_of_measurement: "°%"
 value_template: "{{ value_json.humidity| round(1) }}"
-platform: mqtt
 name: "BME280 Pressure"
 state_topic: "tele/bme280/SENSOR"
 unit_of_measurement: "°%"
 value_template: "{{ value_json.pressure| round(1) }}"

any ideas where I'm going wrong?

Thanks.

zewelor, 2018-02-12:
Thats my config:

```
- platform: mqtt
  state_topic: 'tele/LIVINGROOM/SENSOR'
  value_template: "{{ value_json.BME280.Temperature }}"
  name: 'Living Room temperature'
  unit_of_measurement: "°C"
- platform: mqtt
  state_topic: 'tele/LIVINGROOM/SENSOR'
  value_template: "{{ value_json.BME280.Humidity }}"
  name: 'Living Room Humidity'
  unit_of_measurement: "%"
- platform: mqtt
  state_topic: 'tele/LIVINGROOM/SENSOR'
  value_template: "{{ value_json.BME280.Pressure }}"
  name: 'Living Room Pressure'
  unit_of_measurement: "mb"
```

echo4019, 2018-02-12:
Thanks a bunch that done the trick.

Jason2866, 2018-02-13:
Hi echo4019,

i ordered thats sensor too. How is it connected? Which pins go to where? What do you have configured in the Sonoff?  The Json Thing in OpenHab i can handle.
THX

Jason2866, 2018-02-13:
i found it by myself: Sensor connecting is described in the Wiki...
https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-Configuration