arendst/Tasmota #1877: Backlog command error

roguestreak, 2018-02-12:
Hi Theo
During a discussion on the sonoffusers group, "NodeRed MQTT for Tasmota; which parts are topic, payload etc?" (https://groups.google.com/forum/#!topic/sonoffusers/hqwm-5yaa9c), it appears that a bug in the Backlog command has been identified, creating an error when 16 commands are added.
Here's Bryan Mayland's comment/finding:

> I just looked at the code and you're right, if you add 16 things to the backlog there's a bug/design issue that causes the index to roll over back to the first index and then it thinks there is nothing in the backlog any more. Unfortunately, the backlog buffer is circular so there's no way to prevent this other than waiting at least 0.2 seconds for at least one item to be removed from the backlog to hold that 16th item without wrapping the index.

reloxx13, 2018-02-12:
Which Firmware Version?

#1819

roguestreak, 2018-02-12:
5.11.1, but seems unrelated to your issue?  Backlog works and no restarts, but results in an empty backlog command if 16 commands are added

arendst, 2018-02-12:
It supports indeed up to 15 commands and a total command line length of 254 characters so much more commands would not be present anyway.

roguestreak, 2018-02-12:
If 15 is the maximum then that's fine, and works for my job at hand, but the documentation might need an update; it says 

> execute several commands in sequence (maximum 16/request)