arendst/Tasmota #1879: High availability Wifi

nosaj66au, 2018-02-12:
Hi All now getting down to final design of the new home automation system. Tasmota seems to support redundant networks or at least 2 networks with role over to a secondary AP. Does anyone have any suggestions how to make the most reliable system topology.
My top end at this stage is Node Red on a Virtual Windows box on ESXi. The house is large enought to require a number of AP's and when finished the system will require appox 75-100 devices.
 
I guess I have a few decisions. 
Network topology - Mesh network (i.e. Google),WIFI networks with AP's , Dual WIFI with AP’s
 IP Address - DHCP's or Fix IP's
POE or not
auto OTA firmware upgrade (scheduled of course)
Keep in mind I will also have IP Cameras and intent to run Google Casting on the network so there will be a fair bit of traffic.

Any suggestions welcome
Jason

davidelang, 2018-02-12:
https://www.usenix.org/conference/lisa12/technical-sessions/presentation/lang_david_wireless
https://www.usenix.org/conference/lisa12/technical-sessions/presentation/lang_david_wireless

put all your APs on the same SSID, don't do mesh

static dhcpllocations

PoEor not depends on your hardware, if it's PoE and it's inconvienient to put a 
power supply near your device use PoE, if not, why did you pay for PoE you don't 
need ;-)

David Lang

  On Sun, 11 Feb 2018, nosaj66au wrote:

> Hi All now getting down to final design of the new home automation system. Tasmota seems to support redundant networks or at least 2 networks with role over to a secondary AP. Does anyone have any [...]

ascillato, 2018-02-12:
Thanks for sharing your presentation. Very useful!!! :+1:

davidelang, 2018-02-12:
On Mon, 12 Feb 2018, Adrian wrote:

> Thanks for sharing your presentation. Very useful!!! :+1:

while it's overkill, I figured it was a lot easier than trying to figure out 
which parts you needed and feed them to you one at a time :-)

closed 2018-07-10 by ascillato2