arendst/Tasmota #1880: Uploading in atom/platform.io yields a warning: espcomm_sync failed, error: espcomm_open, failed error: espcomm_upload_mem failed

wsw70, 2018-02-12:
I successfully [installed and built](https://github.com/arendst/Sonoff-Tasmota/wiki/Beginner-Guide---Create-your-own-Firmware-Build) the firmware (thanks to feedback from @Jason2866).

I then tried to [upload](https://github.com/arendst/Sonoff-Tasmota/wiki/Upload) it to my Sanoff Basic, the operation failed with 

```
(...)
|-- <ArduinoJson> v5.11.2
|-- <ESP8266HTTPClient> v1.1
|   |-- <ESP8266WiFi> v1.0
|-- <Ticker> v1.0
Compiling .pioenvs\sonoff\src\sonoff.ino.o
Linking .pioenvs\sonoff\firmware.elf
Checking program size
text       data     bss     dec     hex filename
492771    10032   41328  544131   84d83 .pioenvs\sonoff\firmware.elf
Building .pioenvs\sonoff\firmware.bin
Configuring upload protocol...
Looking for upload port...
Auto-detected: COM6
Uploading .pioenvs\sonoff\firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

```

There are various advice on Internet about this error, I went to probably all of them, notably:

- making sure to press the button before connecting the USB to Serial bridge
- trying with `TX-TX,RX-RX` and `TX-RX,TX-RX` wiring
- tried a different USB port (USB3 and USB2)

I had a look at the verbose version of the flasing process (`pio run -v -t upload`) :

```
(...)
Checking program size
text       data     bss     dec     hex filename
492771    10032   41328  544131   84d83 .pioenvs\sonoff\firmware.elf
"esptool" -eo "C:\Users\yop\.platformio\packages\framework-arduinoespressif8266\bootloaders\eboot\eboot.elf" -bo .pioenvs\sonoff\firmware
.bin -bm dout -bf 40 -bz 1M -bs .text -bp 4096 -ec -eo .pioenvs\sonoff\firmware.elf -bs .irom0.text -bs .text -bs .data -bs .rodata -bc -
ec
Configuring upload protocol...

MethodWrapper(["upload"], [".pioenvs\sonoff\firmware.bin"])
Auto-detected: COM6
esptool -vv -cd ck -cb 115200 -cp "COM6" -cf .pioenvs\sonoff\firmware.bin
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
setting board to ck
setting baudrate from 115200 to 115200
setting port from COM1 to COM6
espcomm_upload_file
espcomm_upload_mem
setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
flush start*
** [upload] Error -1setting serial port timeouts to 1 ms

setting serial port timeouts to 1000 ms
flush complete
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
read 0, requested 1
trying to connect
flush start
(...)
```

**I am at loss what to do next.** 

Should I try to flash it with the Arduino? (using the compiled firmware, not sure yet how to combine both)

Or is there a step in the flashing process I missed? A serial baud rate setting perhaps?

reloxx13, 2018-02-12:
Try another USB Cable.
Try another PIN Cables.
Try another FTDI (does the green SMDs on the FTDI Flasher flash once if you plug it in USB and orange FTDI always on ?) 1/3 was defect which i bought form china.

Be sure to hold the Button down while pluging in USB.

The Button can be also defect.

ALWAYS use TX-RX RX-TX.

Be sure you OPENED the project woth the platformio.ini to select and not create a new project.

sebringsc, 2018-02-12:
Sounds like you have attempted to do the typical fix.
I ran into this issue using Arduino and found that lowering my upload speed to 9600 allowed it to upload (i've got other issue with the flash).
The only time I've seen this error using Platformio is when there isn't a good connection to the Sonoff or I did not set it in "flash" mode at the beginning.
Have you tried to upload this using Platformio rather than esptool?  You used it to build, might as well have it upload as well.

miley786, 2018-02-13:
Ran into multiple issues using Platformio trying to flash the sonoff for some reason, i have however found that using the Arduino software with the following settings seems to flash every time, just have to make sure sonoff in flash mode when starting

[image]

miley786, 2018-02-13:
Yes it is possible, as long as all the boards correct and files are in the correct location, you welcome to share your code and can test it for you?

wsw70, 2018-03-03:
After 20 days I managed to flash my devices. There are numerous solutions on the web, this is what worked for me (Sonoff Basic v. 1.0, made on 20-10-2017):

- download the latest [Tasmota binary release](https://github.com/arendst/Sonoff-Tasmota/releases) 
- download [esptool](https://github.com/espressif/esptool)
- prepare the sonoff for flashing
  - **I had to connect** `TX<->TX`, `RX<->RX`, despite what is stated in the documentation
- disconnect the 3.3V pin
- press the Sonoff button and **keep it pressed** (maybe you can release it, I do not know, keeping it pressed worked for me)
- start the flashing process: `esptool.py --port COM4 --no-stub write_flash 0x0  C:\Temp\sonoff.bin`
- Once you see the `...___...` output from `esptool.py`, plug in the 3.3V pin. I noticed that this tremendously helps with the timing (the flasing just jumps in when the devide is ready)

wsw70, 2018-03-03:
@miley786 thank you for the kind proposal for help - I finally managed to flash using `esptool` (description above)

closed 2018-03-03 by wsw70

reloxx13, 2018-03-04:
pls post your flash tool, seems rx tx is wrong on the tool.