arendst/Tasmota #1881: Only boots on 3.3v (POW)

thecomalley, 2018-02-12:
Can boot and access the web GUI when connected to the TFDI adaptor. However when connected to mains is appears the Sonoff wont boot, it does pass through power to the output though. I have re flashed multiple times and reset the user_config but no change. Any Ideas?

gajotnt, 2018-02-12:
The power supply must have died.

sebringsc, 2018-02-12:
This is a problem that I am having as well.  I can get the webserver to boot when using 3.3V but when plugged into main, my webserver is hit or miss. The webserver on 3.3V also only runs for ~5 min before it shuts down and I need to reboot to access again.

davidelang, 2018-02-12:
we have seen many cases where there is a cold solder joint or missing capacitor 
on the power supply, causing the devices to not work (or not work well) when on 
AC power.

thecomalley, 2018-02-13:
Sounds like my power supply must have died, Replacement unit is on the way hopefully better luck this time round

closed 2018-02-13 by thecomalley