arendst/Tasmota #1882: Upgrade Basic from  v5.11.1 to v5.12.0 sonoff-DE.bin - device dead

einmalacht, 2018-02-12:
Hello ,
i upgraded two basic Modules from 5.11.1 to 5.12.0 and after update both devices are not accessible anymore. (1 of 100 ping is working ) I think it happens when MQTT is active.
My other Modules ( POW, DUAL, S20 ) are working well with 5.12.0
Is it possible that there is a bug only for basic modules=?

Jason2866, 2018-02-12:
No problems here with my Sonoff Basic's

Herschdorfer, 2018-02-12:
similar problem here. 
Upgrade today to newest version.
Device is hopping between two wireless connections. IP is not available. Control not possible via MQTT.

salopette, 2018-02-12:
Auch das selbe Problem! 5.11.1 ist ok. aber die 5.12 geht nicht.

Jason2866, 2018-02-12:
Hi,
i have 5 Sonoff and no problem with 5.12. I use the english version.
Are you using the "de" Version? Maybe the localization is buggy?

salopette, 2018-02-12:
Yes, i use de Version. The error coming after flash of minimal.bin

Herschdorfer, 2018-02-12:
Yes, I tried the German version.

Jason2866, 2018-02-12:
Here is a log of one. Reacts normal and no reconnects:
20:02:00 MQT: tele/sonoff-13D92F/UPTIME = {"Time":"2018-02-12T20:02:00","Uptime":"1T05:24:08"}
20:03:07 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-02-12T20:03:07","Uptime":"1T05:25:15","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
20:08:08 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-02-12T20:08:08","Uptime":"1T05:30:15","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
20:13:10 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-02-12T20:13:10","Uptime":"1T05:35:15","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
20:18:11 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-02-12T20:18:11","Uptime":"1T05:40:15","Vcc":3.127,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
20:23:11 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-02-12T20:23:11","Uptime":"1T05:45:15","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
20:28:11 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-02-12T20:28:11","Uptime":"1T05:50:15","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
20:33:12 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-02-12T20:33:11","Uptime":"1T05:55:15","Vcc":3.127,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
20:38:12 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-02-12T20:38:12","Uptime":"1T06:00:15","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
20:43:13 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-02-12T20:43:12","Uptime":"1T06:05:15","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
20:48:13 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-02-12T20:48:13","Uptime":"1T06:10:15","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
20:53:13 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-02-12T20:53:13","Uptime":"1T06:15:15","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
20:58:13 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-02-12T20:58:13","Uptime":"1T06:20:15","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
21:02:00 MQT: tele/sonoff-13D92F/UPTIME = {"Time":"2018-02-12T21:02:00","Uptime":"1T06:24:02"}
21:03:13 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-02-12T21:03:13","Uptime":"1T06:25:15","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
21:08:14 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-02-12T21:08:14","Uptime":"1T06:30:15","Vcc":3.127,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
21:13:14 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-02-12T21:13:14","Uptime":"1T06:35:15","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
21:18:14 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-02-12T21:18:14","Uptime":"1T06:40:15","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
21:23:15 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-02-12T21:23:15","Uptime":"1T06:45:15","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
21:28:15 MQT: tele/sonoff-13D92F/STATE = {"Time":"2018-02-12T21:28:15","Uptime":"1T06:50:15","Vcc":3.126,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}

Jason2866, 2018-02-12:
I flashed it over the WebConsole no need for me to use the minimal Version.
Direct from 5.11.1 to 5.12 (both english Version)

Herschdorfer, 2018-02-12:
I can confirm: 
5.12.0 EN works, 
5.11.1 DE works.

I could flash the device OTA while pressing the button every 1 second. So the device stays responsive with corrupt DE firmware and does not disconnect from wireless.

Wondermusic, 2018-02-13:
I have the same problem here with de- Version.
After a while now, I was able to downgrade the two modules I tested with 5.12 to 5.11.1 again.

micfire, 2018-02-15:
same problem... 6 basics with 5.12.0 DE are not working and 2 POW + 1 TH10 with 5.12.0 DE are working

jeylites, 2018-02-15:
I flashed a 3gang Touch earlier today, I’m having the same issue on V5.12.0b. It looks like it defaults back to Sonoff Basic when after changing it to touch 3 gang.

Jason2866, 2018-02-16:
Could be this bug:
#1900

Pinellas, 2018-02-27:
I flashed several Sonoffs and "killed" 2 out of 3 Sonoff Basic with 5.12.0 DE.
After trial-and-error during a handfull of re-flashing, downgrade to 5.11.c worked but now, the following problem occurs:
I can not upgrade anymore as the system says, the flash size does not match.
I checked Sonoff information summary and found out, that "used flash", "free flash" and "available flash" do no match the displayed flash size of 1024kB, while all other working V.5.12 devices do.
The "killed" devices have been flashed by esptool.
Is it possible that the flashing did not erase all of the flash or may the false flashing with 5.12 and the boot loops over the next minutes did destroy parts of the plash chip?

SodoffBaldrick, 2018-02-27:
I've had success reviving "dead" ESPs by applying a blank image before reflashing with Tasmota.

You can download a blank file from here: http://www.pratikpanda.com/wp-content/uploads/2016/05/blank_1MB.zip

And Flash it any way you wish. Then try and install the original ROM that you wanted.
Not saying that it'll fix it, but its quick and worth a shot though.

Oxyandy, 2018-02-27:
esptool.py --port COM5 erase_flash
This is super fast, and I assure you is hex FF from start to finish

SodoffBaldrick, 2018-02-27:
I don't use ESPTOOL.py so went hunting for an alternative. I've heard really good things about it but for the simplicity I prefer using other tools and just giving it a blank BIN file.

Whatever works for you though :)

Oxyandy, 2018-02-27:
So you can you back-up original firmwares 
or make back ups & then copies of pre-configured firmwares .. ?

Jason2866, 2018-02-27:
With esptool you can backup the firmware of the device

SeriousOri, 2018-07-24:
Hello
I have updated the SonOff via the web interface.
"Upgrade by web server"
Now he does not work anymore.
It was previously Program Version 5.9.1.
I wanted to flash the v6.1.1.
Can I flash that again, so it goes again, or is it dead?