arendst/Tasmota #1885: Sonoff Hue Emulation <> Homematic CCU2

JoeWiseman, 2018-02-12:
Hello out there,

I've posted several questions in the past how to integrate the Tasmota-flashed-Sonoff-devices into the homematic system by using http-requests. This is working fine for sending singnals from the homematic-system to the Sonoff-devices, but not in the other direction ...

Now I saw that there is a "Hue Emulation" and asked myself if there is a way to use this emulation to make the Homematic and the Sonoff-devices talk to each other.
When using the original firmware of the Homematic-system actually ther is a beta-support for the hue integration, but it's said that there is only one hue-gateway allowed on the connected net.

So the question is: Do I need a (physically) hue-gateway from Philips to connect the Sonoff devices too, so that this is the "communication central" from Homematic to Hue, or is every Sonoff-device acting like a seperate gateway ... and if so, would it be possible to connect "all those gateways" to my system. Or maybe it is possible to set one of the Sonoff-devices as a "master gateway"?

Maybe someone has got some experiences along with the given combination??

Would be great to get some hints or - even better - an complete solution ;-)

Thanks in advance
Joe Wise

reloxx13, 2018-02-12:
just as sai din your another ticket, its not a full hue emu, and no, you cannot connect hue emulation with a real hue bridge cuz teh emulation emulates a hue bridge and a light per relais.

the "connect-button" is emulated as always pressed, means the device is always responding to msearchs and always response with the setup.

ask the homematic team for support multi gateways. as the gatesways can only have a limitied number of devices, this should be a pro to support it.

i think their will never be a full emualted bridge, since the esp as bridge could not handle alot of requests.
full emulated lights which works with a real hue bridge should work if someone is developing a firmware.

JoeWiseman, 2018-02-13:
Hi @reloxx13 
thanks for answering this topic too ... 
Good to know that the "connect-button" is emulated as always pressed! So I know how to handle that message. When waiting for a while the devices are recognized by the Homematic-system....
Again thanks a lot!
I think it is ok to close that issue and keep tracking the Hue Emulation-Issue.
Regards
Joe

closed 2018-02-13 by JoeWiseman