arendst/Tasmota #1886: Sonoff Button disable

rqsnst, 2018-02-12:
Hello,
is it possible to deactivate the button that switchs the relay manually
with a software parameter so that only switching via the web interface or Alexa is possible?

I realize that on the hardware side I can manipulate the button in such a way that it does not 
touches the button soldered on the board,
but I'm interested, if there are any parameters in the software with which I can turn it off.

Unfortunately, I did not find anything on the command page.

Does anyone have a solution?

reloxx13, 2018-02-12:
which sonoff, which firmware?

davidelang, 2018-02-13:
set buttontopic, that will make it so that the button no longer changes things 
locally, it only responds to mqtt messages

rqsnst, 2018-02-13:
@reloxx13
I use a S20 Socket with firmware 5.12.0

@davidelang
Is the command then `buttontopic 1` ?
i can try it after work

rqsnst, 2018-02-15:
Because I have not activated MQTT, the command buttontopic is not understood.
If I activate MQTT and enter the commando `buttontopic 1` , I can still toggle the relay with the button of the S20.
And i don`t have a MQTT server so this is no solution for me.

Are there any other hints or solutions?

l3d00m, 2018-03-03:
This is related to #984.

I'm interested in such a feature as well

timbru31, 2018-05-09:
I'm interested in this feature, too. I have a Sonoff T1 which controls a Hue lamp - When I press the button it should only send a MQTT message.

l3d00m, 2018-06-23:
Still relevant

closed 2018-08-22 by stale[bot]