arendst/Tasmota #1888: Hue Emulation

JoeWiseman, 2018-02-12:
Hello there,

I like to use the Sonoff-devices along with an Philips Hue System.

Is it ok to just enable the Hue-emulation mode or do I have to activate any other optins (like MQTT-enable or something else) to enable full Hue-communication?

Though the Sonoff-devices seem to act like a complete gateway (Hue-bridge + Hue-device) the question is, if the Sonoff-devices with enabled Hue-Emulation can be connected to an Hue bridge, just like other "original" Hue-devices? 

Is it possible to use one Sonoff-device as a gateway and the other devices just like "lights" without the gateway-functionality, so that the communiation is running via the master-gateway?

But the most important question is: Do the Sonoff-devices give a feedback (e.g. is they are toggled manually or by some http-request) to the Hue brigde they are bound too?

Thanks in advance for your help
Joe

reloxx13, 2018-02-12:
its not a full hue emulation.

cuz of limited flash it would cost too much storage to include a full hue emulation.

for a full hue emulation you should search for another firmware, tasmota is for mqtt and basic hue/wemo emulation.

the hue api is documented and opened by phillips, so technically yes it would be possible.

ascillato, 2018-02-13:
Hi

> Is it ok to just enable the Hue-emulation mode or do I have to activate any other optins (like MQTT-enable or something else) to enable full Hue-communication?

Emulation don't need MQTT. Just activate it on Tasmota.
 
> Though the Sonoff-devices seem to act like a complete gateway (Hue-bridge + Hue-device) the question is, if the Sonoff-devices with enabled Hue-Emulation can be connected to an Hue bridge, just like other "original" Hue-devices?

I don't know. Emulation is meant for being used with HomeAssistant or any other application compatible with Hue. Just a connection between a Sonoff device and a application.
 
> Is it possible to use one Sonoff-device as a gateway and the other devices just like "lights" without the gateway-functionality, so that the communiation is running via the master-gateway?

Not right now. I think Tasmota is meant to be a controller device not a gateway.

> But the most important question is: Do the Sonoff-devices give a feedback (e.g. is they are toggled manually or by some http-request) to the Hue brigde they are bound too?

Yes. I tested on Home Assistant and give feedback. So if i push the button on Sonoff and turn it on, it will be displayed on Home Assistant. Also, using Hue Emulation is automatically recognized by Home Assistant without need to configure anything there. Just activating Hue Emulation on Sonoff.

JoeWiseman, 2018-02-13:
Hi @reloxx13 and @ascillato 

thanks for answering so fast. That Tasmota is primary developed for supporting MQTT is not that new to me, but every time I see a light at the end of the tunnel in trying to get the Sonoff-devices more integrated in my home automation I seem to close my eyes on this fact ;-)

Important to me is what Adrian mentioned about the devices giving an feedback to the Home Assistant.  So I guess, hope and pray ;-) that it will give an answer to an Hue bridge as well.

Maybe someone has got some experiences in connecting the Sonoff-devices to an Hue bridge??

Regards
Joe

reloxx13, 2018-02-13:
> Maybe someone has got some experiences in connecting the Sonoff-devices to an Hue bridge??

how i said, its not possible. its a emulated hue bridge and you cannot connect a bridge with a bridge.

JoeWiseman, 2018-02-13:
OK, thanks a lot for answering again.... I think that my thoughts were wrong!
Regards
Joe

closed 2018-02-13 by JoeWiseman