arendst/Tasmota #1889: Error in 5.12.0a when compliling without USE_HOME_ASSISTANT

jhlweb, 2018-02-12:
When compiling 5.12.0a without defining USE_HOME_ASSISTANT in the user_config.h i get this error:

````
....Sonoff-Tasmota-5.12.0a\sonoff\settings.ino: In function 'void SettingsDefaultSet2()':
settings:433: error: 'HOME_ASSISTANT_DISCOVERY_ENABLE' was not declared in this scope
   Settings.flag.hass_discovery = HOME_ASSISTANT_DISCOVERY_ENABLE;
````

Should this be in a #ifdef statement?

ascillato, 2018-02-13:
If you do not want to use **Home Assistant**, your configuration on _user_config.h_ should be:

```
// -- MQTT - Home Assistant Discovery -------------
//#define USE_HOME_ASSISTANT                       // Enable Home Assistant Discovery Support (+1k4 code)
  #define HOME_ASSISTANT_DISCOVERY_PREFIX "homeassistant"  // Home Assistant discovery prefix
  #define HOME_ASSISTANT_DISCOVERY_ENABLE 0      // [SetOption19] Home Assistant Discovery (0 = Disable, 1 = Enable)
```

That error is shown when you comment the line:
`//  #define HOME_ASSISTANT_DISCOVERY_ENABLE 0      // [SetOption19] Home Assistant Discovery (0 = Disable, 1 = Enable)`

And yes, you are right. It should be inside an #ifdef statement

sammyke007, 2018-02-16:
I can confirm this. Only comment "#define USE_HOME_ASSISTANT  "!

ascillato, 2018-02-16:
@jhlweb if it works for you, please close this issue. Thanks

jhlweb, 2018-02-16:
Yes it works by only commenting '//#define USE_HOME_ASSISTANT'.
But then the comment should make it clear that by commenting the USE_HOME_ASSISTANT is enough.

closed 2018-02-16 by jhlweb

ascillato, 2018-02-16:
PR #1937