arendst/Tasmota #1890: Button Hold with SetOption13	on

Leseratte10, 2018-02-12:
The Sonoff Touch supports two button modes with Tasmota, SetOption13 off or on. With "off", it recognizes a single press, multi press or button hold. With "on", it just recognizes a button press and immediately switches before waiting for more button presses. 

What I'm looking for is a mode where the sonoff switches immediately when I touch and release the button, like SetOption12=on, but also supports sending a "Hold" message to MQTT when I'm holding the button. So, a mode where the sonoff just stores the current time when I press the button, and then when I release it, it should compare the time. If it is <0.5 second (or any other value, maybe configurable), it should switch the light without waiting for further button input. If the button was held for multiple seconds, it would not switch anything but just send a HOLD message to the MQTT broker - maybe even one message per 0.1 seconds of the button being pressed, containing the duration, so the MQTT server / a home automation system can do different things depending on how long the button has been pressed. 

I hope I described the function I'm looking for properly, English is not my main language. I have checked the configuration commands but haven't found one that does what I want; and I took a look into the ButtonHandler function in sonoff.ino to see if I can implement that myself and submit as a pull request, but that only seemed to mess stuff up as my Sonoff wasn't responding to any button event when I tried.

laszlojakab, 2018-05-26:
I'm almost looking for the same functionality. I would like to repeat "hold button" MQTT message until button is released to dim some lights. The MQTT message would be received by home automation system then a dimming command would be sent to a WIFI enabled LED controller. 
Is this possibly without modifying Tasmota source code?

Markusenz, 2018-06-19:
Also looking for that feature. tasmota shall report button hold time to mqtt so we can create different  actions depending on the button hold time. Would be awesome!

loxK, 2018-07-27:
I need same feature for my sonoffs touch. I need instant on/off on release and hold message to mqtt.

ascillato2, 2018-08-25:
Hi,

A hold mqtt message is already implemented.

Please, see wiki at: https://github.com/arendst/Sonoff-Tasmota/wiki/Control-other-devices

Hope this helps.

ascillato, 2018-08-27:
Hi,

There is a new example in rules with HOLD for buttons.

https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#13-toggle-a-relay-only-when-holding-the-button-for-2-seconds

Hope this is solves your request.

closed 2018-08-27 by ascillato2

Palloquin, 2019-02-11:
Hi,
I'm late to this party, but think I'm trying to achieve the same.
I would like to 
- press the button to toggle on/off
- long press the button and have my brightness (PWM controlled led) go up and down until I release the button

Based on this example ( [https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#16-using-an-external-button-with-single-press---double-press-and-hold](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#16-using-an-external-button-with-single-press---double-press-and-hold) ) I can get the switching and send a *single* mqtt command on hold.

I would need either repeated hold messages or a release message to be able to achieve something like this.

Anyone figured this out?

kylegordon, 2019-04-03:
I'm also looking for similar configuration for Sonoff Touch switches. I don't think Tasmota can do it unfortunately.

I really don't want the relay linked to the button presses at all, unless the device is offline. If it's online, I'd want single, double and hold (repeated) messages being sent as MQTT messages.

In the meantime, I've written my own firmware to support multitouch.

pickeld, 2019-05-06:
@kylegordon hey there. whould you please mind sharing your firmware with me?
@Palloquin im looking for the exact same thing. did you manage to achieve that?

Palloquin, 2019-05-06:
@pickeld : No, I dod not get that working... I dropped the project for a while. should you find a solution, I would love to hear back.

Perhaps Theo, or another wizz with the Tasmota firmware would be willing to have a look at adding this a an (optional) feature. It seems to make sense and there seem to be multiple users searching for something like it...

madivad, 2020-06-21:
dead links

AlexTransit, 2020-06-21:
> 
> 
> dead links

[tasmota.github.io/docs/](url)