arendst/Tasmota #1891: %prefix%/%topic%/ vs. /%prefix%/%topic%/

mawin01, 2018-02-12:
Hello,

I am using 5.11.1b on a sonoff basic device and a "sonoff" instance in iobroker. I run into the following issue:

- If i use %prefix%/%topic%/ as Full Topic the teledata/stat works fine and i am getting all the right information in iobroker, except of cmnd-Prefix, i didnt get the value of the command in iobroker .
- If i use **/**%prefix%/%topic%/ as Full Topic  the cmnd works fine, but teledata/stat does not.

It seems that cmnd-Prefix needs a "/" at the beginning,
e.g. _/cmnd/SONOFF_Basic_04/POWER1 = ON_
and tele-Prefix not, 
e.g. _tele/SONOFF_Basic_04/SENSOR = {"Time":"2018-02-13T00:35:39","Switch1":"OFF","AM2301":{"Temperature":23.5,"Humidity":44.4},"TempUnit":"C"}_

I am reading about Fallback-Feature, but my MQTTClientName was different to the Topic-Name.

Here is status 0, I hope it shows you all the important settings 

00:26:54 CMD: status 0
00:26:54 MQT: stat/SONOFF_Basic_04/STATUS = {"Status":{"Module":1,"FriendlyName":"Sonoff","Topic":"SONOFF_Basic_04","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:26:54 MQT: stat/SONOFF_Basic_04/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":0,"Sleep":0,"BootCount":37,"SaveCount":115,"SaveAddress":"F9000"}}
00:26:54 MQT: stat/SONOFF_Basic_04/STATUS2 = {"StatusFWR":{"Version":"5.11.1b","BuildDateTime":"2018-02-12T22:53:09","Boot":6,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
00:26:54 MQT: stat/SONOFF_Basic_04/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"mwrt04","SSId2":"","TelePeriod":10,"SetOption":"00000009"}}
00:26:54 MQT: stat/SONOFF_Basic_04/STATUS4 = {"StatusMEM":{"ProgramSize":518,"Free":484,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
00:26:54 MQT: stat/SONOFF_Basic_04/STATUS5 = {"StatusNET":{"Hostname":"SONOFF-Basic-04","IPAddress":"192.168.20.47","Gateway":"192.168.20.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.20.254","Mac":"68:C6:3A:A7:C3:6E","Webserver":2,"WifiConfig":4}}
00:26:54 MQT: stat/SONOFF_Basic_04/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.20.63","MqttPort":1883,"MqttClientMask":"SONOFF-Basic-04","MqttClient":"SONOFF-Basic-04","MqttUser":"sonoff","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:26:54 MQT: stat/SONOFF_Basic_04/STATUS7 = {"StatusTIM":{"UTC":"Mon Feb 12 23:26:54 2018","Local":"Tue Feb 13 00:26:54 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
00:26:54 MQT: stat/SONOFF_Basic_04/STATUS10 = {"StatusSNS":{"Time":"2018-02-13T00:26:54","Switch1":"ON","AM2301":{"Temperature":23.5,"Humidity":44.3},"TempUnit":"C"}}
00:26:54 MQT: stat/SONOFF_Basic_04/STATUS11 = {"StatusSTS":{"Time":"2018-02-13T00:26:54","Uptime":0,"Vcc":3.399,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mwrt04","RSSI":64,"APMac":"5C:49:79:FB:0D:C3"}}}
00:26:55 MQT: cmnd/SONOFF_Basic_04/POWER1 = OFF

Thank you very much in advance. I really hope that i not missed something in the documentation/forum.

regards
Markus

Ivoz, 2018-02-13:
I have a perhaps a similar problem, with the value 

`house/name/%topic%/%prefix%/`

I cannot pick up any responses on stat ( `house/name/device/stat/POWER` ) either.

mawin01, 2018-02-14:
Hi again,

spend some (the truth is a lot) time to find out that this problem is not with the tasmota firmware, it seems to be a problem with iobroker or the sonoff Adapter/Add on.

I have a workaround and for now I can live with that : 
I configure the Sonoff Device and when i am finished with that, I set a new device name in the MQTT Setting page. For both,  "Client" and "Topic" . Thats it. 

The last step produce a new entry in iobroker for the devie and all values are shown correctly.

 I learned that the leading slash in the Full Topic should be avoid, because of that  I use  %prefix%/%topic%/ . 

I hope this helps someone.

regards 
Markus

ascillato, 2018-06-02:
Thanks for share your solution. :+1:

closed 2018-06-02 by ascillato