arendst/Tasmota #1892: please help with getting the correct data value

handsomejackuk, 2018-02-13:
so i get this data reported out the tasmota console at intervals autmatically

01:08:26 MQT: tele/s1/SENSOR = {"Time":"2018-02-13T01:08:26","BMP180":{"Temperature":21.7,"Pressure":972.130,"SeaPressure":1003.337},"TempUnit":"C"}

can i directly put a command in the console that will bring out **only** the temperature value...
I try

stat/s1/SENSOR
tele/s1/SENSOR
stat/s1/BMP180
tele/s1/BMP180

and have been trying all other commands to just get the **only temperature** but just seem to fail to get this.. please help...

I would like to be able to get the temperature data on my android device.

I am using mqtt dashboard, and  have the relays to turn on and i get status of the relays / ws2812 leds with no problem, but cannot get the temperature, or any other single values to show

what am i doing wrong ?

davidelang, 2018-02-13:
no, there is no way to just get one value, you only get the sensor json output 
and need to parse it on the receiver.

jhlweb, 2018-02-13:
JSON has a couple of methods to get a value out of its string if u use a json parser.

I use  [http://www.jsonquerytool.com/](url) to validate a parse string a
To get the temperature use `$.BMP180.Temperature`  on the website validator

In youre case try using 

tele/s1/BMP180/Temperature

Jason2866, 2018-02-13:
I dont use mqqt Dashboard,

but since v1.9
- Support JsonPath to extract data from JSON payload ★
it should work....

What do i need to connect the Sensor BMP180/BMP280 to a Sonoff? If nothing needed which Pins are used?
I ordered some today from Ali.

THX

handsomejackuk, 2018-02-13:
First of all thanks for all the replies i will get there.... I have been doing a lot of research... 

what i am failing to understand though is some of the data in the State here   
 
tele/s1/STATE = {"Time":"2018-02-13T12:43:48","Uptime":"0T00:13:14","Vcc":3.232,"POWER1":"OFF","POWER2":"OFF","Dimmer":43,"Color":"6D0000","Scheme":4,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"xxxxxxxx","RSSI":88,"APMac":"xxxxxxxxx"}}

can be read back using the stat command in the console

so i enter stat/s1/dimmer and i get the result "on" 
so i enter stat/s1/color and get the value "6d00000"

is it because color,dimmer is defined somewhere else in the arduino sketch, and that sensor data is not ?

Jason2866, 2018-02-13:
You can get these (so far i know) only over json output 
(davidelang wrote this already)

handsomejackuk, 2018-02-13:
thanks Jason2866....

i can parse the string no problem with jsonquerytool, and get the result, 

Surely it cant be this hard to just display the temperature of my bmp180 on my android device in real time... using mqtt dashboard

phew...!!!!!!!

johoberm, 2018-02-13:
Hi handsomejackuk,

your welcome. 
It took me a while and some research what solution i use for my Home Automation System.
The easy way  i took (Domoticz) was a dead end because no working solution exists for my FHT Thermostates. But the Sonoff Switch Setup was so easy!
Finally for me is OpenHab with mosquitto mqtt broker installed on a PI3 the best solution.
As a beginner (i am not a programmer) it was a hard way to get even the simpliest things working.
But now, after understanding the Basics it is really fun, an with the possibilities to programm rules
in OpenHab (even for a non programmer, i "played" a little Basic in my young Years on a Sinclair ZX81)
and you get very easy a nice GUI in your Web browser.

handsomejackuk, 2018-02-13:
i currently installing openhab on my Raspberry Pi i dont know if it will run on my old model b and if i am wasting my time but its worth a shot...

same here zx81 and commodore pet, done a bit of cobol / pascal and visual basic but not really got into it in a big way been into computers and software / hardware for years mostly hardware though.. built a few quadcopters with gps, using arduino etc just getting in to home automation...

handsomejackuk, 2018-02-14:
sorted now.....

handsomejackuk, 2018-02-26:
can anyone please help me again, in nodered how can i get only the temperature data out of the json string and display it on a dashboard guage in nodered ? i see there a json node but it still will not give me the temperature value out on the debug window so the guage will not display the value.

Jason2866, 2018-02-27:
Try

Number Thermostat_CurrentTemp "Thermostat IST: Temperatur [%.1f °C]" 
    { mqtt="<[broker:tele/sonoff-5D3631/SENSOR:state:JSONPATH($.AM2301.Temperature)]" }

FryGuy1981, 2019-01-03:
`$.BMP180.Temperature`
For json filter string in MQTT dashboard to get just the temp.