arendst/Tasmota #1893: stat MQTT messages not sent on specified topic

Ivoz, 2018-02-13:
Hi. I have an s20 on 5.12.0

I have configured the Topic as `name` and the `house/room/%topic%/%prefix%/`

When I send a mqtt msg through the console, I the console tells me it sends a resulting `stat` message back on the same configured topic-

```
09:29:49 CMD: house/room/name/cmnd/POWER ON
09:29:49 MQT: house/room/name/stat/RESULT = {"POWER":"ON"}
09:29:49 MQT: house/room/name/stat/POWER = ON
```

Everything as expected.

However when I send a msg using `muisquitto_pub` on the RPi I'm running the MQTT broker on- `mosquitto_pub -u username -P password -t house/room/name/cmnd/POWER -m OFF`

Then the console tells me-

```
09:31:24 MQT: stat/name/RESULT = {"POWER":"OFF"}
09:31:24 MQT: stat/name/POWER = OFF
```

This is obviously **not** on the expected topic I thought was supposed to be setup, i.e `house/room/name/stat/POWER`. The same thing also happens if I use the Android app MQTT-dash to send a cmnd throught, the resulting `stat` message is published on this uncustomized channel.

Why these two different behaviours? When run through the web console, the reply `stat` uses the expected customized topic I set in configuration, when received from the broker, only the default? topic is used.

curzon01, 2018-02-13:
Can not be reproduced using already compiled binary downloaded from [Releases](https://github.com/arendst/Sonoff-Tasmota/releases):

Console:
```
CMD: Status 2
MQT: house/room/name/stat/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-09T14:54:51","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
CMD: Status 6
MQT: house/room/name/stat/STATUS6 = {"StatusMQT":{"MqttHost":"172.17.7.9","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_13B8C0","MqttUser":"mqttdev","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
CMD: Topic
MQT: house/room/name/stat/RESULT = {"Topic":"name"}
CMD: FullTopic
MQT: house/room/name/stat/RESULT = {"FullTopic":"house/room/%topic%/%prefix%/"}
```

Terminal 1:
```
mosquitto_sub -h 172.17.7.9 -u user -P ****** -t '#' -v
```

Terminal 2:
```
mosquitto_pub -h 172.17.7.9 -u user -P ****** -t 'house/room/name/cmnd/POWER' -m OFF
```

Terminal 1:
```
house/room/name/stat/RESULT {"POWER":"OFF"}
house/room/name/stat/POWER OFF
```

Console:
```
MQT: house/room/name/stat/RESULT = {"POWER":"OFF"}
MQT: house/room/name/stat/POWER = OFF
```

as expected.

I tried two for example, [sonoff.bin](https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.12.0/sonoff.bin) and [sonoff-DE.bin](https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.12.0/sonoff-DE.bin), both are working.

Hint: If you have compiled it by yourself, try one of the [Releases binaries](https://github.com/arendst/Sonoff-Tasmota/releases) first.

Ivoz, 2018-02-13:
Yes, it was OTA flashed with binaries.

I have figured out an additional critical condition of reproducing, that is the `Client` field should be the same as the `Topic`.

In my case I am using unique names to my devices, which I would therefore assume would be fine reproduced as the ClientID for mqtt, and also as the device's topic.

[image of mqtt settings](https://i.imgur.com/wl10Ttc.png)

[console](https://i.imgur.com/mgIVWit.png)

curzon01, 2018-02-14:
Yes, reproducible:

```
CMD: MqttClient
MQT: house/room/name/stat/RESULT = {"MqttClient":"name"}
CMD: Topic
MQT: house/room/name/stat/RESULT = {"Topic":"name"}
CMD: FullTopic
MQT: house/room/name/stat/RESULT = {"FullTopic":"house/room/%topic%/%prefix%/"}
```
Console:
```
MQT: stat/name/RESULT = {"POWER":"OFF"}
MQT: stat/name/POWER = OFF
```

If did also a test with changed FullTopic like `myhome/%prefix%/tasmota/%topic%/` and other commands like STATUS 1 - it happens the same. The anser is alwys

### Conclusion - wrong MQTT response
MQTT `stat` reply with

`stat/<MqttClient>/<STATUS>`

by following conditions:
1. `<MqttClient>` is a substring of `Topic`, e.g.
```
Topic       MqttClient
mydevice    device
mydevice    mydev
mydevice    dev
```
2. command is triggered by MQTT cmnd

The wrong MQTT response does not occur when using web interface button or button switch.