arendst/Tasmota #1896: Water Meter Config Sonoff basic

miley786, 2018-02-13:
Hi 

We have been using the the sonoff basic for a few configurations with it and the Tasmota firmware, at this point we are trying to implement a water meter sensor on the sonoff using one of the GPIO's to act as the input for the readings.

We have a project running with the water meter working on a Arduino UNO running the code below:
`
byte` statusLed    = 13;
byte sensorInterrupt = 0;
byte sensorPin       = 2;

float calibrationFactor = 4.5; // Change depending on sensor
volatile byte pulseCount;
float flowRate;
unsigned int flowMilliLitres;
unsigned long totalMilliLitres;
unsigned long oldTime;

void setup()
{
  Serial.begin(9600);  // Initialize a serial connection for reporting values to the host
  pinMode(statusLed, OUTPUT);  // Set up the status LED line as an output
  digitalWrite(statusLed, HIGH);  // We have an active-low LED attached

  pinMode(sensorPin, INPUT);
  digitalWrite(sensorPin, HIGH);

  pulseCount        = 0;
  flowRate          = 0.0;
  flowMilliLitres   = 0;
  totalMilliLitres  = 0;
  oldTime           = 0;

  // The Hall-effect sensor is connected to pin 2 which uses interrupt 0.
  // Configured to trigger on a FALLING state change (transition from HIGH
  // state to LOW state)
  attachInterrupt(sensorInterrupt, pulseCounter, FALLING);
}

void loop()
{
  if ((millis() - oldTime) > 1000)   // Only process counters once per second
  {
    // Disable the interrupt while calculating flow rate and sending the value to
    // the host
    detachInterrupt(sensorInterrupt);

    // Because this loop may not complete in exactly 1 second intervals we calculate
    // the number of milliseconds that have passed since the last execution and use
    // that to scale the output. We also apply the calibrationFactor to scale the output
    // based on the number of pulses per second per units of measure (litres/minute in
    // this case) coming from the sensor.
    flowRate = ((1000.0 / (millis() - oldTime)) * pulseCount) / calibrationFactor;

    // Note the time this processing pass was executed. Note that because we've
    // disabled interrupts the millis() function won't actually be incrementing right
    // at this point, but it will still return the value it was set to just before
    // interrupts went away.
    oldTime = millis();

    // Divide the flow rate in litres/minute by 60 to determine how many litres have
    // passed through the sensor in this 1 second interval, then multiply by 1000 to
    // convert to millilitres.
    flowMilliLitres = (flowRate / 60) * 1000;

    // Add the millilitres passed in this second to the cumulative total
    totalMilliLitres += flowMilliLitres;

    unsigned int frac;

    // Print the flow rate for this second in litres / minute
    Serial.print("Flow rate: ");
    Serial.print(int(flowRate));  // Print the integer part of the variable
    Serial.print("L/min");
    Serial.print("\t"); 		  // Print tab space

    // Print the cumulative total of litres flowed since starting
    Serial.print("Output Liquid Quantity: ");
    Serial.print(totalMilliLitres);
    Serial.println("mL");
    Serial.print("\t"); 		  // Print tab space
    Serial.print(totalMilliLitres / 1000);
    Serial.print("L");

    // Reset the pulse counter so we can start incrementing again
    pulseCount = 0;

    // Enable the interrupt again now that we've finished sending output
    attachInterrupt(sensorInterrupt, pulseCounter, FALLING);
  }
}

void pulseCounter()
{
  // Increment the pulse counter
  pulseCount++;
}
`
Is it possible to modify this code and include this in the tasmota firmware? and list it as a water sensor? since the MQTT configurations are already running on it? it would allow for a simple integration for a water meter to an automation system

Please let me know if its possible.

Thanking in advance

arendst, 2018-02-13:
Connect the sensor to a GPIO, select a counter in Tasmota and see what happens.

Your home automation tool can probably make some sense out of the counter value it can receive.

miley786, 2018-02-13:
Running tests using home assistant, with the sensor connected to GPIO 14 on sonoff, at this point I can’t find any logic in HASS to make sense of the info.

Have tested with both counter and pulse timer settings,,,, can’t seem to get the logic working. That’s why I was hopping that the so off could publish the results to be stored in home assistant directly.

Doubleho7, 2018-02-19:
I've experienced similar, has there been any updates on implementing/adding a water meter or identifying an accurate way to measure?

EricVdB, 2018-05-07:
I'm interested in this as well. I have this setup for both my water and gas meter using 2 hall effect sensors.
They are both connected to an ESP32 which also has the PZEM004 for power consumption.
I would like to move completely to Tasmota and get rid of all the other stuff.
Plan is to have an ESP8266 for each (electricity, water and gas).

Besides that, I use Tasmota to measure the temps of my pool solar panels and the pool, and control the pump with another Tasmota through OpenHab. Reporting is done with Grafana.

I hope we see soon a decent counter implementation that reports counter, L/min, today consumption, yesterdays consumption, total consumption like is reported by the PZEM004 module.

Thanks in advance,
Eric

EricVdB, 2018-06-06:
no one ?

miley786, 2018-06-06:
Hi would you be able to share your config for homeassistant? Trying to implement reports that displays counter, L/min, today consumption, yesterdays consumption, total consumption 

Assistance much appreciated

ascillato, 2018-06-06:
Yes, of course

```yaml
sensor:
  - platform: mqtt
    name: "Watering Flow"
    state_topic: "tele/sonoff/SENSOR"
    value_template: "{{ ( value_json['COUNTER'].C1 | multiply(0.00222) | float ) | round(2) }}"
    unit_of_measurement: "l/min"    
```

ascillato, 2018-06-13:
Hi

If your issue was solved, please close it. Thanks :+1:

kajmaj, 2018-06-24:
Hello [@EricVdB](url), coud you be so kind to share your rule to reset counter each tepeleriod?
Thanks

ascillato2, 2018-06-27:
Hi,

For reseting a counter every teleperiod you can use a rule like:

type on the console:

```
rule 1
rule on tele-counter#c1>0 do counter1 0 endon
```

If your issue is not solved, please, reopen it. Thanks.

closed 2018-06-27 by ascillato2

thepotoo, 2019-09-14:
I believe this rule should be `ON tele-counter#c1>0 DO counter1 0 ENDON`
But it is crashing in current releases when the counter gets too high.

gmikulski, 2020-09-06:
> Hi @EricVdB ,
> 
> Tasmota already has a counter. I'm using that for water flow (also hall effect flow sensor) on my garden watering. My sonoff counts the pulses and every teleperiod I set a rule to reset the counter. I set the teleperiod to be 1 minute, so in my home automation software (home assistant) I receive by mqtt pulses/min that there I translate to flow (l/min)
> 
> Hope this helps

Hi @ascillato / @ascillato2 

I was hoping to do similar thing with SONOFF BASICZBR3 (Zigbee DIY Smart Switch) + YF-S201 Hall Effect Water Flow Meter / Sensor + HomeAssistant. How should I connect the sensor with zigbee switch?

The sensor comes with three wires: red (5-24VDC power), black (ground) and yellow (Hall effect pulse output).

I found on home assistant page that someone took the 5V power by soldering to one of pins on bottom of switch. But what about black and yellow wires? Can you advice what is your setup?

ascillato, 2020-09-06:
It is the exact same thing as explained above. Remember that the power regulator of a sonoff basic is designed only to power itself. It is more reliable to use a separated power source for your sensor.

lalo-uy, 2020-09-06:
Black goes to ground.
Yellow to the Counter gpio.
Red need 5v, not 3.3v. I take it fom the input of the lm1111 regulator y
the bottom of the board.