arendst/Tasmota #1898: Homeassistant issue

bachoo786, 2018-02-13:
Hi

I have setup the code but struggle with the ds18b20  sensor with my HA. I have gone through the guide 5 times and still fail.

Can someone please share their mqtt topic for ds18b20 and any pre requisite's that I require to perform for the ds18b20 and HA?

Thanks a lot.

pascalsaul, 2018-02-13:
This will do the trick:
```
- sensor
  - platform: mqtt
    state_topic: "tele/dual_sw/SENSOR"   
    name: "dual_sw_sensor_ds18b20_1"
    unit_of_measurement: "°C"
    value_template: '{{ value_json["DS18B20-1"]["Temperature"] }}'
```

miley786, 2018-02-13:
Hope this helps, seems to be easy config for the sensor, tested this config and works:

- platform: mqtt
    name: "Water Temperature"
    state_topic: "tele/temp2/SENSOR"
    value_template: "{{ value_json.DS18B20.Temperature }}"
    unit_of_measurement: "°C"
    force_update: true

bachoo786, 2018-02-13:
thanks @pascalsaul @miley786 I was wondering why is the state topic for both of you different? shouldn’t mine be different to or? I am a complete NOOB !!

pascalsaul, 2018-02-13:
They are the same for both of us...

tele/<device name on mqtt>/SENSOR

bachoo786, 2018-02-13:
`#device name on mqtt#` where do I find mine? is this the MQTT Client name?

pascalsaul, 2018-02-13:
[image]

bachoo786, 2018-02-13:
@pascalsaul thanks buddy I have same as yours i.e. client name and topic.

I just wanted to know what did you do before you flashed? 

did you uncomment 1). or 2).

1). #define USE_DS18x20                              
2). #define USE_DS18x20_LEGACY 

Do I have to do this?

Thanks again and sorry for the trouble

pascalsaul, 2018-02-13:
I'm using: define USE_DS18x20

bachoo786, 2018-02-13:
@pascalsaul did you have to add any pull up resistors between data and +5V?

also did you have to specify the sonoff module in the webserver? 

I have not connected my sonoff th16 to the mains yet and trying to debug it from the computer via usb

miley786, 2018-02-13:
Yes you need a 4.7k resistor between the 3v and data on sensor 

Not sure what you referring to? How are you trying to use the sonoff?

bachoo786, 2018-02-13:
so i done this without any resistor and it starts reporting the temperature.

[image]
[image]

miley786, 2018-02-13:
I have tested with our resistor and couldn’t get it working on 3.3v,,, Can read the explanation on this link might shed light as to why you need to use one.

https://arduino.stackexchange.com/questions/30822/the-use-of-4-7kohm-resistor-with-ds18b20-temperature-sensor

bachoo786, 2018-02-13:
@miley786 did you specify the gpro14 pi to ds18b20 in the webserver?

miley786, 2018-02-13:
*without

miley786, 2018-02-13:
what setting are you referring to as the web server? 

Only reference to the sensor selection is done under “configure module” and mine currently set on gpio 14

bachoo786, 2018-02-13:
This is what I have under gpio 14 as ds18x20. Is that what you have?

How about gpio 1,3,4? 

Surprisingly it works for me without the resistor. Anyway where have you connected your resistor? Like towards to the 2.5mm pin or near the sensor?

And finally why is it that the temperature readings on the  webserver where you changed your gpio settings is quicker then home assistant?  The temperature change in home assistant is every 15mins where's as on the weber very it's every almost 5 seconds. Anyway to make the temperature readings in home assistant better?

miley786, 2018-02-13:
Okay understand what u mean by web server, depending on where you have connected the sensor data wire on the sonoff, if it’s connected to pin hole next to pins used for firmware upload then that would be GPIO14 

You can increase you telemetry period under “config logging” to a desired time, currently it should be set to 300 sec by default, also would need to specify force_update: true on your mqtt config as shown above

The resistor can be connected anywhere Between the sensor and sonoff, no specific distance, just needs to be between 3v and data wire running from sonoff to sensor

bachoo786, 2018-02-13:
@miley786 I think my sonoff th16 is different to yours mine is this one: https://loads.pickle.me.uk/images/sonoff-header.jpg

regardless I have set it to gpio 14 as ds18b20 and it works.

I now need to solder the 4.7kohm resistor between data and vcc.

I will change the mqtt settings and see if the temperature logging is any quicker on HA..

bachoo786, 2018-02-13:
@miley786 where do I specify "specify force_update: true" on my MQTT config?

miley786, 2018-02-13:
Ah sorry missed the part where you said you using the TH16 I thought it was a basic, my mistake 🙈, but if u getting the readings then u in the right track,,,,  you will need to change the telemetry period on the web server as well as update your mqtt config on home assistant to increase readings in Home assistant

bachoo786, 2018-02-13:
I have changes the telemetry period to 10 seconds and it'd much better..

Force update:true is on home assistant?

bachoo786, 2018-02-13:
thanks buddy will update the config

miley786, 2018-02-13:
The 4.7k resistor must be connected between the voltage input wire and the data wire on the sensor

So the wire from gpio14 and the 3.3v wire to the sensor, the resistor can be connected anywhere in between these two wires from the sonoff to the sensor as per image below:

https://goo.gl/images/z6g7uh

bachoo786, 2018-02-14:
Hey @miley786 
I noticed HA doesn't report the state of the switch if it was restarted. Do you have any solution to that?

miley786, 2018-02-14:
you will have to add a binary sensor for the device you looking to monitor status eg:

binary_sensor:
  - platform: mqtt
    name: "Temp Switch Monitor"
    state_topic: "tele/#topic name on mqtt#/LWT"
    value_template: "{{ value }}"
    payload_on: "Online"
    payload_off: "Offline"
    device_class: connectivity

bachoo786, 2018-02-14:
@miley786 I resolved the power issue by going into ‘Console’ in the Sonoff’s web UI, typing ‘PowerRetain 1’ and enter.

this remembers the state and even changes after restarting HA.