arendst/Tasmota #1899: Flashing MagicHome LED strip controller

jannnfe, 2018-02-13:
Hello,
I am having problems to flash the ESP-M2 based MagicHome LED strip controller.
I have this one:
[image]

I followed the instructions in [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-LED-strip-controller) but with no success.
I have soldered the 4 cables (yellow) directly to the chip as seen in the picture and have followed the steps.
What I have noticed is that the chip gets very hot when it is connected via 12v even if nothing else is connected. is that normal?
Thanks!

oxcxsxixbxaxn, 2018-02-13:
What problems? Problem flashing or after flashing?

 I don't know how normal it is but mine also gets very toastie...

jannnfe, 2018-02-13:
Sorry I should have been more specific. I do not get the Tasmota firmware flashed.
I have all settings like my other Sonoff devices in Arduino IDE but I get this error:
```
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
```
Maybe someone has an idea what I can test to flash the firmware.
Thanks!

oxcxsxixbxaxn, 2018-02-13:
I used esptool on a raspberry pi because my usb serial doesn't work well on Win10.
Did you check the GND connection, particularly where it connects to GPIO-0 to put it in flash mode?

jannnfe, 2018-02-15:
Sorry for my Late response. So i connect GND and GPIO-0 to ground of the FTDI connector, RX to TX and TX to RX and last 12v to Power Supply and Start flashing using esptool?
On my Module i have to solder directly on the Chip? Or there are other pins or pads available for Connection?
Thanks!

oxcxsxixbxaxn, 2018-02-16:
If u have the same board as me, yes solder to the chip, otherwise use the pads available...
connect GND, tx, rx, gpio-0 to GND and power up,
Back up with esptool and disconnect power, wait a few seconds
Reconnect power and Erase flash then disconnect power
Reconnect power and flash f/w, when complete remove power, remove tx/rx/gpio-0 then GND
If you used a custom firmware with your own wifi AP details built in, you are off and running...
If using the stock tasmota firmware...  dunno...