arendst/Tasmota #1903: Tasmota Interface Not Coming Up

boilermanc, 2018-02-13:
Hi all, I have flashed a bunch of these over time and have not run into this issue until now. 

I flashed a POW with the latest build 5.12. I put in my IP address because I am on a 10. network. It ran through the flash just fine. Its on mains and I scanned the network and its connected. I cannot get the web interface to come up though. It just spins. Never had that happen before.

thanks!

boilermanc, 2018-02-14:
Tried to flash a tH-16 today and now I get no connection at all. Bad to worse. Everything loads up to the board just fine. This is what I have in my config:
`
// -- Wifi ----------------------------------------
#define WIFI_IP_ADDRESS        "10.1.10.42"         // [IpAddress1] Set to 0.0.0.0 for using DHCP or IP address
#define WIFI_GATEWAY           "10.1.10.1"   // [IpAddress2] If not using DHCP set Gateway IP address
#define WIFI_SUBNETMASK        "255.255.255.0"   // [IpAddress3] If not using DHCP set Network mask
#define WIFI_DNS               "10.1.10.1"    // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)

#define STA_SSID1              "myssid"      // [Ssid1] Wifi SSID
#define STA_PASS1              "mypassword"  // [Password1] Wifi password`

these are the same credentials i have put in in the past with no issue. 

I have the wifi_config_tool set to wifi_wpsconfig.

am i missing something? 

after it flashes i put it on mains and it will not find my network. i have both the blue and red light on on the th-16.

thanks!

Jason2866, 2018-02-15:
Maybe not enough free memory. Disable not needed things (DOMOTICZ, mDNS, HASS support) in the source.

arendst, 2018-02-15:
Take a look at this #1900

boilermanc, 2018-02-15:
Hey Jason, not sure how to disable in source. sorry.

Arendst. I looked at that issue. I tried other libraires all the way back to 5.91. None of them would let me connect. They load up ok. 

But now i am stuck again. anything i try to load just pulls mem errors so nothing is loading. even tried a blank sketch...

Jason2866, 2018-02-15:
Mhmm, looks that you have general problems....
Try maddox universal build:
http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin
You have to config all your settings manually....