arendst/Tasmota #1904: sonoff 4ch does not work after using wall switch

kbaras, 2018-02-14:
Hi,

I have flashed tasmota on my sonoff 4ch pro. There was also toggle switch configured for gpio.
It worked until I did not connect the wall switch. Then it stopped working.
Connecting to the device via serial comm writes constantly random characters to output without pause.

What could be this issue? Thank you!

savage872, 2018-02-15:
serial console, it's probably the baud rate.
I have a ch4 pro and my wall swiches connect into the relays not on the any gpio's

Can you give more details?

kbaras, 2018-02-22:
If I print the output, I is printing endlessly random chars, like: 
␀␀y␀␀␀Q�␀ ␀��␀␀X␀␀0�␀␀␀T␀␀␄��␀␀␔�␀␀␀␙�␀␀␀@�␀␀␔�␀␀␀��␀␁␀␂Q]␀␐␀AT␀ ␀␀8␀␀�|␀␀␀A}␀␀␀�x�␀@␀␄␝␀␀�}␀␀␀␀��␀␀␀␀␘␡␀␀␀␐�␀@␀
�␀␀␀A]␀␀␀AX␀ ␀@\␀@␀␂␐�␀0␀␂�␀␀�␀␀␀���␀␀␀E}␀␀␀␀�␐�␀␀␀␀␀��␀␀␀␀␄␐�␀␀␀}␀␀␀@t�␀␀␀�␀␀␀␀0�␀␀␀␅�␀␀␀␁d␀␀␐␐ �␀␀␀�␔␀␀␀␀0�␀␀␀␀␐␄�␀␀␐�␀␀␀W␀␀␀��␀␀␀
}␀␀␐Q�␀␀␀�X␀␀␀␂p�␀␀␀A<␀␀�␀␀␀␂␐_␀␀␐␀A�␀␀␀�<␀␀y␀␀␀�␀␀␀␀␄Q~␀␀␁Pu␀␀<␀@␀X␀A}␀␀␀␀␀�␀␀␀�␐T␀␃␀␀␀␀␀}␀␀␀␐�␀␀(W␀�␀@␂p�␀␀␄�␀@␀q�␀␀ ␀␀P␀␀␀��␀␀␀@P
␀␀␐␐}␀␀␀␀Q␀␀␃P_␀␀␐��␀␀␀␁8�␀␀␀A]

I cannot add any instruction via serial (erase, or flash, or etc.). Also does not react to push buttons. The first two leds are red. 
Baud is 115200. Earlier it worked until I tried to connect the wall switch to one of the gpios.