arendst/Tasmota #1905: 433 Bridge MQTT problem

Teccno, 2018-02-14:
I have a problem and hope that you can help me.
If i put RFCode #554551 in the console, the bridge (Tasmota 5.11.1) does the job perfectly. Now i am lookin for the MQTT code for exact that string. I've tried i.e. cmnd/sonoff_bridge/POWER/RFCode #554551 or .../RFCode:#554551 or thousands of other strings, i always get {"Command":"Unknown"}. What am i doing wrong?
Thank you very much for your help.

Ducky123455, 2018-02-26:
I'm running into the exact same problem. i'm running 5.10.0. I'm pretty sure it worked before on this version. 

Can't see anything in the console either if I set debug max.
I'm getting this in mosquitto mqtt server:

cmnd/sonoffbridge/RESULT rfcode
stat/sonoffbridge/RESULT {"Command":"Unknown"}

klein0r, 2018-03-21:
You have to send RfCode instead of Power. Power is a command, RfCode is a Command, too.

In your case you should try cmnd/sonoff_bridge/RfCode (depends on mqtt topic configuration).

Ducky123455, 2018-03-21:
Ahh you are right... feeling slightly dumb now...

For future reference, this worked for me:
`mosquitto_pub -t cmnd/sonoffbridge/RfCode -m "#12345"`

or for those using home assistant :)
```
 script8:
   alias: turn off alarm
   sequence:
   - service: mqtt.publish
     data_template:
       topic: cmnd/sonoffbridge/RfCode
       payload: "#12345"
```

Thanks a lot for your help. This firmware is truly a piece of art!