arendst/Tasmota #1906: Can not turn off MQTT

RonGroen, 2018-02-14:
First all the complements!!

I want to turn MQTT off on compile time (with arduino ide)
and used // -- MQTT ----------------------------------------
#define MQTT_USE               0                 // [SetOption3] Select default MQTT use (0 = Off, 1 = On)
But I am getting
00:00:55 MQT: Verbindungsversuch...
(changing language in the same file works fine).

Can you please give me a hint?

With kind regards
Ron Groen

curzon01, 2018-02-14:
May be you have previously used a self compiled firmwate with define MQTT_USE 1

"Reset Configuration" (then the new define from your source will be used) or simply use "SetOption3".

RonGroen, 2018-02-14:
Thank you very much for you fast reply.

 "Reset configuration" with button longer then 4 sec worked. But how do I apply "SetOption3"?? 
Port 80 on the IP does not respond and on ttyACM0 (arduino) gives loglines but does not respond on input.

curzon01, 2018-02-14:
> Port 80 on the IP does not respond

?
To exlude side fx from your own source compilation try same procedure with an already compiled binary from `[Releases](https://github.com/arendst/Sonoff-Tasmota/releases)` page - the web page is your main entry.

RonGroen, 2018-02-14:
I used the English binary, give it a full reset and got:
0:00:00 CFG: Use defaults
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_10E297, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP1 indebuurt1 in mode 11N as sonoff-0663...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP2 indebuurt2 in mode 11N as sonoff-0663...
00:00:09 RSL: stat/sonoff/RESULT = {"WifiConfig":"WifiManager selected"}
00:00:10 WIF: WifiManageractive for 3 minutes
00:00:11 HTP: Web server active on sonoff-0663 with IP address 192.168.4.1
00:00:39 WIF: Hostname %s-%04d, SSId1 AHA2a, Password1 correctPW, SSId2 indebuurt2, Password2 VnsqrtnrsddbrN
00:00:40 APP: Restarting

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v401052cc
~ld
⸮
00:00:00 CFG: Use defaults
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_10E297, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP1 indebuurt1 in mode 11N as sonoff-0663...

It forgets the settings...

curzon01, 2018-02-14:
So your hardware may have a problem (flash?)

RonGroen, 2018-02-14:
I used the wrong esptool. with esp2.2.1 and your binary I can activate wifimanager, write to eeprom and reboot and connect. with a local ip number. I tried to connect with my ethernet connection and nothing works. But on my phone, every works fine.  thank you very much.

RonGroen, 2018-02-14:
Is there a location where you block ethernet wired pc´s?

closed 2018-02-14 by RonGroen