arendst/Tasmota #1908: Feature Request: Split RGB from dedicated White in SetOption15 Mode

oxcxsxixbxaxn, 2018-02-14:
I have a bunch of these Magichome v2.3 boards now and a lot of RGBW strips of LEDs
The strips are alternating RGB LED and Warm White LED with the dedicated White on PWM4

The problem is that in SetOption15 mode, the fixed color commands (1-12) are only about RGB channels.  The dedicated white can only be enabled by sending the whole color hex for all channels RGB and W, eg #RRGGBBWW.  I would prefer to be able to control the dedicated White separately from RGB LEDs, with their own Dimmer levels and schemes, etc... leaving the color channels doing what they are doing while turning on/off/dimming/running schemes, etc

I can control the dedicated White independently in PWM mode, but then lose all of the dimmer, color and other functions, meaning I have to try to implement all this in Mqtt.

I'm happy to poke around in the code and spend some time on this.  Are there any pointers/recommended methods to implement something like this?  

Cheers

oxcxsxixbxaxn, 2018-02-19:
Is this the correct place for feature requests?  Are Feature Requests being considered at this time?

timota, 2018-03-02:
Agree, it will be a great feature. 

I found similar feature in ESPurna firmware. From HASS we can control white channel as well as RGB.
HASS can control it like:

    white_value: true
    white_value_command_topic: '/test/light/channel/3/set'
    white_value_state_topic: '/test/light/channel/3'

As i have so many similar devices to MagicHome and AriLux with RGBWW channels, im ready to donate if this is case to implement. 

@arendst - please let me know if you will be happy to implement it, also, let me know how i can donate.

Thanks.

oxcxsxixbxaxn, 2018-03-15:
@timota, thanks for the mention of ESPurna, firmware, I hadn't looked outside of tasmota for a solution..  I just flashed ESPurna to one of my devices and it works like I need and seems to be generally more functional too.  
What is your reason for looking at tasmota over ESPurna?  
It doesn't look like there is any interest with tasmota devs for the feature/hardware...