arendst/Tasmota #1910: Sonoff POW Sensor setup

EXTRAMINER, 2018-02-14:
Hi, i've recently setup the sonoff pow with the tasmota firmware.
The switch in hass works, but i don't know to setup the sensor that show the current,energy,voltage and power.
Can anyone help me?
Thank you

xbmcnut, 2018-02-14:
https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant#periodical-updates-1

EXTRAMINER, 2018-02-14:
The example doesnt work

xbmcnut, 2018-02-14:
Believe you may need to change 'ENERGY' to 'SENSOR' now though?

xbmcnut, 2018-02-14:
I'm going to submit a change request to the Wiki.

EXTRAMINER, 2018-02-14:
nope it doesnt work again 
but i found the problem the new tele message is like this : 21:51:31 MQT: 

> tele/sonoff-pow/SENSOR

 = {"Time":"2018-02-14T21:51:31","ENERGY":{"Total":0.984,"Yesterday":0.000,"Today":0.984,"Period":12,"Power":145,"Factor":0.90,"Voltage":220,"Current":0.731}}
instead of this: `tele/sonoff/ENERGY` = {"Time":"2018-01-10T19:35:58","Total":2.491,"Yesterday":0.088,"Today":0.000,"Period":0,"Power":0,"Factor":0.00,"Voltage":117,"Current":0.000}`

EXTRAMINER, 2018-02-14:
if can help put this in the wiki

xbmcnut, 2018-02-14:
Wiki updated.

reloxx13, 2018-02-15:
Everyone with a github account can update the wiki :)

xbmcnut, 2018-02-15:
Yep, know that.

reloxx13, 2018-02-15:
haha so i did not say that to you :D

xbmcnut, 2018-02-15:
:)

EXTRAMINER, 2018-02-15:
Ah thank you, I didn't know that

RaymondMouthaan, 2018-02-15:
Changed here https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Pow as well.

EXTRAMINER, 2018-02-15:
Ok good job

RaymondMouthaan, 2018-02-15:
I guess this issue can be closed ?

EXTRAMINER, 2018-02-15:
Yes, thanks

EXTRAMINER, 2018-02-19:
The issue can be closed