arendst/Tasmota #1913: Device Discovery

Schnittcher, 2018-02-14:
Is there a way to find all Tasmota devices on the network and get the Topic?

ascillato, 2018-02-14:
Nice Idea, I ask for a feature like that for SonWEB

reloxx13, 2018-02-15:
German SH Script can be found here: http://forum.creationx.de/forum/index.php?thread/503-soncheck2-tasmota-ger%C3%A4te-ohne-webserver-finden-und-konfigurieren/

Todo for SonWEB :p

emontnemery, 2018-02-15:
You can also easily find all devices by enabling MQTT discovery:
Either setoption19 1, or set `#define HOME_ASSISTANT_DISCOVERY_ENABLE 1` in `user_config_override.h`
You will now get a list of all tasmota devices by subscribing to MQTT topic `homeassistant/#`

ascillato, 2018-02-19:
Device Discovery is now available on SonWEB. :+1:

closed 2018-06-02 by ascillato