arendst/Tasmota #1915: Help on getting state example for different devices model

luisfpinto, 2018-02-14:
Hello I'm working on creating an app that integrates a communication with Sonoff Tasmota devices. Right now I have a sonoff basic output doing a get request that looks like this:

```http://ipaddress/cm?cmnd=status%200```

```
{
    "Status": {
        "Module": 1,
        "FriendlyName": "Sonoff",
        "Topic": "sonoff",
        "ButtonTopic": "0",
        "Power": 0,
        "PowerOnState": 3,
        "LedState": 1,
        "SaveData": 1,
        "SaveState": 1,
        "ButtonRetain": 0,
        "PowerRetain": 0
    },
    "StatusPRM": {
        "Baudrate": 115200,
        "GroupTopic": "sonoffs",
        "OtaUrl": "http://domus1:80/api/arduino/sonoff.ino.bin",
        "Uptime": "0T00:00:49",
        "Sleep": 0,
        "BootCount": 28,
        "SaveCount": 78,
        "SaveAddress": "F6000"
    },
    "StatusFWR": {
        "Version": "5.12.0",
        "BuildDateTime": "2018-02-14T17:22:04",
        "Boot": 6,
        "Core": "2_4_0",
        "SDK": "2.1.0(deb1901)"
    },
    "StatusLOG": {
        "SerialLog": 2,
        "WebLog": 2,
        "SysLog": 0,
        "LogHost": "domus1",
        "LogPort": 514,
        "SSId1": "Obi Lan Kenobi",
        "SSId2": "indebuurt2",
        "TelePeriod": 300,
        "SetOption": "00000009"
    },
    "StatusMEM": {
        "ProgramSize": 493,
        "Free": 508,
        "Heap": 23,
        "ProgramFlashSize": 1024,
        "FlashSize": 1024,
        "FlashMode": 3
    },
    "StatusNET": {
        "Hostname": "sonoff-2236",
        "IPAddress": "192.168.1.139",
        "Gateway": "192.168.1.1",
        "Subnetmask": "255.255.255.0",
        "DNSServer": "192.168.1.1",
        "Mac": "XXXXX",
        "Webserver": 2,
        "WifiConfig": 3
    },
    "StatusMQT": {
        "MqttHost": "domus1",
        "MqttPort": 1883,
        "MqttClientMask": "DVES_%06X",
        "MqttClient": "DVES_6728BC",
        "MqttUser": "DVES_USER",
        "MAX_PACKET_SIZE": 1000,
        "KEEPALIVE": 15
    },
    "StatusTIM": {
        "UTC": "Wed Feb 14 20:48:44 2018",
        "Local": "Wed Feb 14 21:48:44 2018",
        "StartDST": "Sun Mar 25 02:00:00 2018",
        "EndDST": "Sun Oct 28 03:00:00 2018",
        "Timezone": 1
    },
    "StatusSNS": {
        "Time": "2018-02-14T21:48:44"
    },
    "StatusSTS": {
        "Time": "2018-02-14T21:48:44",
        "Uptime": "0T00:00:49",
        "Vcc": 3.423,
        "POWER": "OFF",
        "Wifi": {
            "AP": 1,
            "SSId": "XXXXi",
            "RSSI": 100,
            "APMac": "XXXX"
        }
    }
}
```

Are you able to provide me more output examples for the different sonoff devices? So then I can try to integrate them without waiting for them to arrive.

Thank you!

reloxx13, 2018-02-15:
Heya,

JSON Response is generally the same for all.

There are diffrents for POW and Sensores.

Also "StatusMQT" is only avaible if MQTT is enabled (be sure to "if" it).

here you go, pls add new data if you got sth.
https://github.com/arendst/Sonoff-Tasmota/wiki/JSON-Status-Responses

luisfpinto, 2018-02-15:
Woah! Thank you so much for this!

closed 2018-06-14 by luisfpinto