arendst/Tasmota #1916: Another Board Layout RF Bridge

michaelmassee, 2018-02-14:
Today i get my "new" Bridge, successfully flashed with help of the good work here. Thanks for that.

I want to share my experience

What i did differnt from the description:
1. The Box is Screwed together.
[image]

2. To get to the Switch, just bend slightly the large Led on the Top
[image]

3. Then Flash it from Bottom Side

[image]

regards

emontnemery, 2018-02-15:
Which RF bridge is this?

michaelmassee, 2018-02-15:
The  Sonoff  433 RF Bridge
[https://www.itead.cc/sonoff-rf-bridge-433.html](https://www.itead.cc/sonoff-rf-bridge-433.html)

mapsrof, 2018-02-16:
Also got a new board design (2017-11-23) and it's working like Michael described!

arendst, 2018-02-18:
Confirmed. Add this issue as link to Wiki > Sonoff RF Bridge 433.

closed 2018-02-18 by arendst

claud9999, 2018-02-27:
I also received this board today.

Yes you have to pull out the rubber feet to get to the screws. I'm able to power the board but not flash the board using esptool.py, I've flashed a number of other Sonoff devices so it's not a computer config issue.

I've tried swapping RX/TX, pressing and holding the "peer" button, and changing S2 to "off", all to no avail.

More detailed instructions would be appreciated!

PeterEmbedded, 2018-03-06:
@claud9999 
- Set the switch to OFF
- Remove power from the device (3V3)
- Press and HOLD the button
- Power on the device
- Wait +/- 1 second
- Release the button
- Device will now be in firmware download mode, use your IDE to upload the new firmware as usual

newingit18, 2018-04-19:
like @claud9999 I have the same issue with this type of board - tried all hints from @PeterEmbedded - no way to flash - any idea? At this board works a ESP8285, not the ESP8266.

arendst, 2018-04-19:
Again. This boards is fine. Just follow the wiki and this thread.

newingit18, 2018-04-19:
Sure - good board and wiki, but in the described issue this answer is no help...

newingit18, 2018-04-20:
Dear all, seems that is an issue with the COM-Port - I used an existing COM1-port, without an USB-adapter. Then I used a FT232RL USB-Adapter - this worked fine, thanks all.

Marty56, 2018-04-28:
I had to set the switch from OFF to ON after flashing was completed. Otherwise no RF commands was recognized.

Marty56, 2018-04-28:
I managed to get an Intertechno switch to work with the RF bridge.
The Intertechno remote control is not detected. Although it can operate the same switch that I can control via RF bridge. Strange.

mo8Zomo0, 2018-05-01:
The newer RF bridge itself works fine, what I wondered, which IO ports 
are available and could be used for something else, like I would like to add
a temperature/humidity sensor.

MikhaelGerbet, 2018-06-16:
Hello everyone !
It's a good idea to flash  from the bottom side but, we don't close the box after :/
I have try to flash by the top side and here how to do :)

Info : the opening is done with cruciform screws hidden by the non slip skids :
[image]

I lifted the LED of the logo, I removed part of the protective mouse and I welded behind slightly inclined pines
[image]

I flashed the module
[image]

Then I replaced the LED in its place
[image]

And now, I was able to close the box :)
[image]

Enjoy :)

angustyler, 2018-06-25:
Flashing the RF Bridge with Tasmota 6.0.0a worked perfectly and is now accessible from the web page just fine. However, I'm now trying to install the Porticsh RF-Bridge-EFM8BB1 updated firmware by following the instructions:
[https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433) . But am getting error:
"_Upload Failed. Magic byte is not 0xE9_"
I've using the RF_Bridge.hex found at:
[https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF_Bridge.hex](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF_Bridge.hex)
(Note: I guess this is the correct location as the folder "tools/fw_efm8bb1" does not appear to exist - unless of course I am looking in the wrong place!)
Any help greatly appreciated!

bachya, 2018-06-30:
I'm in the same position as @angustyler. I've tried both:

* https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF_Bridge.hex
* https://github.com/arendst/Sonoff-Tasmota/blob/development/tools/fw_efm8bb1/RF_Bridge_00.hex

...and when I upload, I get the same `Upload Failed. Magic byte is not 0xE9` message.

newingit18, 2018-07-01:
Please look at the howto to prepare hardware for updating the EFM8BB1. I used the cutting of PCB strip lines as suggested in the picture (see https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433 - the last topic). It worked for me -I got success at the web gui of the Bridge.

joecool99, 2018-07-10:
in the section "Rf chip firmware upgrade" it is said, that "two not used copper traces need to be cut if you want to use the current power connector during updating".

Why is it necessary to cut these copper traces if they are not used?

Thank you!

joecool99, 2018-07-11:
Thank you, newingit18.

Maybe i'm blind, but i still do not see why it is necessary to cut these two traces because as far as i can see, the lower part of the cutted traces is not connected to anything and the endpoints of the cutted traces are not soldered, as far as i can see...

[image]

This can not be seen very good on this image because of the new cables but on my sonoff bridge i've tried to find a connection with a magnifier - but i didn't ...

Thany you.

newingit18, 2018-07-11:
If you are sure, that these strip lines are not connected to anything (I didn't check that electrically) - you can leave the lines uncutted. Keep in mind that this PCB can be a multi layer PCB, so you can not see all connections inside the multi layer PCB - you have to check this with a AC/DC multimeter to be sure.

joecool99, 2018-07-11:
I'm not sure - that's my problem ;-)

I want to avoid a permanent change to the board, if it's not necessary, and as you said, if the board is multi layer - cutting something might be especially harmfull ...

arendst, 2018-07-11:
Guys, the issue I wanted to raise here is the fact that the lower part of the traces are connected to the USB connector. If you connect external power to the USB connector these traces might carry either power or a USB signal depending where you get your power from which will interfere with the MCU flash signals.

So either do not use the USB power while MCU flashing but connect power to other pins on the header or cut the traces and use the USB connector to power the thing.

joecool99, 2018-07-11:
OK, thank you very much.

newingit18 brought me to the right idea, i've just checked the other side of the board and i'v found what you say:

[image]

Thanks you very much :-)

PeterEmbedded, 2018-07-11:
Just for some more information. These "endpoints" are actually vias.
https://en.wikipedia.org/wiki/Via_(electronics)

videobuff, 2018-07-15:
I aso have the same problem, and not yet resorted to cutting the pcb traces. 
The tasmota firmware update went without a glitch:
Programma Versie | 6.1.0
2018-07-16T00:42:10
2_3_0/1.5.3(aec24ac9)
0T00:09:24
And i can see the webpage.

Anything else to try before cutting the traces ?

nnhhuu, 2018-08-22:
I have an issue,  in setting i can't find the Flash: 1M (no SPIFFS), these are the option i have

[image]

can i use other option here.?

ascillato, 2018-08-22:
Offtopic. 

Please, read the wiki

https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

Geoff571, 2018-10-09:
OK, I'm confused. 

I read the Wiki on the Sonoff Bridge and it implies it should be flashed as an ESP8266, however the chip is actually an ESP8285. I saw a YT video that specifically set Arduino IDE to flash for an ESP8285. Which method is correct?

ascillato, 2018-10-09:
Both will work just fine. Better is to follow the wiki that has proven to work.

Geoff571, 2018-10-09:
Many thanks @ascillato, I flashed is as per the Wiki article and I'm a happy bunny

TerryLansdown, 2019-01-04:
Hi, uploaded the 6.4.1.5 firmware with no big problems. Seems to run fine via a USB-serial adapter. If I reboot with the USB as the power source (not the serial connection) the RF bridge seems to go into a reboot loop. Can't connect to the serial to confirm that. Any suggestions? Should I be cutting tracks on the board as discussed above, that doesn't seem to be necessary. Thanks.

ozett, 2019-01-07:
i also have no luck to flash my new sonoff bridge.
as i checked the wiki 10 times (as some advised here),
i think the [image](https://github.com/arendst/Sonoff-Tasmota/issues/1916#issuecomment-404078512) for the new-version her in this thread as on the wiki is somehow **wrong**. !?

the wiki states:
```
Connect two wires between GPIO4 and C2Ck and GPIO5 and C2D.
```
but if you check the image and revert the PCB, the wires should be crossed.

all you guys flashed successfully made it right, all others are irritated about some (maybe) misleading information in the wiki? 
any hints, if the image in the wiki is right, or my board is different?

[image]

ozett, 2019-01-07:
at least i am on the module now to check if it works.
but as there are a bunch of modules/names one get confused again.. 😁 
maybe would be good to reference the modules also by number in the wiki?
(just to make clear that this "bridge" is not any other "bridge", than as long as there are already numbered in the firmware and in the dropdown a number seems clearer )
[image]

Fietspomp86, 2019-01-11:
I flashed mine according the wiki. I have a R2 V1.0.
So my wires are not like yours. Also I cut the 2 lines.
Flashed Tasmota using the normal procedure (Used ESPeasy flasher with precompiled binary).
Then flashed the hex file using the web-interface of Tasmota.
[image]

Villhellm, 2019-01-21:
I have successfully flashed my R2 V1.0 without resoldering/cuttiing traces, but the bridge is not receiving any signals from my RF devices. The switch is definitely set back to the on position, still nothing. Anyone have a similar problem?

tedsluis, 2019-01-27:
I am having problems with the RF bridge R2 v1.0 marked "2017.11.23". It looks like flashing went okay, no errors, but after powering the RF bridge using the micro usb I can not connect to it. 

To flash I use a Fedora linux system with a CH340G USB serial adapter. I use VSCode with the PlatformIO plugin.  I tried the Sonoff-Tosmota 6.4.1 (release) and the 6.4.1.9 (development).  I have quite some experience with successful flashing other Sonoff devices, like Basic, Pow R2, 1CH and S26. 

### Preparations 
I did not used the micro usb power during flashing, so I did not cut the copper traces on the board, as mentioned in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433#flash-process-of-hardware-r2-without-hardware-preparation). 

Like @ozet, I was confused by the instructions in the wiki on the [flash process](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433#flash-process). The image was different then the wiki text?! So I first tried the wiring according the image in the wiki and later on, as mentioned by @ozet, I also tried the wiring as advised in the wiki text. 

My board (with wiring according wiki image, this does not match with the wiki text):
[image]

Wiring according the wiki image: 
[image]

Wiring according wiki text:
[image]

Both ways of wiring gave the same result: The flashing process finished successful, but I was not able to connect to RF bridge.

### Steps I took to flash:
* connected the RF bridge board to a CH340G USB serial adapter: RX-TX, TX-RX, GND to GND. 
* set switch to off.
* pressed and hold the button.
* Inserted the CH340G USB serial adapter into my laptop.
* connected RF bridge board 3v3 pin to CH340G USB serial adapter pin 3v3.
* waited a few seconds and released the button on the board.
* waited a few seconds and start flashing.

The Flash logging is the same for both ways of wiring:
```
> Executing task: platformio run --target upload <

Processing sonoff (framework: arduino; platform: espressif8266@1.8.0; board: esp01_1m)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Converting sonoff.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 53 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <OneWire> 2.3.2
|-- <ArduinoJson> 5.11.2
|-- <LiquidCrystal_I2C>
|   |-- <Wire> 1.0
|-- <TasmotaMqtt> 1.0.0
|-- <Adafruit ILI9341> 1.2.0
|   |-- <Adafruit GFX Library> 1.2.9
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|-- <PubSubClient> 2.6
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <I2Cdevlib-MPU6050>
|   |-- <I2Cdevlib-Core>
|   |   |-- <Wire> 1.0
|-- <NeoPixelBus> 2.2.9
|   |-- <SPI> 1.0
|-- <ESP KNX IP Library> 0.5.1
|   |-- <ESP8266WiFi> 1.0
|   |-- <EEPROM> 1.0
|   |-- <ESP8266WebServer> 1.0
[... 91 lines cut ...]
Of course I have repeated the flash process several times. Many times I first erased the firmway using this command:
```
$ ./esptool.py --port /dev/ttyUSB0 erase_flash
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
MAC: 60:01:94:be:7d:d1
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.0s
Hard resetting via RTS pin...
``` 

I am still puzzled about the correct way of wiring. Can any one confirm the correct wiring for my type of board? 

Isn't strange that I am able to flash the the RF bridge, but that it does not work? What can be wrong?

I even tried a second RF bridge and I ended up with the same results.

How can I debug this?

TerryLansdown, 2019-01-28:
> How can I debug this?

Hi, See my earlier post for the wiring that works for me.

Sounds like you know what you're doing, but maybe you didn't try:
1. checking the serial monitor for device boot/network connect messages? I think I tested mine with the USB-serial cable powering it, first. 
2. checking if the bridge is appearing on your network, ping'ing it?
3. flashing a different way? I used the Arduino SDK to flash mine.

What do you mean by 'I was not able to connect to RF bridge'? Is it dead? Can you ping it, but not connect to the web interface? On mine, powering by the micro-usb cable resulted in clicking reboots, until I fixed the wiring.

tedsluis, 2019-01-28:
Hi, thnx for your extensive reply!

TerryLansdown said:
>Hi, See my earlier post for the wiring that works for me.

Sure I have read your post, but it was/is not 100% clear to me whether your board has the "2017.11.23" mark, like my board. Can you confirm that?

TerryLansdown said:
>1. checking the serial monitor for device boot/network connect messages? I think I tested mine with the USB-serial cable powering it, first.

I tried that, but without luck. I am not sure how to do it on a Sonoff RF Bridge. 
I had my RF bridge connected using the TX, RX, 3v3 and GND to the CH340G serial adaptor. I did not connected it to a micro usb power. Is that correct?
I tried the [cu](https://www.cyberciti.biz/hardware/5-linux-unix-commands-for-connecting-to-the-serial-console/) command (Fedora command line, I have used this for other board successfully):
``` 
$ cu -l /dev/ttyUSB0 -s 19200
```
I didn't got any response. How should it react?
I also tried to put the RF bridge in flash mode, in order to get any response. 
What steps should I take to start a console? 
What tool did you use? 

TerryLansdown said:
>2. checking if the bridge is appearing on your network, ping'ing it?   

The IP address of the bridge did not appear in my network. So I could not ping it or connect to the web interface.

TerryLansdown said:
> 3. flashing a different way? I used the Arduino SDK to flash mine.

I will try that, may be tonight (or in the coming days).

TerryLansdown said:
> On mine, powering by the micro-usb cable resulted in clicking reboots, until I fixed the wiring.

My wiring is like yours now. When I power on the RF bridge using the micro usb, the big green LED turns "on". No other LED flashes, no beep and no reaction when I press the button.

I used the same Sonoff-Tasmota firmware + configuration as I used for other Sonoff devices (except for the IP address and topic name).

To flash the RF bridge I tried an upload  speed of 115200 as well as 19200 (takes much longer to finish). Which is correct? 

To test whether my current Vscode with PlatformIO setup still works, I have flashed a Sonoff Basic device yesterday. It still works.

TerryLansdown, 2019-01-28:
Hi,

0. Don't know, it's now deep in a cupboard put back together. If I get a chance I'll have a look, from memory it was a 2017 version.

1. Again (from memory), I flashed with just the USB-Serial adapter. When uploaded, within the Arduino SDK there is a serial monitoring option. Using only the serial adapter (I think) I got status updates, connected to local network, connected to MQTT server, etc. You may have seen these with your Sonoff Basics? Only after a successful flash (using the Arduino SDK and the image from the main Tasmota Git pages; did I unplug the serial and power using just micro-usb cable.

I'm guessing your flash is a big fail if you can even see noise using 'cu'.
Think you did the right thing to zero out the board between each flash with esptool.

2. Smells like a big fail.
3. The Arduino SDK worked for me, good luck.

I don't think I needed to press any buttons to get the board up. Move onboard switch from flash to broadcast, then plug in and it connected to my network, and MQTT server. I could then access the webpages, much like (presumably) your other Sonoffs.

From memory, I think I tried flashing at both speeds, I think it flashed at 115200 and then it definitely communicates at 19200.

Other than trying a different flashing technique specifically for this board, sorry I'm out of ideas.

ozett, 2019-01-29:
may you try the espeasy flashtools?
i had success with them: https://github.com/arendst/Sonoff-Tasmota/issues/1916#issuecomment-452069128

LucReynders, 2019-01-29:
Received my RFBridge yesterday.
Flashed it with Tasmota : (! no extra wires have been added,  only the pins - no 5V supply during flash))
- Dismantled the device (4 screws on the bottom)
- lifted the "sonoff display" 90°
- added 4 connector pins to the board (GND, RX, TX, 3.3V)
- switched the on-board switch to "OFF"
- connected the USB flash device to the pins
- pushed the on-board button and connected the flash device to the USB port (Imac with PlatformIO)
- released the button and flashed TASMOTA on the RFBridge.
- found the IP address of the device
- configured the sonoff device as usual (25 bridge)
- put the on-board switch back to "ON"
- shortened the connection pins and put the display back in place (you can remove the black seal partly)
- assemble the unit.
(worked perfectly after teaching "Rfkeys<x> 2"  on the console and sending MQTT in Nodered)

tedsluis, 2019-01-29:
Thnx @TerryLansdown, @Ozett and @LucReynders. 

@TerryLansdown said:
> The Arduino SDK worked for me, good luck.

It took me a while to get the Arduino IDE working on Fedora Linux. I guess most people use Windows. I followed the [wiki instructions](https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE) and I managed to build and upload the Sonoff-Tasmota firmware. Unfortunately still the same result: I can not connect to the bridge, but I keep positive. This is only my first attempt with the Arduino IDE :-) 
```
Build options changed, rebuilding all
Archiving built core (caching) in: /tmp/arduino_cache_674523/core/core_esp8266_esp8266_generic_CpuFrequency_80,VTable_flash,ResetMethod_nodemcu,CrystalFreq_26,FlashFreq_40,FlashMode_dout,FlashSize_1M0,led_2,LwIPVariant_v2mss1460,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_115200_c1df69fc8c56a963385ffb471ec2f0ff.a
Sketch uses 529928 bytes (51%) of program storage space. Maximum is 1023984 bytes.
Global variables use 51604 bytes (62%) of dynamic memory, leaving 30316 bytes for local variables. Maximum is 81920 bytes.
Uploading 534080 bytes from /tmp/arduino_build_804524/sonoff.ino.bin to flash at 0x00000000
................................................................................ [ 15% ]
................................................................................ [ 30% ]
................................................................................ [ 45% ]
................................................................................ [ 61% ]
................................................................................ [ 76% ]
................................................................................ [ 91% ]
..........................................                                       [ 100% ]
```
No output on the serial monitor yet, but I need some more time to get to know the Arduino IDE.

@Ozett said:
> may you try the espeasy flashtools?

Unfortunately I only have Linux systems. I think `espeasy` only runs on Windows.
Your previous post is very useful!

@LucReynders said:
> Flashed it with Tasmota : (! no extra wires have been added, only the pins - no 5V supply during flash))

No extra wiring? Very interesting. Is your RF bridge marked "2017.11.23"? 
Thnx for your detailed steps. Very useful!

I need some more time to carry out all the instructions and tips. I let you guys know in the coming days.

ozett, 2019-01-30:
> Unfortunately I only have Linux systems. I think `espeasy` only runs on Windows.
> Your previous post is very useful!

Thanx,
Espeasy is a firmware for the ESP8266/Esp32 chips.
they have also firmware flash-tools.
i think there is a linux-variant of the good working windows tool.

may there some inspirations for your experiment:
https://www.letscontrolit.com/wiki/index.php/Flash_script_linux

molexuse, 2019-02-01:
> I have successfully flashed my R2 V1.0 without resoldering/cuttiing traces, but the bridge is not receiving any signals from my RF devices. The switch is definitely set back to the on position, still nothing. Anyone have a similar problem?

Hi
Same problem. Flash without cut.
It`s works. 
Rf received (red led flash) but in console serial and mqtt rf not send. I try diferent firmware with no success.
As i understand from schematic esp communicate with rf chip by uart still analyze....

tedsluis, 2019-02-03:
I promised to let you guys know how it goes. I managed to get it working! 

I tried other flash tools (Arduino, esptool and espeasy), several Sonoff Tasmota firmware versions and several configuration without luck. In all cases it looked like the RF bridge was successfully flashed (without errors), but I could not connect to it via WIFI. Believe me, I was getting more and more frustrated every day ;-) 

As a last attempt I did [cut the copper traces, as explained in the wiki,](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433#flash-process-of-hardware-r2-with-hardware-preparation) and powered the RF bridge using the micro USB during flashing, but that also did not worked out. (I have two Sonoff RF bridges, so I dared to do this experiment.)

At some point I read a [dutch blog post](https://ictoblog.nl/2018/12/23/draadloze-schakelaar-met-sonoff-rf-bridge-tasmota-home-assistant) about an Sonoff RF bridge on flash voltage (3v3 or 5v). This guy also was not able to flash his RF bridge with a CH340G based USB serial adapter. He switched from 3v3 to 5v (against all recommendations) and it worked (at least for him). He reported that with 5v the green LED was much brighter. 

I had high hopes, because his serial CH340G USB adapter looked like mine. Unfortunately his solution did not work for me. 

Something that I kept thinking about was that his green LED was bright and mine was not. May be my serial USB adapter is the cause of the issue. 

This afternoon I tried another CH340G based USB serial adapter and the LED shine bright and it immediately worked! It tried my VSCode with PlatformIO setup, as well as my Arduino setup and my esptool setup. They all worked!

Below my CH340G based serial USB adapters. The upper one (SDS011_USB2TTL_004) works for both the Sonoff RF bridge and other Sonoff devices. 
[image]
The lower one (HW-597) does not work with the RF bridge, but I am still able to flash other Sonoff devices (like the Basic, S26 and the Pow R2) with this adapter.

Thank you everybody for all your tips and support. It helped me trying new tools, but most of all: excluding possible causes! Very useful! Thanks a lot!

LucReynders, 2019-02-06:
With the standard Tasmota flash the RFBridge is limited in detecting codes.
After flashing the standard Tasmota package you can upgrade the bridge with Portisch software.
This can be done via the console (but first you need to connect the extra wires to the board
* IO5<->C2CK and IO4<->C2D as explained in the previous discussions for the R2.).
For the upgrade you need first to download the portisch hex file.  https://github.com/arendst/Sonoff-Tasmota/tree/development/tools/fw_efm8bb1.
In the console you can then "flash" upgrade by selecting this file.

After the portisch upgrade I was able to sniff remote codes from Chacon and my Novy cooker hood with the Rfraw 177 command.  The sniffed B1 commands need to be converted to B0 first as explained in github before sending these over the bridge (with bitbucketconverter tool).

AntonioCorrenti, 2019-03-03:
Hi guys,
First of all thanks for all the support and the great work.
I want to start with what always works for me when flashing sonoff's esp8255/66 that's not covered in any guide.

1. Avoid connecting 3.3v between the programmer and the sonoff (bridge, basic, dual). Try always to get external power. I used a Breadboard power adapter.
2. Share the grounds. I tried different setups, but what always works for me is to share GND between the sonoff, the breadboard adapter and the FTDI adapter.

Now unfortunately all the remotes I used with ewelink, are not working with Tasmota. The console sniffs the code but pairing fails always. IDK if I should upgrade the RF chip with Portisch firmware, because as I said everything was fine with ewelink.
Has someone had the same problem?

sndipz, 2019-04-07:
> I have successfully flashed my R2 V1.0 without resoldering/cuttiing traces, but the bridge is not receiving any signals from my RF devices. The switch is definitely set back to the on position, still nothing. Anyone have a similar problem?

Hi, I am facing same issue. Have you figured out an alternate?

Jason2866, 2019-05-16:
Picture save for wiring in wiki
[image]

TurboTronix, 2019-06-12:
I flashed everything as described, I did manage to get the codes but not to send them. I am now trying to reflash the hex file but keep getting: error "Magic byte is not 0xE9"

LucReynders, 2019-06-13:
"After the portisch upgrade I was able to sniff remote codes from Chacon and my Novy cooker hood with the Rfraw 177 command. The sniffed B1 commands need to be converted to B0 first as explained in github before sending these over the bridge (with bitbucketconverter tool)."
- Use cmd rfraw 177
- send codes with the remote 
- the bridge will "sniff" the B1 codes send,  the  sniffed may be different
- select the best matching code
- use the bitbuckedconv for this code
- send the  B0 code to the bridge as a payload

Freestila, 2019-12-19:
Hello,
i hope it is ok that i post in this "closed" issue.
I have flashed my rf bridge v2 successfull. Tried with an rf remote, and it did show some data in the console. Since i wanted to send data, i flashed portisch firmware (with cutted traces), and according to the messages everything was fine.
After that, it seems i can no longer receive any data. Tried multiple 433 mhz remotes and sensors, nothing gives an output in the console. reflashed original rf firmware, again nothing to see. reflashed polisch, tried the rfraw 177 / rfraw AAB155 command, this gives the "Data":"AAA055" result but nothing more.
Any suggestions? is it possible that i cut too deep and damaged something below? Or that the flash was not correct (i used breadboard wires, did not solder anything)?

ascillato, 2019-12-19:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

syepes, 2020-01-07:
Just wanted to thank all the hard work on this one.
I just flashed Tasmota + Portisch on my R2 v1.0 and its working as expected.

LucReynders, 2020-03-14:
Hmmm.
I tried to flash several times another rf bridge with esptool on a linux laptop.
The flashing looked ok but no wifi ap connection visible.
When I finally flashed with nodemcuflasher on my Imac every went smooth.  
I assume that esptool does not fully erase the esp before flashing and nodemcuflash does.
So if you may have a similar problem try nodemcuflash.

Jason2866, 2020-03-14:
Esptool.py does what told. So it does no erase before flashing.
Recommended flash tool for Tasmota is Tasmotizer

LucReynders, 2020-03-14:
I tried to install on Linux Debian but this is what returns with install option 2 and 3.
Can you assist or help me with this ?

lurey@Linux:~$ pip3 install PyQt5 pyserial
Collecting PyQt5
  Using cached https://files.pythonhosted.org/packages/3a/fb/eb51731f2dc7c22d8e1a63ba88fb702727b324c6352183a32f27f73b8116/PyQt5-5.14.1.tar.gz
  Installing build dependencies ... done
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python3.7/tokenize.py", line 447, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-install-z1lxkwlx/PyQt5/setup.py'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-z1lxkwlx/PyQt5/

barbudor, 2021-10-06:
Don't cross post

dbuezas, 2021-11-06:
Direct hack for R2.2
https://community.home-assistant.io/t/new-sonoff-rf-bridge-board-need-flashing-help/344326/17

ascillato, 2021-11-06:
https://github.com/arendst/Tasmota/discussions/13283