arendst/Tasmota #1918: Sonoff keep restarting

smekcio, 2018-02-14:
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_3B73A2, GroupTopic sonoffs) Version 5.12.0b
00:00:⸮⸮UJi Connecting to AP1 <here is my ssid> in mode 11N as sonoff-5026...
00:00:04 WIF: Connected

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

Any idea?

reloxx13, 2018-02-15:
Try reflash it:

Even if it says 100% done there can be errors.

Try another USB Cable.
Try another PIN Cables.
Try another FTDI (does the green SMDs on the FTDI Flasher flash once if you plug it in USB and orange FTDI always on ?) 1/3 was defect which i bought form china.
Check your Soldered Pin connections or re-solder to be sure and check the cable order (see Wiki Page for Sonoff Basic)

Be sure to hold the Button down while pluging in USB.

The Button can be also defect.

ALWAYS use TX-RX RX-TX.

Be sure you OPENED the project with the platformio.ini to select and not created a new project.

arendst, 2018-02-15:
Kind of known problem with 2.4.0 library (https://github.com/esp8266/Arduino/pull/4105)

If compiled on Arduino IDE check your LwIP settings as documented in the wiki.

As last resort use 2.3.0 library.

reloxx13, 2018-02-15:
I never had such problems with 2.4.0 and Atom/PlatformIO.

And since im always using your Dev Branch, im updating very often.

Thanks for the earwig 👯‍♂️ 
*this is my last resort* *flies away*

smekcio, 2018-02-15:
I have used VSCode with PlatformIO and 0 problems.

closed 2018-02-15 by smekcio