arendst/Tasmota #1919: unable to compile on Arduino IDE

Supermillhouse, 2018-02-14:
I get this never ending loop when trying to compile on Arduino IDE 1.8.5
"Detecting libraries used.." keeps spitting out this line, it gets longer over time

C:\Home automation\arduino-1.8.5\arduino-builder -dump-prefs -logger=machine -hardware C:\Home automation\arduino-1.8.5\hardware -hardware C:\Home automation\arduino-1.8.5\portable\packages -tools C:\Home automation\arduino-1.8.5\tools-builder -tools C:\Home automation\arduino-1.8.5\hardware\tools\avr -tools C:\Home automation\arduino-1.8.5\portable\packages -built-in-libraries C:\Home automation\arduino-1.8.5\libraries -libraries C:\Home automation\arduino-1.8.5\portable\sketchbook\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,FlashSize=1M0,LwIPVariant=Prebuilt,Debug=Disabled,DebugLevel=None____,UploadSpeed=115200 -ide-version=10805 -build-path C:\Users\IANSHE~1\AppData\Local\Temp\arduino_build_963994 -warnings=none -build-cache C:\Users\IANSHE~1\AppData\Local\Temp\arduino_cache_281567 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Home automation\arduino-1.8.5\portable\packages\esp8266\tools\esptool\0.4.12 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Home automation\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Home automation\arduino-1.8.5\portable\packages\esp8266\tools\mkspiffs\0.2.0 -verbose C:\Home automation\arduino-1.8.5\portable\sketchbook\sonoff\sonoff.ino
C:\Home automation\arduino-1.8.5\arduino-builder -compile -logger=machine -hardware C:\Home automation\arduino-1.8.5\hardware -hardware C:\Home automation\arduino-1.8.5\portable\packages -tools C:\Home automation\arduino-1.8.5\tools-builder -tools C:\Home automation\arduino-1.8.5\hardware\tools\avr -tools C:\Home automation\arduino-1.8.5\portable\packages -built-in-libraries C:\Home automation\arduino-1.8.5\libraries -libraries C:\Home automation\arduino-1.8.5\portable\sketchbook\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,FlashSize=1M0,LwIPVariant=Prebuilt,Debug=Disabled,DebugLevel=None____,UploadSpeed=115200 -ide-version=10805 -build-path C:\Users\IANSHE~1\AppData\Local\Temp\arduino_build_963994 -warnings=none -build-cache C:\Users\IANSHE~1\AppData\Local\Temp\arduino_cache_281567 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Home automation\arduino-1.8.5\portable\packages\esp8266\tools\esptool\0.4.12 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Home automation\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Home automation\arduino-1.8.5\portable\packages\esp8266\tools\mkspiffs\0.2.0 -verbose C:\Home automation\arduino-1.8.5\portable\sketchbook\sonoff\sonoff.ino
Using board 'generic' from platform in folder: C:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0
Using core 'esp8266' from platform in folder: C:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0
Detecting libraries used...
"C:\Home automation\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/include" "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip/include" "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\IANSHE~1\AppData\Local\Temp\arduino_build_963994/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC"  -DESP8266 "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266" "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0\variants\generic" "C:\Users\IANSHE~1\AppData\Local\Temp\arduino_build_963994\sketch\sonoff.ino.cpp" -o "nul"
"C:\Home automation\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/include" "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip/include" "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\IANSHE~1\AppData\Local\Temp\arduino_build_963994/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC"  -DESP8266 "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266" "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0\variants\generic" "-IC:\Home automation\arduino-1.8.5\portable\sketchbook\libraries\PubSubClient-2.6.09\src" "C:\Users\IANSHE~1\AppData\Local\Temp\arduino_build_963994\sketch\sonoff.ino.cpp" -o "nul"
"C:\Home automation\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/include" "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip/include" "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\IANSHE~1\AppData\Local\Temp\arduino_build_963994/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC"  -DESP8266 "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266" "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0\variants\generic" "-IC:\Home automation\arduino-1.8.5\portable\sketchbook\libraries\PubSubClient-2.6.09\src" "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0\libraries\Ticker" "C:\Users\IANSHE~1\AppData\Local\Temp\arduino_build_963994\sketch\sonoff.ino.cpp" -o "nul"
"C:\Home automation\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/include" "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip/include" "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\IANSHE~1\AppData\Local\Temp\arduino_build_963994/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC"  -DESP8266 "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266" "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0\variants\generic" "-IC:\Home automation\arduino-1.8.5\portable\sketchbook\libraries\PubSubClient-2.6.09\src" "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0\libraries\Ticker" "-IC:\Home automation\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src" "C:\Users\IANSHE~1\AppData\Local\Temp\arduino_build_963994\sketch\sonoff.ino.cpp" -o "nul"

Supermillhouse, 2018-02-14:
this may be in error but it looks like it is making progress now compiling library's, I am new to this and am shocked at how long it takes to progress, 5 mins so far and the green bar is half way across now

ddamron, 2018-02-14:
Supermillhouse, 

Your log there doesn't contain any errors.  Lots of warnings are normal.
If you followed the instructions, it will eventually finish without error.  This is a big project, it will take some time to compile, compared to standard arduino projects..
If it fails, check to make sure you copied your libraries over, check processor options, etc.

Supermillhouse, 2018-02-14:
ok, is it expected to take 5 min plus, videos on youtube must cut out that bit. :)