arendst/Tasmota #1921: Customise tasmota - ready to pay

sam9s, 2018-02-15:
Hi Developer,

I am using sonoff with tasmota like crazy and have installed it on many homes including friends and relatives. I would like to use tasmota in other, home electronic appliance automation as well. So would like to know if a small change is possible. I am ready to pay for the efforts to the developer as well if this is a big deal.

I would like to use tasmota for home applicances that would require power for a small stipulated period of time say like 30 sec or 5 mins etc. This is not like a scheduling, where a device recieves power every said day, week , but changes in the firmware where when I shoot the command "alexa switch on coffee maker" sonoff running tasmota should trigger the power ON (relay) and then trigger it OFF after a defined interval of time say 5 mins.  The time should be user defined on the web page of tasmota. So to summarize, when I load tasmota, I should be able to access the web page, click a button and define a time in mins say 5, and then reboot the device. Now let alexa discover the device and then when I say alexa switch on said device it should switch it on and then auto switch it off after the stipulated time whch we defined on the web interface, which is 5 min in this example.

Q to developer... is this possible?, if its easy to do please provide pointers, how to do it and I shall give my best, if its complecated then I am ready to pay if you can come up with a custom tasmota firmware for me specifically.

Please let me know
Regards
Sammy

reloxx13, 2018-02-15:
Dude, its PuleTime, check the Wiki Command Page.

Should be Pulestime 400 for 5mins.

But i dont know if PulseTime is affected by VoiceCommand.

Just try it with a smaller Pulsetime (eg. 112 for 12 Seconds).

sam9s, 2018-02-15:
ok so is pulse time disabled currently, coz when we pass on the command to switch on a device it "remains" switched on until the time we give the command again to switch off the device.

sam9s, 2018-02-15:
if this is the case  ...lol then its all sorted .... thanks reloxx13 "again" for the help and pointing me in the correct direction

reloxx13, 2018-02-15:
Type in console:

SaveData 1
PulseTime 400
SaveData 0

Pls give feedback if pulsetime is also working if device was turned on by alexa :)

reloxx13, 2018-02-15:
PulseTime1 = Relais 1

sam9s, 2018-02-15:
ok but I dont want to use console, I want this to be user defined via web interface, so that my "not so technical" public can use it  ..... 

other wise I will try console method as well and report back .....thanks again

sam9s, 2018-02-15:
OK just needed to report that pulse function DOES work and with alexa as well. Just tried on my bedroom lights and they do go off after 13 sec after seting pulse value to 113.

sam9s, 2018-02-15:
this solves the issue ... thank you so much reloxx13 again for assisting me on this ..... since I am using tasmota like crazy, do expect more help and queries, 

Regards
Sammy

reloxx13, 2018-02-15:
for easier use u could use http command

http://ip-from-your-device/cm?cmnd=pulsetime1%20113

i also plan to implement it in sonweb 👍

tango156157, 2018-02-22:
Helloz, this is my first post in the TASMOTA support forum. I have a similar requirement as in this Post for Pulse time. I however want to modify it in the User.config.h file and create my own built under #define APP_PULSETIME and change it to say 2 seconds. Not sure if I should be putting 0.20 value here. Please do guide. Many thanks for the excellent work done for creating TASMOTA. I love it !  Cheers~

sam9s, 2018-02-23:
I would prefer to have pulse time be available via user web interface, under config, so if there is a way to incorporate pulse time on the web interface that uses has access to, that would be truly ultimate. I myself is working on it, but I am not that technical so might not be able to figure this out.