arendst/Tasmota #1922: IDE Tools - Flash Size: "1M (no SPIFFS)" is not available on board "generic esp2266 module"

rthepen, 2018-02-15:
IDE Tools - Flash Size: "1M (no SPIFFS)" is not available on board "generic esp2266 module"
Same issue like this one. 
https://github.com/arendst/Sonoff-Tasmota/issues/583

I can not find any information on wiki prerequisites or setup wiki.
can someone give me a link to the right page?

Thanx!

[image]

reloxx13, 2018-02-15:
From the [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE#copy-files):

> ESP8266 Board version before 2.4.0
> 
> If not available copy from the Tasmota release Source code folder arduino\version 2.3.0\tools\sdk\ld file eagle.flash.1m0.ld to Arduino IDE folder portable\packages\esp8266\hardware\esp8266\2.3.0\tools\sdk\ld.
> 
> Replace in Arduino IDE folder portable\packages\esp8266\hardware\esp8266\2.3.0 file boards.txt with the Tasmota Source code file arduino\version 2.3.0\boards.txt.
> ESP8266 Board version 2.4.0
> 
> To remove default compile options for floating point printf and scanf which are not being used by Tasmota and save 15k code space replace in Arduino IDE folder portable\packages\esp8266\hardware\esp8266\2.3.0 file platform.txt with the Tasmota Source code file arduino\version 2.4.0\platform.txt.

rthepen, 2018-02-15:
thanks, it works now!

closed 2018-02-15 by rthepen