arendst/Tasmota #1923: Switch on GPIO - strange behavior.

tombartosz, 2018-02-15:
I have Sonoff Basic is connected to NC relay using GPIO0.

GPIO0 has pull up resistor 4.7kOhm (near GPIO) and between GPIO0 and GND there is 1nF capacitor, NC relay is connected to GPIO and GND. Also in parallel to relay (near relay) there is 100kOhm resistor.

Between Sonoff and the relay there is about 15m of wire (regular electric wire)

SonOff has GPIO0 configured as SWITCH in mode FOLLOW. SwitchTopic is  set to "termostat". 

When relay is connected (GPIO0 connected to GND) all works ok and message "cmnd/termostat/POWER1 OFF" is published.

When relay is disconnected (GPIO0 pulled up by 4.7k resistor in Sonoff and connected through wire and 100k resistor to GND) strange things happens. Sonoff reads ON and OFF like in the log below.
Relay is not "clicking" - it is all the time connected.
 

```
[02-15 11:05:39.611] cmnd/termostat/POWER1 ON
[02-15 11:26:54.233] cmnd/termostat/POWER1 OFF
[02-15 11:26:54.280] cmnd/termostat/POWER1 ON
[02-15 11:27:13.076] cmnd/termostat/POWER1 OFF
[02-15 11:27:13.126] cmnd/termostat/POWER1 ON
[02-15 11:27:13.276] cmnd/termostat/POWER1 OFF
[02-15 11:27:13.326] cmnd/termostat/POWER1 ON
[02-15 11:27:17.583] cmnd/termostat/POWER1 OFF
[02-15 11:27:17.627] cmnd/termostat/POWER1 ON
[02-15 11:27:43.927] cmnd/termostat/POWER1 OFF
[02-15 11:27:43.976] cmnd/termostat/POWER1 ON
[02-15 11:27:45.527] cmnd/termostat/POWER1 OFF
[02-15 11:27:45.576] cmnd/termostat/POWER1 ON
[02-15 11:27:47.027] cmnd/termostat/POWER1 OFF
[02-15 11:27:47.076] cmnd/termostat/POWER1 ON
[02-15 11:27:47.227] cmnd/termostat/POWER1 OFF
[02-15 11:27:47.276] cmnd/termostat/POWER1 ON
[02-15 11:27:48.633] cmnd/termostat/POWER1 OFF
[02-15 11:27:48.703] cmnd/termostat/POWER1 ON
[02-15 11:27:53.727] cmnd/termostat/POWER1 OFF
[02-15 11:27:53.777] cmnd/termostat/POWER1 ON
```

It should be all the time ON and only one message.

What can be cause of this strange behaviour ?

Have a nice day, Bartosz.

Jason2866, 2018-02-17:
Did i understand you correct, you have directly connect a relay to GPIO 0?
This will not work if it ever does for a longer time. The relay needs to much current and has a inductive load. The GPIO pins needs a driver for that!

tombartosz, 2018-02-17:
No, maybe i was not clear enough. It is relay output of another device,
from sonoff point of view it is like a switch.

Jason2866, 2018-02-17:
What happens if you disconnect the 15m wires? Leave the pullup resistor in.

Jason2866, 2018-02-17:
The setup of the sonoff seems correct. The wires are too long to connect to a "digital" input like the GPIO pins. You have build an antenna with your normal electrial wires. The resitance of the wire is not the problem. What you can try is to use shielded cable. Thin and cheap is shielded audio cable. Cheap Coax Cable you can use too. Shield from cable connected to ground of Sonoff.

tombartosz, 2018-02-17:
Schielded cable is not the option. Existing cable is inside wall and i
don't want to damage it.

Jason2866, 2018-02-17:
Take a look here
http://www.nutsvolts.com/magazine/article/optocoupler-circuits

tombartosz, 2018-02-28:
Problem solved. I realized that at the end of long wire I had 2.5V when it should be pulled up to Vcc. I changed GPIO0 pull-up resistor to 1kOhm and increased resistor between GPIO and GND (at the end of wire - switch/relay connectors) to 600kOhm. 
It solves problem.

closed 2018-02-28 by tombartosz