arendst/Tasmota #1924: Hue Emulation as a "Hue client" (without bridge)

JoeWiseman, 2018-02-15:
Would it be possible to emulate a "pure Hue client" (without the bridge/gateway) emulation for the Sonoff-devices, so that they can be connected and controlled by one Hue bridge?

I hope that this can be done wihout much effort .... 

Regards
Joe

Frogmore42, 2018-02-16:
The Hue client is not Wi-Fi as far as I know.  So I don't believe it is possible.  But, I think the Hue app might be able to handle more than one bridge.  I know the Smart Things app can.

Frogmore42, 2018-02-16:
https://www.engineersgarage.com/contribution/zigbee-vs-wi-fi

JoeWiseman, 2018-02-16:
Hi @Frogmore42 
thanks a lot for explaining the differences in communicating to an bridge and client.
So that seems to be an dead end for me.
Thanks again!
Joe

closed 2018-02-16 by JoeWiseman